Koenkk/zigbee2mqtt #23620: Moes BHT-002/BHT-006 - Max temperature limit - always NULL

error105, 2024-08-14:
### What happened?

When im try set Max temperature limit via MQTT (zigbee2mqtt/MyFriedlyName/set - {"max_temperature_limit": 25} or via ZigBee2mqtt WebUI nothing heppend, still in status i have there NULL value. Same in deadzone_temperature.

`{
    "child_lock": "UNLOCK",
    "current_heating_setpoint": 21,
    "deadzone_temperature": null,
    "heat": "OFF",
    "linkquality": 105,
    "local_temperature": 21.5,
    "local_temperature_calibration": -4,
    "max_temperature": 45,
    "max_temperature_limit": null,
    "min_temperature_limit": 5,
    "preset": "hold",
    "preset_mode": "hold",
    "program": {
        "saturday_p1_hour": 6,
        "saturday_p1_minute": 0,
        "saturday_p1_temperature": 24,
        "saturday_p2_hour": 12,
        "saturday_p2_minute": 0,
        "saturday_p2_temperature": 23,
        "saturday_p3_hour": 14,
        "saturday_p3_minute": 30,
        "saturday_p3_temperature": 22,
        "saturday_p4_hour": 17,
        "saturday_p4_minute": 30,
        "saturday_p4_temperature": 21,
        "sunday_p1_hour": 6,
        "sunday_p1_minute": 0,
        "sunday_p1_temperature": 19,
        "sunday_p2_hour": 12,
        "sunday_p2_minute": 30,
        "sunday_p2_temperature": 20,
        "sunday_p3_hour": 14,
        "sunday_p3_minute": 30,
        "sunday_p3_temperature": 21,
        "sunday_p4_hour": 21,
        "sunday_p4_minute": 30,
        "sunday_p4_temperature": 20,
        "weekdays_p1_hour": 6,
        "weekdays_p1_minute": 0,
        "weekdays_p1_temperature": 20,
        "weekdays_p2_hour": 11,
        "weekdays_p2_minute": 30,
        "weekdays_p2_temperature": 21,
        "weekdays_p3_hour": 13,
        "weekdays_p3_minute": 30,
        "weekdays_p3_temperature": 22,
        "weekdays_p4_hour": 17,
        "weekdays_p4_minute": 30,
        "weekdays_p4_temperature": 23
    },
    "running_state": "idle",
    "sensor": "AL",
    "system_mode": "heat"
}`

### What did you expect to happen?

It should change that value from null to numeric.

### Zigbee2MQTT version

1.39.1

### Adapter firmware version

20200921

### Adapter

zStack3x0

### Setup

Docker

ExquisiteMetropolis, 2024-10-19:
Same issue here:

Zigbee Model: TS0601
Zigbee Manufacturer: _TZE204_aoclfnxz
Description: Smart heating thermostat
Support status: Supported
Manufacturer : Moes
Model: [BHT-002/BHT-006](https://www.zigbee2mqtt.io/devices/BHT-002_BHT-006.html#moes-bht-002%252Fbht-006)

Can't be changed in Z2M or HA. It always reverts to an empty value in the Max temperature limit field.

Z2M version: 1.40.2
Coordinator Firmware: 7.4.1. [GA] - EmberZNet
Adapter type: usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2

pssc, 2024-10-24:
Hi, looks like stuff as happened https://github.com/Koenkk/zigbee-herdsman-converters/commit/1fa471d13f4cc9c64750471911606424c3121d4b so maybe the next release will fix

ExquisiteMetropolis, 2024-11-01:
Today's update to 1.41.0 has resolved this issue for me.
https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.41.0

closed 2025-01-09 by github-actions[bot]