Koenkk/zigbee2mqtt #23621: zigbee2mqtt stopps on device power up

mtomzig, 2024-08-15:
### What happened?

When I switch on a lamp (no matter which) connected to the coordinator (connect it to the power supply), the software crashes because the usb dongle is not available anymore.

The software is set up as a service and is supposed to restart on its own - but that fails. The software can only be restarted if the USB dongle is disconnected from the Raspberry for a few seconds.

In the logfile (see attached file) i see a line with the following content:
`error:    zh:zstack:znp: Failed to determine if path is valid: 'Error: ENOENT: no such file or directory, lstat '/dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0018ED16DA-if00''`

### What did you expect to happen?

The software (or the USB dongle) should not crash when a lamp is connected to the power supply.

### How to reproduce it (minimal and precise)

- Turn off the lamp (power switch)
- Turn on the lamp (power switch)
- Software/Dongle crashed

### Zigbee2MQTT version

1.39.1

### Adapter firmware version

20211115

### Adapter

Texas_Instruments_TI_CC2531_USB_CDC

### Setup

RaspberryPI via Git clone

### Debug log

[log.log](https://github.com/user-attachments/files/16622520/log.log)

mtomzig, 2024-08-18:
I have now reached a state where the USB stick does not crash, when a paired device is switched on using the power switch. I have performed the following steps:

- I had 26 devices connected to the USB stick. Since I read that problems can arise with 20 devices or more, I removed a few. I then only had 19 paired.
_Result: The problem still existed._
- Then I found the adapter settings (Settings -> Serial -> Adapter). There I selected "zstack" to test it. I was able to start Zigbee2MQTT, but it crashed when I tried to control a lamp. But that didn't solve the actual problem.
_Result: The problem still existed._
- After I reset the adapter setting to "auto" (via the configuration.yaml, since the web interface was not accessible), I was able to start the service again, but the problems from the previous step still existed.
_Result: The problem still existed._
- Drastic measures: I reinstalled the firmware on the USB stick (following these instructions: https://www.zigbee2mqtt.io/guide/adapters/flashing/alternative_flashing_methods.html#arduino_ccloader). At first it looked as if this would have no effect. After another restart, Zigbee2MQTT now runs stably, even when a lamp is switched on via the power switch:
_Result: The original problem is **solved**._

What I am wondering now is: Could it be that the pairing of more than 20 devices has led to this problem? Or is there possibly another reason for this?

I would like to reconnect the seven lamps that are now not paired, but I would like to avoid running into the original problem again.

Can anyone give me their opinion on this?

Thanks and greetings.

Koenkk, 2024-08-19:
The CC2531 is known to crash and is not recommended. Consider updating to one of the recommended adapters (https://www.zigbee2mqtt.io/guide/adapters/)

mtomzig, 2024-08-20:
Thank you. I already thought so.

I would say that the SONOFF USB stick is quite good.

Koenkk, 2024-08-20:
Both are good, P has been supported for longer, but E also works great nowadays, doesnt really matter anymore

mtomzig, 2024-09-23:
Hello again!

I have, hopefully, one last question. I have now bought a "SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P".

Does it have the correct firmware out of the box or do I need to flash it? I only find the reference to a firmware to be flashed on the E-Dongle.

Greetings
Mark

closed 2024-12-22 by github-actions[bot]