Koenkk/zigbee2mqtt #23623: Third Reality Garage Tilt Sensor (3RDTS01056Z) , Errors when starting Zigbee2MQTT

TomKassel, 2024-08-15:
### What happened?

The sensor can be paired perfectly. However, if you restart ZigBee2MQTT, 3 attempts with error messages are issued.

[image]

### What did you expect to happen?

No errors.

### How to reproduce it (minimal and precise)

Every start of Zigbee2MQTT.

### Zigbee2MQTT version

1.139.1

### Adapter firmware version

26780700

### Adapter

ConBee2

### Setup

Add.on Homeseer Software, Intel X86, Windows 11

TomKassel, 2024-11-01:
Same errors in 1.41.0

daubman, 2024-11-01:
Here's the full error I'm getting for this sensor - also in the last two versions (was hoping this update would fix)
```
[2024-11-01 06:07:53] error: 	z2m: Failed to configure 'garage-door-3rd-1', attempt 1 (Error: ZCL command 0x282c02bfffec9720/1 genPowerCfg.configReport([{"minimumReportInterval":3600,"maximumReportInterval":65000,"reportableChange":10,"attribute":"batteryPercentageRemaining"}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:350:28)
    at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:956:26)
    at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:688:9)
    at setupAttributes (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:162:17)
    at configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:211:17)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:205:21)
    at Configure.configure (/app/lib/extension/configure.ts:130:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:206:17))```

dwaned, 2024-12-02:
Same error on 1.42.0-1
`
z2m: Failed to configure 'Garage door sensor', attempt 1 (Error: ZCL command 0x282c02bfffe65cea/1 genPowerCfg.configReport([{"minimumReportInterval":3600,"maximumReportInterval":65000,"reportableChange":10,"attribute":"batteryPercentageRemaining"}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:350:28) at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:956:26) at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:688:9) at setupAttributes (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:162:17) at configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:211:17) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:206:21) at Configure.configure (/app/lib/extension/configure.ts:130:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:211:17))
`

sumsh, 2024-12-16:
Same here, on 1.42.0

`z2m: Failed to configure 'West garage door tilt', attempt 2 (Error: ZCL command 0x282c02bfffee6107/1 genPowerCfg.configReport([{"minimumReportInterval":3600,"maximumReportInterval":65000,"reportableChange":10,"attribute":"batteryPercentageRemaining"}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:350:28) at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:956:26) at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:688:9) at setupAttributes (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:162:17) at configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:211:17) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:206:21) at Configure.configure (/app/lib/extension/configure.ts:130:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:211:17))`

closed 2025-02-22 by github-actions[bot]

micahp2, 2025-03-23:
I also have this issue. This sensor, back in 2024, did pair successfully a couple of times. Now, in 2025, it will not. I get these messages in the log:

info 2025-03-23 13:22:31z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/remove', payload '{"data":{"block":false,"force":true,"id":"0x282c02bfffeedb60"},"status":"ok","transaction":"k2rh2-11"}'
info 2025-03-23 13:23:19z2m: Accepting joining not in blocklist device '0x282c02bfffeedb60'
info 2025-03-23 13:23:19zh:controller: Interview for '0x282c02bfffeedb60' started
info 2025-03-23 13:23:19z2m: Device '0x282c02bfffeedb60' joined
info 2025-03-23 13:23:19z2m: Starting interview of '0x282c02bfffeedb60'
info 2025-03-23 13:23:19z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x282c02bfffeedb60","ieee_address":"0x282c02bfffeedb60"},"type":"device_joined"}'
info 2025-03-23 13:23:19z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x282c02bfffeedb60","ieee_address":"0x282c02bfffeedb60","status":"started"},"type":"device_interview"}'
error 2025-03-23 13:24:19zh:controller: Interview failed for '0x282c02bfffeedb60 with error 'Error: Interview failed because can not get node descriptor ('0x282c02bfffeedb60')'
error 2025-03-23 13:24:19z2m: Failed to interview '0x282c02bfffeedb60', device has not successfully been paired
info 2025-03-23 13:24:19z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x282c02bfffeedb60","ieee_address":"0x282c02bfffeedb60","status":"failed"},"type":"device_interview"}'