Koenkk/zigbee2mqtt #23625: [New device support]: Dual USB switch - _TZ3000_h1ipgkwn

ViscountHermit, 2024-08-15:
### Link

https://a.aliexpress.com/_EGJ9RZB

### Database entry

{"id":23,"type":"Router","ieeeAddr":"0xa4c13834decc60b7","nwkAddr":18283,"manufId":4417,"manufName":"_TZ3000_h1ipgkwn","powerSource":"Mains (single phase)","modelId":"TS0002","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65506":56,"65508":0,"65534":0,"modelId":"TS0002","manufacturerName":"_TZ3000_h1ipgkwn","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}},"manuSpecificBosch":{"attributes":{"53251":"AAAAAgAA"}},"genOnOff":{"attributes":{"onOff":1,"onTime":0,"offWaitTime":0,"moesStartUpOnOff":2}},"manuSpecificTuya_3":{"attributes":{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"switchType":2}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x003c84fffec777e3","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[4,5,6],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":1,"onTime":0,"offWaitTime":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x003c84fffec777e3","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1723735699246} 

### Comments

This is a new 2 port usb switch. Not sure if it has power monitoring but it should allow each port to be managed individually. 

### External definition

```shell
const {deviceEndpoints, identify, onOff, electricityMeter} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS0002'],
    model: 'TS0002',
    vendor: '_TZ3000_h1ipgkwn',
    description: 'Automatically generated definition',
    extend: [deviceEndpoints({"endpoints":{"1":1,"2":2}}), identify(), onOff({"powerOnBehavior":false,"endpointNames":["1","2"]}), electricityMeter()],
    meta: {"multiEndpoint":true},
};

module.exports = definition;
```

nmaster2042, 2024-08-17:
There is 3 variants: with one, two or three usb switch.
Is the support for these 3 variants ?

Bjk8kds, 2024-10-02:
I was planning on buying this device aswell, but after seeing this issue I will probably hold off on buying it for now until it is supported, at least until a converter is available. Unfortunately I can't help out since I am a novice.

ov1d1u, 2024-10-02:
The device works with Zigbee2MQTT, it's just detected as a wall switch.

frysztak, 2024-11-14:
I have two of those and both exhibit the same problem: once turned on (while load is connected), they immediately turn off. If I disconnect the load, turn it on and then connect the load - it stays on. I wonder if the device is faulty. Does anyone have the same problem?

Pr20100, 2024-11-15:
I've one with 2 usb ports, running great since few weeks with Z2M. Maybee have to take a look at the config of your device under Z2M? (Device/expose tab)
One minor issue for me, I can't set up the "Power on behavior" : 

> z2m: Publish 'set' 'power_on_behavior' to 'USB plug' failed: 'Error: ZCL command 0xa4c1385b08a45614/2 manuSpecificTuya_3.write({"powerOnBehavior":2}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'

closed 2025-01-23 by github-actions[bot]

de-andrei, 2025-12-15:
> У меня устройство с двумя USB-портами, отлично работает уже несколько недель с Z2M. Возможно, вам стоит посмотреть конфигурацию вашего устройства в Z2M? (Вкладка «Устройство/Отображать») У меня есть одна небольшая проблема: я не могу настроить «Поведение при включении питания»:
> 
> > z2m: Не удалось опубликовать 'set' 'power_on_behavior' в 'USB plug': 'Ошибка: Команда ZCL 0xa4c1385b08a45614/2 manuSpecificTuya_3.write({"powerOnBehavior":2}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) не удалась (Статус 'UNSUPPORTED_ATTRIBUTE')'

Did you decide this problem?