Koenkk/zigbee2mqtt #23630: [New device support]: UNSH Tuya ZigBee Smart Socket Switch

brvhprince, 2024-08-15:
### Link

https://www.aliexpress.com/item/1005006991694847.html

### Database entry

{"id":5,"type":"Router","ieeeAddr":"0xa4c1384f8cefa442","nwkAddr":37762,"manufId":4417,"manufName":"_TZE204_mvtclclq","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"!\u0007\u0000\u0000e\"\u0007\u0000\u0000\u0012�\u0007\u0000\u0000e�\u0007\u0000\u0000\u0012�\u0002\u0000\u0000\u0012�\u0003\u0000\u0000e�\u0003\u0000\u0000\u0012I\u0004\u0000\u0000e","65506":56,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","appVersion":74,"manufacturerName":"_TZE204_mvtclclq","zclVersion":3,"modelId":"TS0601","powerSource":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1723754422454}

### Comments

I followed the support [here](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html) then [here](https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html#instructions) then I got stuck at finding the data points and their meaning (I have no tuya gateway).

 **Data from logs that maybe useful**
 
```sh
[2024-08-15 20:37:19] debug: 	z2m: Received Zigbee message from 'Wall Socket', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,49],"type":"Buffer"},"datatype":2,"dp":21}],"seq":47616}' from endpoint 1 with groupID 0
[2024-08-15 20:37:19] debug: 	zhc:tuya: Datapoint 21 not defined for '_TZE204_mvtclclq' with value 49

[2024-08-15 20:37:19] debug: 	z2m: Received Zigbee message from 'Wall Socket', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,9,78],"type":"Buffer"},"datatype":2,"dp":23}],"seq":47872}' from endpoint 1 with groupID 0
[2024-08-15 20:37:19] debug: 	zhc:tuya: Datapoint 23 not defined for '_TZE204_mvtclclq' with value 2382

[2024-08-15 20:37:19] debug: 	z2m: Received Zigbee message from 'Wall Socket', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,64],"type":"Buffer"},"datatype":2,"dp":22}],"seq":48128}' from endpoint 1 with groupID 0
[2024-08-15 20:37:19] debug: 	zhc:tuya: Datapoint 22 not defined for '_TZE204_mvtclclq' with value 64

```
 

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_mvtclclq',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

Battosuka, 2024-09-01:
up, I couldn't either

brvhprince, 2024-09-24:
I got a Tuya Gateway today and I got it to work. A couple of clean ups to do, but for now it works.

Working converter:
```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
  // Since a lot of Tuya devices use the same modelID, but use different datapoints
  // it's necessary to provide a fingerprint instead of a zigbeeModel
  fingerprint: [
    {
      // The model ID from: Device with modelID 'TS0601' is not supported
      // You may need to add \u0000 at the end of the name in some cases
      modelID: 'TS0601',
      // The manufacturer name from: Device with modelID 'TS0601' is not supported.
      manufacturerName: '_TZE204_mvtclclq',
    },
  ],
  model: 'DS-1450WN',
  vendor: 'Tuya',
  description: 'Smart Zigbee Switch with Power Monitoring',
  fromZigbee: [tuya.fz.datapoints],
  toZigbee: [tuya.tz.datapoints],
  configure: tuya.configureMagicPacket,
  exposes: [
    e.switch_()
      .withState('switch_1', true, 'Control the USB A on/off state', ea.STATE_SET)
      .withLabel('USB A'),
    e.switch_()
      .withState('switch_2', true, 'Control the USB C on/off state', ea.STATE_SET)
      .withLabel('USB C'),
    e.switch_()
      .withState('switch_3', true, 'Control the Plug 1 on/off state', ea.STATE_SET)
      .withLabel('Plug 1'),
    e.switch_()
      .withState('switch_4', true, 'Control the Plug 2 on/off state', ea.STATE_SET)
      .withLabel('Plug 2'),
    e.numeric('countdown_1', ea.STATE_SET)
      .withLabel('USB A Countdown')
      .withValueMin(0)
      .withValueMax(86400)
      .withValueStep(1)
      .withDescription('Countdown timer for USB A')
      .withUnit('s'),
    e.numeric('countdown_2', ea.STATE_SET)
      .withLabel('USB C Countdown')
      .withValueMin(0)
      .withValueMax(86400)
      .withValueStep(1)
      .withDescription('Countdown timer for USB C')
      .withUnit('s'),
    e.numeric('countdown_3', ea.STATE_SET)
      .withLabel('Plug 1 Countdown')
      .withValueMin(0)
      .withValueMax(86400)
      .withValueStep(1)
      .withDescription('Countdown timer for Plug 1')
      .withUnit('s'),
    e.numeric('countdown_4', ea.STATE_SET)
      .withLabel('Plug 2 Countdown')
      .withValueMin(0)
      .withValueMax(86400)
      .withValueStep(1)
[... 21 lines cut ...]
      [7, "countdown_1", tuya.valueConverter.countdown],
      [8, "countdown_2", tuya.valueConverter.countdown],
      [9, "countdown_3", tuya.valueConverter.countdown],
      [10, "countdown_4", tuya.valueConverter.countdown],
      [14, "relay_status", tuya.valueConverterBasic.lookup({"memory": tuya.enum(0), "on": tuya.enum(1), "off": tuya.enum(2)})],
      [16, "switch_backlight", tuya.valueConverter.onOff],
      [21, "cur_current", tuya.valueConverter.divideBy10],
      [22, "cur_power", tuya.valueConverter.divideBy10],
      [23, "cur_voltage", tuya.valueConverter.divideBy10],
      [105, "total_battery", tuya.valueConverter.divideBy10],
      [106, "child_lock", tuya.valueConverter.lockUnlock]

    ]
  },
  extend: [
    // A preferred new way of extending functionality.
  ],
  icon: "data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAAoAAAAKACAYAAAAMzckjAAAACXBIWXMAAC4jAAAuIwF4pT92AAGSI0lEQVR4nOz9d5Al133nC37Oyczry9uu9g7ohgcIkAQIULQQSZGSOCKp4YxGTzMjzczO243dNxuxG7F/TWy8v3Yj3s7Oe29mNU4SZUd0ohNFT5ESDQxBAAQaQKO9re7qsrfquszz2z8y87rKa8o00I06n4juujdv5smT/ps/d5SIYLFYLBaLxWLZOeg3uwMWi8VisVgsljcWKwAtFovFYrFYdhhWAFosFovFYrHsMKwAtFgsFovFYtlhWAFosVgsFovFssOwAtBisVgsFotlh2EFoMVisVgsFssOwwpAi8VisVgslh2GFYAWi8VisVgsOwwrAC0Wi8VisVh2GFYAWiwWi8VisewwrAC0WCwWi8Vi2WFYAWixWCwWi8Wyw7AC0GKxWCwWi2WHYQWgxWKxWCwWyw7DCkCLxWKxWCyWHYYVgBaLxWKxWCw7DCsALRaLxWKxWHYYVgBaLBaLxWKx7DCsALRYLBaLxWLZYVgBaLFYLBaLxbLDsALQYrFYLBaLZYdhBaDFYrFYLBbLDsMKQIvFYrFYLJYdhhWAFovFYrFYLDsMKwAtFovFYrFYdhhWAFosFovFYrHsMKwAtFgsFovFYtlhWAFosVgsFovFssOwAtBisVgsFotlh2EFoMVisVgsFssOwwpAi8VisVgslh2GFYAWi8VisVgsOwwrAC0Wi8VisVh2GFYAWiwWi8VisewwrAC0WCwWi8Vi2WFYAWixWCwWi8Wyw7AC0GKxWCwWi2WHYQWgxWKxWCwWyw7DCkCLxWKxWCyWHYYVgBaLxWKxWCw7DCsALRaLxWKxWHYYVgBaLBaLxWKx7DCsALRYLBaLxWLZYVgBaLFYLBaLxbLDsALQYrFYLBaLZYdhBaDFYrFYLBbLDsMKQIvFYrFYLJYdhhWAFovFYrFYLDsMKwAtFovFYrFYdhhWAFosFovFYrHsMKwAtFgsFovFYtlhWAFosVgsFovFssOwAtBisVgsFotlh2EFoMVisVgsFssOwwpAi8VisVgslh2GFYAWi8VisVgsOwwrAC0Wi8VisVh2GFYAWiwWi8VisewwrAC0WCwWi8Vi2WFYAWixWCwWi8Wyw7AC0GKxWCwWi2WHYQWgxWKxWCwWyw7DCkCLxWKxWCyWHYYVgBaLxWKxWCw7DCsALRaLxWKxWHYYVgBaLBaLxWKx7DCsALRYLBaLxWLZYVgBaLFYLBaLxbLDsALQYrFYLBaLZYdhBaDFYrFYLBbLDsMKQIvFYrFYLJYdhhWAFovFYrFYLDsMKwAtFovFYrFYdhhWAFosFovFYrHsMKwAtFgsFovFYtlhWAFosVgsFovFssOwAtBisVgsFotlh2EFoMVisVgsFssOwwpAi8VisVgslh2GFYAWi8VisVgsOwwrAC0Wi8VisVh2GFYAWiwWi8VisewwrAC0WCwWi8Vi2WFYAWixWCwWi8Wyw7AC0GKxWCwWi2WHYQWgxWKxWCwWyw7DCkCLxWKxWCyWHYYVgBaLxWKxWCw7DCsALRaLxWKxWHYYVgBaLBaLxWKx7DCsALRYLBaLxWLZYVgBaLFYLBaLxbLDsALQYrFYLBaLZYdhBaDFYrFYLBbLDsMKQIvFYrFYLJYdhhWAFovFYrFYLDsMKwAtFovFYrFYdhhWAFosFovFYrHsMKwAtFgsFovFYtlhWAFosVgsFovFssOwAtBisVgsFotlh2EFoMVisVgsFssOwwpAi8VisVgslh2GFYAWi8VisVgsOwwrAC0Wi8VisVh2GFYAWiwWi8VisewwrAC0WCwWi8Vi2WFYAWixWCwWi8Wyw7AC0GKxWCwWi2WHYQWgxWKxWCwWyw7DCkCLxWKxWCyWHYYVgBaLxWKxWCw7DCsALRaLxWKxWHYYVgBaLBaLxWKx7DCsAL [... cut ...] BzBglAgiAIgiCIOYMEIEEQBEEQxJxBApAgCIIgCGLOIAFIEARBEAQxZ5AAJAiCIAiCmDNIABIEQRAEQcwZJAAJgiAIgiDmDBKABEEQBEEQcwYJQIIgCIIgiDmDBCBBEARBEMScQQKQIAiCIAhiziABSBAEQRAEMWeQACQIgiAIgpgzSAASBEEQBEHMGSQACYIgCIIg5gwSgARBEARBEHMGCUCCIAiCIIg5gwQgQRAEQRDEnEECkCAIgiAIYs4gAUgQBEEQBDFnkAAkCIIgCIKYM0gAEgRBEARBzBkkAAmCIAiCIOYMEoAEQRAEQRBzBglAgiAIgiCIOYMEIEEQBEEQxJxBApAgCIIgCGLOIAFIEARBEAQxZ5AAJAiCIAiCmDNIABIEQRAEQcwZJAAJgiAIgiDmjP8f/GCwo52bJS0AAAAASUVORK5CYII="
};

module.exports = definition;
```

Screenshot
[image]

closed 2024-12-22 by github-actions[bot]