Koenkk/zigbee2mqtt #23635: SLZB-06 does not work with version 1.39.1

eerison, 2024-08-16:
### What happened?

after update to version 1.39.1 the z2m does not start

### What did you expect to happen?

it starts

### How to reproduce it (minimal and precise)

update z2m image from 1.39.0 to 1.39.1 and it gonna see the message bellow

Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory

### Zigbee2MQTT version

1.39.1

### Adapter firmware version

20240527

### Adapter

slzb-06

### Setup

docker container

### Debug log

I just can see this error 

Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory
Using '/app/data' as data directory

even if I set log_level: debug

eerison, 2024-08-16:
and on version 1.39.0 it starts

```
Using '/app/data' as data directory
Starting Zigbee2MQTT without watchdog.
[2024-08-16 08:30:26] info: 	z2m: Logging to console, file (filename: log.log)
[2024-08-16 08:30:26] info: 	z2m: Starting Zigbee2MQTT version 1.39.0 (commit #0326926)
[2024-08-16 08:30:26] info: 	z2m: Starting zigbee-herdsman (0.50.1)
[2024-08-16 08:30:27] info: 	zh:zstack:znp: Opening TCP socket with 
[2024-08-16 08:30:27] info: 	zh:zstack:znp: Socket connected
[2024-08-16 08:30:27] info: 	zh:zstack:znp: Socket ready
[2024-08-16 08:30:27] info: 	zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2024-08-16 08:30:28] info: 	zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2024-08-16 08:31:09] info: 	z2m: zigbee-herdsman started (restored)
```

Snake40, 2024-08-16:
Bonsoir.
I'm trying (in vain) to install zigbbee2mqtt in a Docker container with the SLZB-06 POE dongle. Beyond the difficulty you seem to be encountering at the moment, could you give me the docker compose (or equivalent) that you used to install your container as well as the zigbee2mqtt "configuration.yaml" file, specifying where of your container file structure did you put it?...
Thank you in advance for your help...

Koenkk, 2024-08-17:
@eerison can you try changing your `configuration.yaml` to [this one](https://github.com/Koenkk/zigbee2mqtt/blob/master/data/configuration.example.yaml) and see if tries to start now? (of course it will fail because the port is wrong, but I want to see if it gets further). Also, with what config do you run the Docker container?

Snake40, 2024-08-17:
Bonjour.
Thank you for your assistance.

I'm working with a Synology DS220+ NAS that already has MQTT installed in a container. I want to do some testing by creating 2 containers (HA and zigbee2mqtt) and using the SLZB-06 POE dongle.
Given your initial post on which I reacted, I am using version 1.39.0 of the z2m image to avoid complicating things...

Here is my docker compose to create the homeassistant and zigbee2mqtt containers:

version: '3.8'
services:
    #Conteneur Home Assistant
    homeassistant:
        container_name: homeassistant
        image: homeassistant/home-assistant:latest
        restart: always
        #On définit un "bind mount" pour conserver la configuration de Home-Assistant
        volumes:
            - /volume1/docker/homeassistant:/config
        ports:
        #Le port sur lequel sera accessible l'interface web PORT_EXPORTE:PORT_CONTENEUR
            - 8123:8123
        environment:
            - TZ=Europe/Paris

    #Conteneur ZigBee2MQTT
    zigbee2mqtt:
        container_name: zigbee2mqtt
        image: koenkk/zigbee2mqtt
        restart: unless-stopped
        volumes:
            - /volume1/docker/zigbee2mqtt/data:/app/data
            - /run/udev:/run/udev:ro         
        ports:
            #Port Zigbee2MQTT
            - 8085:8085
        environment:
            - TZ=Europe/Paris

Here is the "configuration.yaml" file which is located in the "/volume1/docker/zigbee2mqtt/data" directory:

homeassistant: false
permit_join: false
frontend:
    port: 8085
mqtt:
    base_topic: zigbee2mqtt
    server: mqtt://192.168.1.100:1883
    user: xxx
    password: xxx
serial:
# Location of SLZB-06
    port: tcp://192.168.1.43:6638
    baudrate: 115200
    adapter: zstack
# Disable green led?
    disable_led: false
# Set output power to max 20
# Advanced settings
advanced:
    transmit_power: 20
    # Let Zigbee2MQTT generate a network key on first start
    network_key: GENERATE
    # Let Zigbee2MQTT generate a pan_id on first start
    pan_id: GENERATE
    # Let Zigbee2MQTT generate a ext_pan_id on first start
    ext_pan_id: GENERATE

In this file, everything related to the SLZB-06 is given to me by the web app associated with the dongle.

Whether I start creating the containers with my "configuration.yaml" file or with the example file that you gave me, the result is the same. Here is the log file associated with the zigbee2mqtt container:

2024/08/17 10:24:46	stderr	    at start (/app/index.js:130:14)
2024/08/17 10:24:46	stderr	    at Object.reRead (/app/lib/util/settings.ts:749:5)
2024/08/17 10:24:46	stderr	    at getInternalSettings (/app/lib/util/settings.ts:466:21)
2024/08/17 10:24:46	stderr	    at read (/app/lib/util/settings.ts:380:27)
2024/08/17 10:24:46	stderr	    at interpretValue (/app/lib/util/settings.ts:369:25)
2024/08/17 10:24:46	stderr	    at Object.read (/app/lib/util/yaml.ts:7:37)
2024/08/17 10:24:46	stderr	    at Object.readFileSync (node:fs:464:35)
2024/08/17 10:24:46	stderr	    at Object.openSync (node:fs:596:3)
2024/08/17 10:24:46	stderr	Error: ENOENT: no such file or directory, open '/app/data/... 3.yaml'
2024/08/17 10:24:45	stdout	Starting Zigbee2MQTT without watchdog.
2024/08/17 10:24:45	stdout	Using '/app/data' as data directory

My current understanding of Docker and z2m and my searches on the web do not allow me to understand what is happening. Currently I think my volume declarations when creating the zigbee2mqtt container are incorrect and not allowing the system to find the "configuration.yaml" file.

Have a nice day ...

Snake40, 2024-08-17:
Sorry...I just realized I was responding to a message that wasn't intended for me.

eerison, 2024-08-17:
> @eerison can you try changing your `configuration.yaml` to [this one](https://github.com/Koenkk/zigbee2mqtt/blob/master/data/configuration.example.yaml) and see if tries to start now? (of course it [...]

I changed the config file for the example that you sent, But it didn´t work too 🤔

eerison, 2024-08-17:
@Snake40 

it is my docker file

```yaml
version: '3.8'
services:
  zigbee2mqtt:
    container_name: zigbee2mqtt
    image: koenkk/zigbee2mqtt:1.39.1
    restart: unless-stopped
    deploy:
      resources:
        limits:
          cpus: '1'
          memory: 1G
        reservations:
          cpus: '0.5'
          memory: 500M 
    volumes:
      - my/local/path/data:/app/data
      - /run/udev:/run/udev:ro
    labels:
      - "traefik.enable=true"
      - "traefik.http.routers.z2m.rule=Host(`my.domain.com`)"
      - "traefik.http.routers.z2m.entrypoints=websecure"
      - "traefik.http.services.z2m.loadbalancer.server.port=8080"
      - "traefik.http.routers.z2m.tls.certresolver=cloudflare"
    environment:
      - TZ=Your/TimeZone
    devices:
      # Make sure this matched your adapter location
      - /dev/tty0:/dev/ttyACM0
    networks:
      - proxy

networks:
    proxy:
      driver: bridge
      external: true
```

Snake40, 2024-08-17:
@eerison
I'm perplexed... Apart from the point I raise below, my volume declarations are similar to yours. So it could be that I'm looking for my mistake in the wrong direction.

You include the following lines in your docker:

    devices:
      - /dev/tty0:/dev/ttyACM0
      - 
I thought I understood that these lines corresponded to a USB device. Would you use your SLZB-06 via USB and not Ethernet?

Can you confirm to me that your z2m configuration file "configuration.yaml" is indeed in the "my/local/path/data" directory (by reference to the content of the file you sent me)?
Would it be possible to see the contents of this file (without personal things and other passwords)?

Thank you.

eerison, 2024-08-17:
> @eerison I'm perplexed... Apart from the point I raise below, my volume declarations are similar to yours. So it could be that I'm looking for my mistake in the wrong direction.
>
> You include the [...]

I can't send it now, but there is anything special,  I connect like you, via socket on the network.

I added this device settings, because the container wasn't starting without this map,  then I added this,  just to have z2m running.

Koenkk, 2024-08-18:
What if you execute the following: `docker run -it --rm koenkk/zigbee2mqtt`, does it start (with an error?)

Arn0uDz, 2024-08-18:
Running 1.39.1 with SLZB-06 and no issues. Running 20230507

eerison, 2024-08-18:
> Running 1.39.1 with SLZB-06 and no issues. Running 20230507

Hey @Arn0uDz 

I guess the error isn't related with slzb-06, for some reason z2m isn't starting using docker 🥲

eerison, 2024-08-18:
I tested in other server (x86_64) and it started

I guess it is related with arm64 architecture 🥲

Output (server running x86_64)

```
docker run -it --rm koenkk/zigbee2mqtt
Using '/app/data' as data directory
Creating configuration file...
Starting Zigbee2MQTT without watchdog.
[2024-08-18 15:58:35] info:     z2m: Logging to console, file (filename: log.log)
[2024-08-18 15:58:35] info:     z2m: Starting Zigbee2MQTT version 1.39.1 (commit #e132316)
[2024-08-18 15:58:35] info:     z2m: Starting zigbee-herdsman (0.55.3)
[2024-08-18 15:58:35] error:    zh:zstack:znp: Failed to determine if path is valid: 'Error: ENOENT: no such file or directory, lstat '/dev/ttyACM0''
[2024-08-18 15:58:36] info:     zh:zstack:znp: Opening SerialPort with {"path":"/dev/ttyACM0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2024-08-18 15:58:36] error:    z2m: Error while starting zigbee-herdsman
[2024-08-18 15:58:36] error:    z2m: Failed to start zigbee
[2024-08-18 15:58:36] error:    z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-08-18 15:58:36] error:    z2m: Exiting...
[2024-08-18 15:58:36] error:    z2m: Error: Error: No such file or directory, cannot open /dev/ttyACM0
```

eerison, 2024-08-18:
Ok good news the problem was related with the docker image, I removed the image and downloaded again, and it started ❤️ 

now I am facing the issue bellow 

```
[2024-08-18 13:19:44] error: 	z2m: Error while starting zigbee-herdsman
[2024-08-18 13:19:44] error: 	z2m: Failed to start zigbee
[2024-08-18 13:19:44] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-08-18 13:19:44] error: 	z2m: Exiting...
[2024-08-18 13:19:45] error: 	z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:119:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:127:29)
    at Zigbee.start (/app/lib/zigbee.ts:63:27)
    at Controller.start (/app/lib/controller.ts:139:27)
    at start (/app/index.js:154:5
```

eerison, 2024-08-18:
Ok it worked :)

I removed config and coordinator file, and let the z2m generate again, copied mqqt and serial config from my old config and it is working without any issue.

thank you @Koenkk for your support

Note: as I removed the coordinator fine I lost my devices and I need to put again in z2m 😄

closed 2024-08-18 by eerison