Koenkk/zigbee2mqtt #23638: LLKZMK12LM "decoupled" mode wrong

BeFygo, 2024-08-16:
### What happened?

Unable to decouple relay 1 independent from relay 2.
When pressing "decoupled" on one of the relay the other changes to "decoupled" also.
idem for "control_relay" mode.

Sometimes playing with those buttons it shows one relay decoupled and the other control_relay but in the log both are decoupled or control_relay.
And on the hardware only relay 1 gets decoupled and its impossible  to decouple relay 2.

### What did you expect to happen?

decoupling relay 1 and 2 independent from each other

### How to reproduce it (minimal and precise)

zigbee2mqtt V1.39.1 the issue is PRESENT

Zigbee2mqtt V1.39.0 the issue is NOT present

### Zigbee2MQTT version

1.39.1

### Adapter firmware version

20240710

### Adapter

Sonoff ZBdongle-P CC2652P 

### Setup

Docker

HobboRobin, 2024-09-15:
Hello, @BeFygo 
I saw your problem while i was with one of my Z2M Systems still on 1.39.1 and had the same issue. The other Z2M i had already on 1.40.1 and there i had no problem with the same device. 
So i upgraded both on 1.40.1 and i had no problem anymoore.

So upgrading should be fix your problem :)

BeFygo, 2024-10-22:
upgraded and its indeed fixed

closed 2024-10-22 by BeFygo