Koenkk/zigbee2mqtt #23642: Beginner Sonoff Stick 3 in Raspberry pi

Middendorf, 2024-08-16:
### What happened?

Hello folks,

until a few days ago I was still with Conbee, but they stopped the service.

Now I wanted to try out a Sonoff USB 3 stick.

`ls` /dev/serial/by-id/`
usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_b8b49abd27a6ed11a280eba32981d5c7-if00-port0

it is also recognized.

sudo nano /opt/zigbee2mqtt/data/configuration.yaml
```
# home Assistant
homeassistant: false
# allow new devices to join
permit_join: true

# MQTT settings
mqtt:
  # MQTT base topic for Zigbee2MQTT MQTT messages
  base_topic: zigbee2mqtt
  # MQTT server URL
  server: 'mqtt://mqtt
  # MQTT server authentication, uncomment if required:
  # user: username
  # password: 123456
# Serial settings
serial:
  # Location of the adapter (see first step of this guide)
port: /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_b8b49abd27a6ed11a280eba32981d5c7-if00-port0

advanced:
  network_key: GENERATE
  ikea_ota_use_test_url: true
  legacy_api: false
  log_level: debug

frontend: true
frontend:
  port: 8020
  host: 0.0.0.0
  auth_token: 123456

``` 

--- this is what the start of npm looks like----

cd /opt/zigbee2mqtt                               
pi@Zuhause:/opt/zigbee2mqtt $ npm start

> zigbee2mqtt@1.39.1 start
> node index.js

Starting Zigbee2MQTT without watchdog.

/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:183
  return new YAMLException(message, mark);
         ^
YAMLException: unexpected end of the stream within a single quoted scalar (32:1)

 29 |   host: 0.0.0.0
 30 |   auth_token: 123456
 31 |   url: http://localhost
 32 | 
------^
    at generateError (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:183:10)
    at throwError (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:187:9)
    at readSingleQuotedScalar (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:624:3)
    at composeNode (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1446:13)
    at readBlockMapping (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1164:11)
    at composeNode (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1441:12)
    at readBlockMapping (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1164:11)
    at composeNode (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1441:12)
    at readDocument (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1625:3)
    at loadDocuments (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1688:5)
pi@Zuhause:/opt/zigbee2mqtt $ 

I followed this page, but only from the point where they talk about setting up Zigbee2mqtt configuration
https://nerdiy.de/de_de/howto-zigbee-installation-von-zigbee2mqtt-auf-dem-raspberry-pi/

Have I forgotten something or made a mistake?? Is there software on the stick?? Or do you have to flash it when you buy it?

I'm a beginner when it comes to this, if I have to flash it it's enough for me to know this, then I'll google it.

I would like to have the web interface, because I can certainly handle something like that better than with codes. I use Fhem as a home assistant on a Raspberry pi4

### Zigbee2MQTT version

1.22.1

### Adapter firmware version

.

### Adapter

sonoff zigbee 3.0 usb dongle plus

### Setup

Fhem, Raspberry Pi4

Koenkk, 2024-08-16:
```
frontend: true # <- remove this line
frontend:
  port: 8020
  host: 0.0.0.0
  auth_token: 123456
```

Middendorf, 2024-08-17:
Hello Koenkk,

I have removed it. Unfortunately I still can't get to the web interface. The page is not accessible.
Do you have any other ideas??

Unfortunately I can't get any further

Koenkk, 2024-08-17:
Does z2m start now? If yes, than the frontend should be accessible on `http://localhost:8020`

Middendorf, 2024-08-17:
Hi
according to
npm start it seems to start but I can't get to the web interface.

this is what the npm start looks like.
`:/opt/zigbee2mqtt $ npm start

> zigbee2mqtt@1.39.1 start
> node index.js

Starting Zigbee2MQTT without watchdog.

/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:183
  return new YAMLException(message, mark);
         ^
YAMLException: unexpected end of the stream within a single quoted scalar (32:1)

 29 |   auth_token: 123456
 30 |   url: http://localhost
 31 | 
 32 | 
------^
    at generateError (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:183:10)
    at throwError (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:187:9)
    at readSingleQuotedScalar (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:624:3)
    at composeNode (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1446:13)
    at readBlockMapping (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1164:11)
    at composeNode (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1441:12)
    at readBlockMapping (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1164:11)
    at composeNode (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1441:12)
    at readDocument (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1625:3)
    at loadDocuments (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1688:5)
`

I have already done
sudo systemctl enable zigbee2mqtt.service
but unfortunately it didn't work

`pi@Zuhause:~` $ sudo journalctl -f -u zigbee2mqtt
-- Journal begins at Mon 2024-05-27 15:02:05 CEST. --
Aug 17 08:48:40 Zuhause systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Aug 17 08:48:40 Zuhause systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Aug 17 08:48:40 Zuhause systemd[1]: zigbee2mqtt.service: Consumed 2.943s CPU time.
Aug 17 08:48:41 Zuhause systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 38.
Aug 17 08:48:41 Zuhause systemd[1]: Stopped zigbee2mqtt.
Aug 17 08:48:41 Zuhause systemd[1]: zigbee2mqtt.service: Consumed 2.943s CPU time.
Aug 17 08:48:41 Zuhause systemd[1]: Started zigbee2mqtt.
Aug 17 08:48:41 Zuhause npm[79179]: > zigbee2mqtt@1.39.1 start
Aug 17 08:48:41 Zuhause npm[79179]: > node index.js
Aug 17 08:48:42 Zuhause npm[79191]: Starting Zigbee2MQTT without watchdog.
Aug 17 08:48:43 Zuhause npm[79191]: /opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:183
Aug 17 08:48:43 Zuhause npm[79191]:   return new YAMLException(message, mark);
Aug 17 08:48:43 Zuhause npm[79191]:          ^
Aug 17 08:48:43 Zuhause npm[79191]: YAMLException: unexpected end of the stream within a single quoted scalar (32:1)
Aug 17 08:48:43 Zuhause npm[79191]:  29 |   auth_token: 123456
Aug 17 08:48:43 Zuhause npm[79191]:  30 |   url: http://localhost
Aug 17 08:48:43 Zuhause npm[79191]:  31 |
Aug 17 08:48:43 Zuhause npm[79191]:  32 |
Aug 17 08:48:43 Zuhause npm[79191]: ------^
Aug 17 08:48:43 Zuhause npm[79191]:     at generateError (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:183:10)
Aug 17 08:48:43 Zuhause npm[79191]:     at throwError (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:187:9)
Aug 17 08:48:43 Zuhause npm[79191]:     at readSingleQuotedScalar (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:624:3)
Aug 17 08:48:43 Zuhause npm[79191]:     at composeNode (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1446:13)
Aug 17 08:48:43 Zuhause npm[79191]:     at readBlockMapping (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1164:11)
Aug 17 08:48:43 Zuhause npm[79191]:     at composeNode (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1441:12)
Aug 17 08:48:43 Zuhause npm[79191]:     at readBlockMapping (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1164:11)
Aug 17 08:48:43 Zuhause npm[79191]:     at composeNode (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1441:12)
Aug 17 08:48:43 Zuhause npm[79191]:     at readDocument (/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1625:3)
Aug 17 08:48:43 Zuhause npm[79191]:     at loadDocuments `(/opt/zigbee2mqtt/node_modules/js-yaml/lib/loader.js:1688:5)`

maybe that will help with expenses

Koenkk, 2024-08-17:
Your yaml file is wrong, use e.g. https://www.yamllint.com/ to fix it (copy paste your configuration.yaml file in there)

Middendorf, 2024-08-17:
What does the end look like? I looked at the file that I found on the web and there is no end listed there either.

`Missing closing 'quote at line 31, column 1`

`# home Assistant
homeassistant: false
# allow new devices to join
permit_join: true

# MQTT settings
mqtt:
  # MQTT base topic for Zigbee2MQTT MQTT messages
  base_topic: zigbee2mqtt
  # MQTT server URL
  server: 'mqtt://mqtt
  # MQTT server authentication, uncomment if required:
  # user: Middendorf
  # password: 123456
# Serial settings
serial:
  # Location of the adapter (see first step of this guide)
port: /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_b8b49abd27a6ed11a280eba32981d5c7-if00-port0

advanced:
  network_key: GENERATE
  ikea_ota_use_test_url: true
  legacy_api: false
  log_level: debug

frontend:
  port: 8020
  host: 0.0.0.0
  auth_token: 123456
  url: http://localhost
`

Middendorf, 2024-08-17:
So now I've made it this far and even had the software look at the Yaml again several times. Now I have this error message and I can't find the error.

It runs as localhost for me, with 1883, so the USB stick is connected there. I've also tried with /ttyUSB1. Same error message.

The info page doesn't say what I'm doing wrong either.

Does anyone have an idea??

`---
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://localhorst:1883
serial:
port: /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_b8b49abd27a6ed11a280eba32981d5c7-if>
advanced:
  network_key: GENERATE
  ikea_ota_use_test_url: true
  legacy_api: false
  log_level: debug
frontend:
  port: 8038
  host: 0.0.0.0
  auth_token: GENERATE
  url: http://localhost
`
error message

`pi@Zuhause:/opt/zigbee2mqtt $ npm start

> zigbee2mqtt@1.39.1 start
> node index.js

Starting Zigbee2MQTT without watchdog.

!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
            READ THIS CAREFULLY

Refusing to start because configuration is not valid, found the following errors:
- serial must be object

If you don't know how to solve this, read https://www.zigbee2mqtt.io/guide/configuration

!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
`

chris-1243, 2024-08-19:
```

[2024-08-17 23:10:05] error: zh:adapter:zstack:manager: If you wish to re-commission your network, please remove coordinator backup at /opt/zigbee2mqtt/data/coordinator_backup.json.
[2024-08-17 23:10:05] error: zh:adapter:zstack:manager: Re-commissioning your network will require re-pairing of all devices!
```

I would suggest to read carefully your log😉

If you wish to start a new installation, rename`coordinator_backup.json` like `coordinator_backup.json_original`. Like this you have a backup of this file and Z2M will not read it. 

Restart Z2M and you should be fine. Keep in mind, you will have to re-pair your devices.

closed 2024-12-22 by github-actions[bot]