Koenkk/zigbee2mqtt #23647: [Wrong device]: Nous A1Z "Reset energy" chapter was removed

krusta4711, 2024-08-17:
### Link

https://www.zigbee2mqtt.io/devices/A1Z.html

### Model

A1Z

### Description

Reset energy

### Vendor

Nous

### Picture (link)

just the description might be wrong

### Database entry

just the description might be wrong

### Notes

The section "Reset energy" was removed from the descripton. Was this by purpose? At least for mine plugs the command still works.

krusta4711, 2024-08-17:
@Koenkk It looks like you did the change in the repository. Was it by purpose. For me resetting the energy command still works fine.

Koenkk, 2024-08-18:
Not on purpose, could you make a PR to add it back?

krusta4711, 2024-08-18:
@Koenkk 
I created a PR for reintroducing the remove chapter:
https://github.com/Koenkk/zigbee2mqtt.io/pull/2956

closed 2024-08-19 by Koenkk