Koenkk/zigbee2mqtt #23648: auth_token is ignored

ZTHawk, 2024-08-17:
### What happened?

I have zigbee2mqtt as a HA addon and setup "auth_token" in configuration.yaml (in config/zigbee2mqtt).
```
frontend:
  port: 8099
  auth_token: 123456

```
When connecting to the frontend via "http://192.168.0.123:8099" then there is no authentication request. I am logged in an can interact with all UI features.

### What did you expect to happen?

A popup to enter the password

### Zigbee2MQTT version

1.39.1

### Adapter firmware version

EmberZNet 7.4.2 [GA]

### Adapter

SONOFF Dongle Plus V2

### Setup

HA Addon and HA in proxmox

jvcsw, 2024-09-13:
same here with version 1.40.1

PacmanForever, 2024-09-27:
Hi @Koenkk.
This is a feature or an issue?

DBengs, 2024-11-21:
Same here. This is quite a serious bug considering it lets anyone on the local network access your zigbee devices, and I would guess probably some form of RCE attack as well.

Not even setting the auth_token through the frontend itself does anything.

PacmanForever, 2024-11-21:
I agree...

ZTHawk, 2025-01-21:
still applicable (v2.0.0)

ZTHawk, 2025-02-02:
still applicable (v2.1.0)
I think it is related to #23883

ZTHawk, 2025-04-04:
still applicable

ZTHawk, 2025-04-06:
Update v2.2.1 (maybe 2.2.0) fixed this issue.
Bug created a new one with ingress.

Site is displayed but not popup window to ask for the password is shown (for HA ingress I would not expect to enter anything).

ZTHawk, 2025-05-02:
The actuall issue is fixed. The new issue discrived is now here: #27297

closed 2025-05-02 by ZTHawk