Koenkk/zigbee2mqtt #23652: [New device support]: Philips Hue Twilight LGT002

toksvaeth, 2024-08-17:
### Link

https://www.philips-hue.com/en-gb/p/sleep-and-wake-up-light-twilight-sleep-and-wake-up-light-black/8720169263055

### Database entry

{"id":77,"type":"Router","ieeeAddr":"0x001788010de259c4","nwkAddr":25628,"manufId":4107,"manufName":"Signify Netherlands B.V.","powerSource":"Mains (single phase)","modelId":"LGT002","epList":[1,11,12,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":259,"inClusterList":[0,64512,3],"outClusterList":[6,3],"clusters":{"genBasic":{"attributes":{"modelId":"LGT002","manufacturerName":"Signify Netherlands B.V.","powerSource":1,"zclVersion":8,"appVersion":2,"stackVersion":1,"hwVersion":0,"dateCode":"20231211","swBuildId":"1.113.2"}},"genOnOff":{"attributes":{}},"lightingColorCtrl":{"attributes":{}},"genLevelCtrl":{"attributes":{}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024ca886f","endpointID":1}],"configuredReportings":[],"meta":{}},"11":{"profId":260,"epId":11,"devId":269,"inClusterList":[0,3,4,5,6,8,4096,64515,768,64513,64516],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"hwVersion":0}},"lightingColorCtrl":{"attributes":{"colorCapabilities":31,"colorTempPhysicalMin":153,"colorTempPhysicalMax":500}},"genOnOff":{"attributes":{"onOff":1}},"genGroups":{"attributes":{"nameSupport":0}},"genScenes":{"attributes":{"currentScene":0}}},"binds":[],"configuredReportings":[],"meta":{}},"12":{"profId":260,"epId":12,"devId":269,"inClusterList":[0,3,4,5,6,8,64515,768,64513],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"hwVersion":0}},"lightingColorCtrl":{"attributes":{"colorCapabilities":31,"colorTempPhysicalMin":153,"colorTempPhysicalMax":500}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":0,"dateCode":"20231211","swBuildId":"1.113.2","zclVersion":8,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1723928845402}

### Comments

The automated external definition does not correctly detect the state; I _think_ this is related to the lamp having two independent lights on it with independent endpoints.  identify _does_ work.

I've tried a few variants of definition including adding `endpointNames` as an arg to `philipsLight`, but the device is different enough to other definitions I've looked at such that I can't seem to make progress.

The default auto-generated definition results in UNSUPPORTED_ATTRIBUTE errors when attempting to retrieve state.

### External definition

```shell
const {
  deviceEndpoints,
  identify,
  commandsOnOff,
} = require("zigbee-herdsman-converters/lib/modernExtend");
const { philipsLight } = require("zigbee-herdsman-converters/lib/philips");

const definition = {
  zigbeeModel: ["LGT002"],
  model: "LGT002",
  vendor: "Signify Netherlands B.V.",
  description: "Automatically generated definition",
  extend: [
    deviceEndpoints({ endpoints: { "1": 1, "11": 11, "12": 12 } }),
    identify(),
    philipsLight({
      colorTemp: { range: [153, 500] },
      color: { modes: ["xy", "hs"], enhancedHue: true },
    }),
    philipsLight({
      colorTemp: { range: [153, 500] },
      color: { modes: ["xy", "hs"], enhancedHue: true },
    }),
    commandsOnOff(),
  ],
  meta: { multiEndpoint: true },
};

module.exports = definition;
```

toksvaeth, 2024-08-20:
Thanks.  I'd tried some variants of that setup without success.

Unfortunately still no luck.  I have tried with and without `meta: { multiEndpoint: true }` in the definition as well.

Here's an example of me trying to get the power on state of one of the lights:
```
[2024-08-20 23:31:31] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/0x001788010de259c4/get' with data '{"state":""}'
[2024-08-20 23:31:31] debug: 	z2m: Publishing get 'get' 'state' to '0x001788010de259c4'
[2024-08-20 23:31:31] debug: 	zh:controller:endpoint: ZCL command 0x001788010de259c4/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
[2024-08-20 23:31:31] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0x001788010de259c4:16977/1 (0,0,1)
[2024-08-20 23:31:31] debug: 	z2m: Received Zigbee message from '0x001788010de259c4', type 'readResponse', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
[2024-08-20 23:31:31] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x001788010de259c4', payload '{"action":null,"color_mode":"xy","linkquality":42,"power_on_behavior":null,"state":null,"state_1":"OFF","update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null}'
[2024-08-20 23:31:31] debug: 	zh:controller:endpoint: Error: ZCL command 0x001788010de259c4/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
[2024-08-20 23:31:31] error: 	z2m: Publish 'get' 'state' to '0x001788010de259c4' failed: 'Error: ZCL command 0x001788010de259c4/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
[2024-08-20 23:31:31] debug: 	z2m: Error: ZCL command 0x001788010de259c4/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
```

I don't know if it's relevant, but all reads appear to be going to the `1` endpoint regardless of config -- I've also tried setting a default of 11 as an experiment and repairing a number of times as well.

Using the dev console I have been able to run a number of reads against various clusters on the `11` and `12` endpoints that seem to work but writes do not (I get READ_ONLY) errors.

That said, my knowledge of zigbee is quite limited.  If you can point me in any good directions for documentation or better debugging methodology I'd appreciate it.

Koenkk, 2024-08-21:
> Here's an example of me trying to get the power on state of one of the lights:

What do you use for this, the refresh button in the frontend?

mundschenk-at, 2024-09-23:
I now have an `LGT001` (which is probably the same device for a different market or in a different color). @toksvaeth, did you make any progress?

mundschenk-at, 2024-09-23:
@Koenkk As @toksvaeth has experienced, the frontend always seems to query cluster 1.

toksvaeth, 2024-09-24:
Der,

I've made no new progress with diagnosis as I've been travelling of recent,
however, it appears your behaviour aligns.

I won't return home for another 2--3 weeks and until then cannot contribute.

mundschenk-at, 2024-09-24:
After some experimenting, I've found out some additional information:

`Endpoint 12` controls the "head" of the lamp, `endpoint 11` the "foot". `endpoint 1` is the two-button controller on top of the lamp head. I am able to control the lighting endpoints using the appropriate ZCL commands from the developer console (i.e. `endpoint 12`, `cluster 0x0006`, `command 0x0002,0x01` to toggle the endpoint on and off). 

I have some success with this definition (but with some caveats, see below):

```JS
const definition = {
    zigbeeModel: ['LGT001'],
    model: 'LGT001',
    vendor: 'Signify Netherlands B.V.',
    description: 'TODO',
    extend: [
        deviceEndpoints({"endpoints":{"1":1,"bottom":11,"top":12}}),

        philipsLight({"colorTemp":{"range":[153,500]},"color":{"modes":["xy","hs"],"enhancedHue":true}, "endpointNames": ["top"]}),
        philipsLight({"colorTemp":{"range":[153,500]},"color":{"modes":["xy","hs"],"enhancedHue":true},"gradient":true, "endpointNames": ["bottom"]}),
    ],
};
```

The frontend only partially works (i.e. color selection does not, but on/off and brightness do). From Home Assistant, color selection works fine. Hue effects work fine from HA from within the corresponding `light` entity, but not via the `gradient_scene` and `effect` entities added to the top-level device (this is also holds true for the frontend), as these commands are apparently always routed to `endpoint 1`. 

What also does not work is `power_on_behavior`, which gets displayed twice in the frontend but also always addresses the wrong endpoint. I've tried working around it by adding `philipsOnOff({"endpointNames":["top", "bottom"]})` to the `extend`, but that just breaks everything.

mundschenk-at, 2024-09-26:
Update: I have the converter running, but `modernExtend` needs some fixes (https://github.com/Koenkk/zigbee-herdsman-converters/pull/8031).

mundschenk-at, 2024-09-27:
@Koenkk Is there any way to assign custom names to the buttons (`genOnOff` cluster)? They are reported as `on` and `off`, but should really be something like `toggle` and `hue` or `scene`.

Koenkk, 2024-09-28:
Yes through a custom converter, [example](https://github.com/Koenkk/zigbee-herdsman-converters/blob/f67d7ff214d452554d1de0dbc6afaa71ce8184c4/src/converters/fromZigbee.ts#L3173)

mundschenk-at, 2024-10-03:
I have [added a PR](https://github.com/Koenkk/zigbee-herdsman-converters/pull/8069). I am a bit conflicted about `fz.ignore_command_off`, I think it is incorrectly named – it ignores `commandOffWithEffect`, not `commandOff` – but since it is widely used by the other Hue remote converters, so I didn't want to to touch it. (Instead I added an anonymous fZ converter for ignoring `commandOff`).

It would probably be a good idea to look in to making the fZ/tZ converters depend on arguments in some way to be able to have a true `modernExtend` for the various Hue remotes as they appear to share a lot of code.

I've also noted that the frontend still has problems with multiple endpoints in a device, basically none of the color swatches work (neither regular colors nor gradients). The lights can be controlled fine from Home Assistant.

closed 2024-10-03 by Koenkk