Koenkk/zigbee2mqtt #23655: Fail to include SMSZB-120

larronni, 2024-08-18:
### What happened?

When including SMSZB-120 it fails with error info:

```
error 2024-08-18 10:52:20z2m: Failed to configure '0x0015bc0031004ded', attempt 1 (Error: Bind 0x0015bc0031004ded/35 ssIasZone from '0x8c65a3fffe521f16/1' failed (Delivery failed for {"profileId":0,"clusterId":33,"sourceEndpoint":0,"destinationEndpoint":0,"options":4416,"groupId":0,"sequence":41}) at EmberOneWaitress.deliveryFailedFor (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:98:31) at EmberAdapter.onMessageSent (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:454:42) at Ezsp.emit (node:events:517:28) at Ezsp.ezspMessageSentHandler (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:5016:14) at Ezsp.callbackDispatch (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:770:26) at Ezsp.onAshFrame (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:430:22) at UartAsh.emit (node:events:517:28) at Immediate.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ember/uart/ash.ts:1187:45) at processImmediate (node:internal/timers:476:21))
info 2024-08-18 10:52:20z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"0x0015bc0031004ded"},"error":"Failed to configure (Bind 0x0015bc0031004ded/35 ssIasZone from '0x8c65a3fffe521f16/1' failed (Delivery failed for {\"profileId\":0,\"clusterId\":33,\"sourceEndpoint\":0,\"destinationEndpoint\":0,\"options\":4416,\"groupId\":0,\"sequence\":41}))","status":"error","transaction":"gn6j3-6"}'
info 2024-08-18 10:52:20zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "27964".
info 2024-08-18 10:52:22zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "27964".
info 2024-08-18 10:52:25zh:ember:ezsp: Received network/route error ROUTE_ERROR_SOURCE_ROUTE_FAILURE for "27964".
error 2024-08-18 10:52:26zh:controller: Interview failed for '0x0015bc0031004ded with error 'Error: Interview failed because can not get active endpoints ('0x0015bc0031004ded')'
error 2024-08-18 10:52:26z2m: Failed to interview '0x0015bc0031004ded', device has not successfully been paired
info 2024-08-18 10:52:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc0031004ded","ieee_address":"0x0015bc0031004ded","status":"failed"},"type":"device_interview"}'
error 2024-08-18 10:52:32zh:ember: Delivery of BROADCAST failed for '65533' [apsFrame={"profileId":0,"clusterId":6,"sourceEndpoint":0,"destinationEndpoint":0,"options":256,"groupId":0,"sequence":1} messageTag=255]
```

### What did you expect to happen?

It did work with ZZH adapter, so I expect it to work on EFR32MG21 adapter.

### Zigbee2MQTT version

1.39.1

### Adapter firmware version

7.4.1 [GA]

### Adapter

EFR32MG21

### Setup

Add-on on HomeAssistant on a NUC

Nerivec, 2024-08-18:
Excluding the interferences issue (improvements already in dev), this is likely related to https://github.com/Koenkk/zigbee2mqtt/issues/22492#issuecomment-2270474898
@Koenkk Seems more than one Develco device reports wrong statuses for various ZDO requests. Can you take a look at the converter, see about adding the same workaround as before?

Koenkk, 2024-08-18:
@Nerivec this is the interviewing part, which is before the converter kicks in

@larronni could you provide the debug log of this?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

larronni, 2024-08-23:
Now it fails to intervju it every time...

[log2.log](https://github.com/user-attachments/files/16728236/log2.log)

larronni, 2024-08-23:
After a "dillion" attempts, it just worked.... 

[logfile.log](https://github.com/user-attachments/files/16729060/logfile.log)

Nerivec, 2024-08-23:
Seems like the same behavior described in previously linked issue for that other Develco device. Random failures (at varying steps of the interview) and sometimes it works.
- Can you confirm the behavior is the same if you bring the device right next to the coordinator before interviewing (if possible)?
- Can you try with latest dev/edge? (Wondering if the interferences improvements might make them behave better)

closed 2024-12-22 by github-actions[bot]