Koenkk/zigbee2mqtt #23657: How should the non-existent effect 'none' be handled?

Minnozz, 2024-08-18:
### What happened?

I use zigbee2mqtt with Home Assistant. When you create a scene for lights in Home Assistant (through the UI or through the `scene.create` action), HA adds `effect: none` to the scene definition.

When you activate this scene the following message gets published by HA:
```json
{"state":"ON","brightness":254,"color_temp":238,"effect":"None"}
```

The value `none` ends up in this call to `utils.getFromLookup`:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/b524c39dbebcf32a8912e53820bca3bd362989ea/src/converters/toZigbee.ts#L1258

This causes an error to appear in the logs and an error notification in the web UI:
```
error 2024-08-18 14:08:45z2m: Publish 'set' 'effect' to 'Kitchen Table Ceiling' failed: 'Error: Value: 'none' not found in: [blink, breathe, okay, channel_change, finish_effect, stop_effect]'
```

This "none" value also causes issues with other HA integrations, like the `hue` integration: https://github.com/home-assistant/core/issues/122165

My question for z2m is: should this `none` value get special treatment, and if so, where? Maybe in the function I linked above?

Or is the problem with HA, and should it not send the value over mqtt in the first place?

### What did you expect to happen?

I expect the `effect: none` in HA to not cause any effect-related zigbee commands sent to my light bulbs.

### Zigbee2MQTT version

1.39.1

### Adapter firmware version

7.4.1 [GA]

### Adapter

ember

### Setup

Docker container running zigbee2mqtt, talking to Home Assitant

### Debug log

```
[2024-08-18 14:30:21] debug:    z2m:mqtt: Received MQTT message on 'zigbee2mqtt/Kitchen Table Ceiling/set' with data '{"state":"ON","brightness":254,"color_temp":238,"effect":"None"}'
[2024-08-18 14:30:21] debug:    z2m: Publishing 'set' 'brightness' to 'Kitchen Table Ceiling'
[2024-08-18 14:30:21] debug:    z2m: Publishing 'set' 'color_temp' to 'Kitchen Table Ceiling'
[2024-08-18 14:30:21] debug:    z2m: Publishing 'set' 'effect' to 'Kitchen Table Ceiling'
[2024-08-18 14:30:21] error:    z2m: Publish 'set' 'effect' to 'Kitchen Table Ceiling' failed: 'Error: Value: 'none' not found in: [blink, breathe, okay, channel_change, finish_effect, stop_effect]'
[2024-08-18 14:30:21] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Kitchen Table Ceiling', payload '{"brightness":254,"color_mode":"color_temp","color_temp":238,"linkquality":120,"power_on_behavior":null,"state":"ON","update":{"installed_version":16786434,"latest_version":16786432,"state":"idle"},"update_available":null}'
```

lux73, 2024-08-18:
> Or is the problem with HA, and should it not send the value over mqtt in the first place?

this...  HA should not send any 'none' Value to z2m - pls report the Bug to HA Dev's

Minnozz, 2024-08-18:
I did some more digging, and it seems that the cause is that the list of supported effects of a Light Group in Home Assistant is the union of all supported effects of its children. So when you mix lights of different types (or in this case, integrations) in a group, it is possible that z2m receives an effect that is not reported as a supported effect. I will report this as a HA bug.

A second HA bug is that if `None` is one of the supported effects of an entity, it seems to get auto-selected when creating a scene, which explains why this bug triggers without users explicitly causing it.

Minnozz, 2024-08-18:
> I did some more digging, and it seems that the cause is that the list of supported effects of a Light Group in Home Assistant is the union of all supported effects of its children. So when you mix [...]

I've reported the Light Group bug here: https://github.com/home-assistant/core/issues/124172
 
> A second HA bug is that if `None` is one of the supported effects of an entity, it seems to get auto-selected when creating a scene, which explains why this bug triggers without users explicitly causing it.

(This is because the current effect of the group is reported as the most common current effect of its members)

closed 2024-12-22 by github-actions[bot]