Koenkk/zigbee2mqtt #23659: Unable to bind 'IKEA Styrbar' remote to Z2M light group

SamSammy123, 2024-08-18:
### What happened?

**Summary:**
The IKEA Styrbar remote is not controlling a group of Ikea GU10 lights correctly after being successfully bound in the Z2M frontend.

**Problem:**
The Styrbar remote doesn't control the light group after binding. No response when pressing any buttons.

**Troubleshooting:**
Successfully bound another dimmer (Ikea RODRET) to the same group, but it lacks color temperature control.
Pairing the Styrbar remote individually with each light works, but it’s not feasible due to cluster limitations and synchronization issues.

**Attempted Solutions:**
Binding the Styrbar remote directly to the group.
Testing with another dimmer (RODRET), which works but lacks necessary features for temperature control.

**Current Workaround:**
Considering Home Assistant automations, though direct Z2M control is preferred for smoother operation and reliability.

**Environment:**
Remote: [STYRBAR (E2001/E2002)](https://www.zigbee2mqtt.io/devices/LED2005R5_LED2106R3.html#ikea-led2005r5%252Fled2106r3) - Firmware v2.4.16 (newest)
Lights: [TRADFRI bulb GU10 (LED2005R5/LED2106R3)](https://www.zigbee2mqtt.io/devices/E2001_E2002.html#ikea-e2001%252Fe2002) - Firmware v3.0.21 (newest)

**Screenshots:**
Screenshot 1: [Group bind with RODRET](https://github.com/user-attachments/assets/e93e9a36-051b-412d-8b7a-21205abe4f26).
Screenshot 2: [Individual pairing with STYRBAR](https://github.com/user-attachments/assets/8317eb71-2c5e-4a1d-922b-700414aa0324)

### What did you expect to happen?

I expected that the Styrbar remote would control the light group normally. 

### How to reproduce it (minimal and precise)

1. Reset lights and remote control to factory settings
2. Add to Z2M
3. Group lights (4 group members)
4. Bind Styrbar remote to group via Z2M frontend

### Zigbee2MQTT version

1.39.1

### Adapter firmware version

7.4.2 [GA]

### Adapter

SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2

### Setup

Add-on on HomeAssistant RPI3b+

thecode, 2024-08-20:
From the doc you linked: https://www.zigbee2mqtt.io/devices/E2001_E2002.html#latest-firmware

```
Note that the remote loses the capability to bind to a group in the latest firmware.
```

This is a firmware issue, so you either get a new one and don't update it or use automation for this

b2un0, 2024-08-23:
duplicate #23584 23584

closed 2024-08-23 by SamSammy123