Koenkk/zigbee2mqtt #23661: Physical state change not reflected in Zigbee2MQTT for Gledopto GL-SD-003P

jaco, 2024-08-18:
### What happened?

When changing the state (ON/OFF) of the Gledopto GL-SD-003P device using the physical input, the new state is not reflected in Zigbee2MQTT. The actual state of the device is only updated in Zigbee2MQTT after manually refreshing the state via the web interface.

### What did you expect to happen?

The state change (ON/OFF) via the physical input should be immediately reflected in Zigbee2MQTT without requiring a manual refresh.

### How to reproduce it (minimal and precise)

1. Change the state (ON/OFF) of the device using the physical input (e.g., a wall switch or button)
2. Observe the state of the device in the Zigbee2MQTT web interface. Note that the state does not change to reflect the physical input.
3. Manually refresh the device state via the Zigbee2MQTT web interface.

### Zigbee2MQTT version

1.39.1 

### Adapter firmware version

20220219

### Adapter

SLZB-06

### Setup

Add-on on Home Assistant OS, x86-64

### Debug log

No logs after changing physical state

gh0st125, 2024-11-28:
I have the exact same issue. Any news on this?

ghostrider21991, 2025-01-06:
I have the exact same issue. @Koenkk is there any chance u can check this? With the Hue Bridge it works.

Koenkk, 2025-01-06:
Fixed! After updating press the yellow re-configure button on the frontend.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2025-01-06 by Koenkk

dkkr23, 2025-01-14:
I have the exact issue with the model GL-SD-001. @Koenkk  please can you extend the fix to the model as well. Thanks

Koenkk, 2025-01-15:
@dkkr23 done!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

dkkr23, 2025-01-16:
> [@dkkr23](https://github.com/dkkr23) done!
> 
> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

Apologies @Koenkk ! The model is GL-SD-001P. I didn’t realise there is a “P” at the end.

Koenkk, 2025-01-17:
Also fixed for `GL-SD-001P`

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

dkkr23, 2025-01-26:
@jaco has this fix resolved the issue for you? I still dont get the state updated with the physical switch.

ghostrider21991, 2025-01-27:
> [@jaco](https://github.com/jaco) [@gh0st125](https://github.com/gh0st125) [@ghostrider21991](https://github.com/ghostrider21991) has this fix resolved the issue for you? I still dont get the state updated with the physical switch.

Yes, it works. You have to use the EDGE or DEV Version of Zigbee2mqtt. And u also have to klick on the Orange "reconfig button".

jaco, 2025-01-27:
I have moved inputs to the another controller to prevent messing up the lights…

I will check that next week :)

romain3w, 2026-02-02:
Hello, I have the problem on the GL-C-003P  (firmware 20951203). I must go on EDGE or DEV Version of Zigbee2mqtt or it's solved now on main version?

ghostrider21991, 2026-02-02:
It’s solved in the main version.