Koenkk/zigbee2mqtt #23663: I encountered a problem when switching between Zigbee Dongles in Zigbee2MQTT.

Yukyniss, 2024-08-19:
### What happened?

I initially set up Zigbee2MQTT using a Zigbee Dongle-E.
I successfully added a smart plug, and it was functioning correctly, including both control and status reporting.
I then unplugged the Dongle-E and replaced it with a Zigbee Dongle-P.
I changed configuration.yaml, changing [port] to the path of -p and [adapter] to zstack
Upon restarting Zigbee2MQTT with the Dongle-P, I encountered the following error: [2024-08-19 16:33:16] error: z2m: Error: network commissioning timed out - most likely network with the same panId or extendedPanId already exists nearby

### What did you expect to happen?

I hope I can start zigbee2mqtt with -p and can control the devices I added previously with -e

### Zigbee2MQTT version

1.38.0

### Adapter firmware version

-e: 6.10.3 -p: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210708,"transportrev":2},"type":"zStack3x0"}'

### Adapter

sonoff zigbee dongle -e and sonoff zigbee dongle -p

### Setup

macos  and npm run start

### Debug log

The logs are from when Zigbee2MQTT was initially started using Dongle-E:
[log.log](https://github.com/user-attachments/files/16660258/log.log)

**database.db:** 
{"id":1,"type":"Coordinator","ieeeAddr":"0xdc8e95fffe258190","nwkAddr":0,"manufId":0,"epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":48879,"inClusterList":[0,3,6,10,25,26,768],"outClusterList":[0,3,4,5,6,8,32,768,1024,1026,1029,1030,1280,2817,2819,2820,1794,4096,64513,64514],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":1724056141976}

**coordinator_backup.json:**
{"metadata":{"format":"zigpy/open-coordinator-backup","version":1,"source":"zigbee-herdsman@0.49.2","internal":{"date":"2024-08-19T08:29:40.983Z","ezspVersion":8}},"stack_specific":{"ezsp":{"hashed_tclk":"0d401f3e33faf9dba7bd58eac8f0a110"}},"coordinator_ieee":"dc8e95fffe258190","pan_id":"6f54","extended_pan_id":"2d3d4d5d6d7d8d9d","nwk_update_id":0,"security_level":5,"channel":15,"channel_mask":[11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26],"network_key":{"key":"01030507090b0d0f00020406080a0c0d","sequence_number":0,"frame_counter":79},"devices":[]}

start with dongle-p: 
[log.log](https://github.com/user-attachments/files/16660265/log.log)

RobLeighton67, 2024-08-20:
Look at the docs about generating pan id (and network key) using configuration settings - prevents the z2m trying to reuse the pan id of the old adapter (which is the error you're seeing, I think)

Yukyniss, 2024-08-21:
> Look at the docs about generating pan id (and network key) using configuration settings - prevents the z2m trying to reuse the pan id of the old adapter (which is the error you're seeing, I think)

panid, networkkey and extendpanid are all generated by z2m (I filled in GENERATE in yaml), which generates specific values ​​after using -e. I hope to use -p to restore the same zigbee network so that I don't need to re-add the device. After this problem occurred, I cleared the backup and then started it with -p and it worked normally. But I want to know why -p can't restore the zigbee network.

lux73, 2024-08-24:
shuffling different Chipsets **CC2652P** vs **EFR32MG21** is never a good Idea - some Values are stored inside the Coordinator

closed 2024-12-22 by github-actions[bot]