Koenkk/zigbee2mqtt #23667: _TZE200_vvmbj46n (Tuya Temperature and Humidity - TH05Z)

Pizzabroodje, 2024-08-20:
### What happened?

The battery percentage is inaccurate. With brand-new 1.5v alkaline batteries in all 8 I have, they show as having 150% battery. 

There's others reporting the same issue (#19731 , #19702).

Yesterday, three of the four that are currently in use, stopped sending data. They did seem to get a link, as last seen just went back and forth between right now and just now. The display also was still on on all of them, also with the right temp and all. After removing and putting back a battery from them, they came back.

### Zigbee2MQTT version

1.39.1

### Adapter firmware version

7.3.3.0 build 239

### Adapter

EZSP

### Setup

In home assistant, virtualbox container on a Windows 10 PC

philipp, 2024-08-22:
I've been meaning to add the "recognised as ZTH05Z instead of TH05Z" part of this as a new issue myself, glad to see someone beat me to it.
Haven't noticed the battery percentage problem (but I'm using 1.2V rechargeables, right now, so my accuracy expectations are low :-)
The only reason I even noticed the mismatch in names was because the ZTH05Z expects a percentage for the update frequency (which is weird) while for the TH05Z it's minutes (also kind of a long default of 120?) but the device initializes with a value of 120 (the intended minutes) in a field that the system thinks should be "%" and so it's been throwing warnings everywhere.

I assume the fundamental problem here is that it's being tagged as a "ZTH05Z" instead of a "TH05Z", and the other weird behaviors are fallout from that initial misidentification.

The place where this actually needs to be fixed is in the https://github.com/Koenkk/zigbee-herdsman-converters project in src/devices/tuya.ts
There's also another (stale) issue in that repo regarding this device :https://github.com/Koenkk/zigbee-herdsman-converters/issues/6832

I'm still learning how all this works, so I'm curious to see how this gets resolved.

Pizzabroodje, 2024-08-22:
> I've been meaning to add the "recognised as ZTH05Z instead of TH05Z" part of this as a new issue myself, glad to see someone beat me to it. Haven't noticed the battery percentage problem (but I'm using 1.2V rechargeables, right now, so my accuracy expectations are low :-) The only reason I even noticed the mismatch in names was because the ZTH05Z expects a percentage for the update frequency (which is weird) while for the TH05Z it's minutes (also kind of a long default of 120?) but the device initializes with a value of 120 (the intended minutes) in a field that the system thinks should be "%" and so it's been throwing warnings everywhere.
> 
> I assume the fundamental problem here is that it's being tagged as a "ZTH05Z" instead of a "TH05Z", and the other weird behaviors are fallout from that initial misidentification.
> 
> The place where this actually needs to be fixed is in the https://github.com/Koenkk/zigbee-herdsman-converters project in src/devices/tuya.ts There's also another (stale) issue in that repo regarding this device :[Koenkk/zigbee-herdsman-converters#6832](https://github.com/Koenkk/zigbee-herdsman-converters/issues/6832)
> 
> I'm still learning how all this works, so I'm curious to see how this gets resolved.

What percentage do you get with fully charged batteries? Have you drained any yet, and if so, at what percentage did it die?
I'm assuming my measurements are accurate, as I tested on 8 different units with non-rechargeable 1.5V alkaline batteries fresh from the store. And that's what should be used according to the specs.

The range isn't in minutes either. For me it's set to 12% and updating every 5 minutes. Doubling the percentage also doesn't double the time. It's a bit of a hassle to find out the actual update time like this, but doable (just have to manually check when it updates).

The random dropouts are also odd. All 4 have been running again without issues after reseating a battery in them.

Pizzabroodje, 2024-08-22:
I did some digging. Here's how to add support for new tuya devices:
https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html

I'll check if the right data points are used, and if they don't need to be adjusted.
The battery might just have to be 2/3rd of what it is, as 150 would be 100 then 🤷🏻‍♂️
Maybe there's also some data point for the voltage though which then can be used to calculate the percentage.

I'll look into this more later, I have to sleep now 😴

Pizzabroodje, 2024-09-21:
@philipp 
Did you do any more digging?

I got a Tuya Gateway in the mail today, and just checked the data points of the sensor through it. It seems like the values are in line with what it should be after all.

The updating time for example, is in minutes according to the developer platform. What's weird though is that it's updating every 5 minutes for me while it's set to 12... 

The battery is also reporting the same (88% currently, have yet to test with a set that's above 100 like before, but I guess it'll go up to 150 like in Z2M).

In the smart life app it's recognized with the right picture, but the data just seems to be off.

Maybe Tuya themselves just used the base code of another Tuya device, hence it's reporting falsely.

closed 2024-12-22 by github-actions[bot]