Koenkk/zigbee2mqtt #23678: Zigbee Solar Rain Sensor

mikejclark, 2024-08-20:
### What happened?

I have the same device and used the external converter from this device support request:
https://github.com/Koenkk/zigbee2mqtt/issues/23532

For me the battery reporting does not work. I see only Null, or if I press the button on the back of the device it reports N/A for a short time and then reverts to Null.

[image]

### What did you expect to happen?

Should report percentage?

### Zigbee2MQTT version

1.39.1

### Adapter firmware version

20220219

### Adapter

Sonoff Zigbee 3.0 P

### Setup

Add-on. x84

### Debug log

info 2024-08-20 19:51:15z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Solar Rain Sensor RB-SRAIN01', payload '{"battery":null,"battery_low":false,"cleaning_reminder":null,"illuminance":1724,"illuminance_average_20min":1959,"illuminance_maximum_today":6128,"linkquality":87,"rain_intensity":48,"tamper":false,"water_leak":false}'

rdorys, 2024-08-29:
Hi, @mikejclark did you try to set it in the reporting tab : 
[image]
Don't forget to "wake it up" pressing the pairing button just before clicking on "Apply".
Is it working ?

RubenKelevra, 2024-09-02:
[`1.40.0`](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.40.0) got official support, do you still have issues with this version?

rdorys, 2024-09-02:
You're right, it is missing. I bet it is a copy/paste mistake.
I created a PR to add it again. https://github.com/Koenkk/zigbee-herdsman-converters/pull/7931

mikejclark, 2024-09-02:
Battery reporting now working 1.40.0

closed 2024-09-02 by mikejclark

fisch55, 2024-09-04:
> You're right, it is missing. I bet it is a copy/paste mistake. I created a PR to add it again. [Koenkk/zigbee-herdsman-converters#7931](https://github.com/Koenkk/zigbee-herdsman-converters/pull/7931)

Do I have to wait until new Release or can I change it manually to get the water leak back again?

rdorys, 2024-09-04:
> > You're right, it is missing. I bet it is a copy/paste mistake. I created a PR to add it again. [Koenkk/zigbee-herdsman-converters#7931](https://github.com/Koenkk/zigbee-herdsman-converters/pull/7931)
> 
> Do I have to wait until new Release or can I change it manually to get the water leak back again?

You can switch to the latest-dev branch to get it back or create the appropriate external converter file as mentioned earlier in this issue. 
If you only use stable release, then you will get it on 1st October.