Koenkk/zigbee2mqtt #23681: Integration problem : Legrand 067776(A) module

apages2, 2024-08-21:
### What happened?

Good morning,

I have a problem with the integration of the Legrand 067776(A) module.

when I physically press the button (up, down, stop), the feedback Z2M receives is always "action": "stopped".

z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Physique Volet Bureau', payload '{"action":"stopped","calibration_mode":"classic_nllv","color_mode":"xy","cover_mode":{"calibration":false,"led":false,"maintenance":false,"reversed":false},"device":{"applicationVersion":0,"dateCode":" \u000020221202\u0000\u0000\u0000\u0000\u0000","friendlyName":"Physique Volet Bureau","hardwareVersion":1,"ieeeAddr":"0x0040ceb","manufacturerID":4129,"manufacturerName":" Legrand","model":"067776A","networkAddress":54843,"powerSource":"Mains (single phase)","softwareBuildID":"003d","stackVersion":66,"type":"Router","zclVersion":2},"led_if_on":"OFF","led_in_dark":"OFF","linkquality":255,"position":50,"state":"OPEN","tilt":100,"update":{"installed_version":4015615,"latest_version":4015615,"state":"idle"},"update_available":false}'

While on the Legand app, the state looks different depending on the button.

Likewise, when I click on a Z2M action, nothing is sent to the Zigbee network (while it is when I click on the identity button).

Finally, there doesn't seem to be a "group" cluster in toZigbee. is this normal?

Best Regard

### What did you expect to happen?

I should receive an action like open, close, stop

### Zigbee2MQTT version

1.39.0

### Adapter firmware version

0x26780700

### Adapter

ConBee2

### Setup

Add-on Z2M on ODROID+jeedom

### Debug log

z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Physique Volet Bureau', payload '{"action":"stopped","calibration_mode":"classic_nllv","color_mode":"xy","cover_mode":{"calibration":false,"led":false,"maintenance":false,"reversed":false},"device":{"applicationVersion":0,"dateCode":" \u000020221202\u0000\u0000\u0000\u0000\u0000","friendlyName":"Physique Volet Bureau","hardwareVersion":1,"ieeeAddr":"0x0040ceb","manufacturerID":4129,"manufacturerName":" Legrand","model":"067776A","networkAddress":54843,"powerSource":"Mains (single phase)","softwareBuildID":"003d","stackVersion":66,"type":"Router","zclVersion":2},"led_if_on":"OFF","led_in_dark":"OFF","linkquality":255,"position":50,"state":"OPEN","tilt":100,"update":{"installed_version":4015615,"latest_version":4015615,"state":"idle"},"update_available":false}'

apages2, 2024-08-21:
@FabianMangold

FabianMangold, 2024-08-22:
Hi @apages2,

I'd have to check how my own devices behave, but for a start, do you have "HomeAssistant legacy triggers" enabled ?
If i remember correctly, those seem to interfere with the status. Please disable those should they be enabled, and let me know the outcome.

Fabian

Message ID: ***@***.***>

apages2, 2024-08-22:
Hello

 

I don't use homeassistant, but jeedom, so I don't have homeassistant integration activated

 

Best regards

apages2, 2024-08-31:
hello @FabianMangold 

were you able to check that the shutters were working? 

in particular the operation of the cluster group for this type of equipment, because I cannot control other shutters from 1 general shutter switch, as is the case with lighting switches for example

Best regards

FabianMangold, 2024-08-31:
Hi @apages2 ,

I am sorry, unfortunately I have very little time lately, and i mostly far from the devices, which makes testing complicated!

Regarding the status, you are right, those do not work correctly at all. If i remember correctly, they have never worked flawlessly. If the Legrand app reports them correctly, we should be able to reproduce this too.
Could you please report how the switch behaves, i.e. what gets reported in the app when the roller goes up or down, gets stopped, when pressing the button (long / short) ?
The behavior might be somewhat different depending on the calibration mode, but since I don't have access to the Legrand App to compare, I'd appreciate if you could try to summarize the behaviour your switch(es) expose.

Regarding the group, I don't really know! I group the switches on higher level (not Zigbee) via automation, hence i've never paid attention.
Due to limited time at the moment, not sure if anyone else could take a look at this part of the issue ?

Thanks for your understanding.
Best,
Fabian

apages2, 2024-09-15:
[https://github.com/Koenkk/zigbee-herdsman-converters/pull/7992](url)

closed 2024-09-15 by apages2