Koenkk/zigbee2mqtt #23683: CR-MNZ1

miomia, 2024-08-21:
### What happened?

The device CR-MNZ1 on the site it is stated that you are supported in Zigbee2mqqt but after adding to Home Assistants not everything is functional Monitoring does not work: current, voltage, power.
`z2m: Failed to configure '0x70b3d52b60092d29', attempt 5 (Error: ZCL command 0x70b3d52b60092d29/1 haElectricalMeasurement.configReport([{"attribute":"rmsVoltage","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":5}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNREPORTABLE_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:351:28) at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:870:26) at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:611:9) at Object.rmsVoltage (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:221:5) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:3392:13) at Configure.configure (/app/lib/extension/configure.ts:126:13) at Configure.onMQTTMessage (/app/lib/extension/configure.ts:56:21) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))`

### Zigbee2MQTT version

1.39.1-1

### Adapter firmware version

20231030.1

### Adapter

EZSP v9

### Setup

Operating System 12.4

Brucepbd, 2024-09-24:
Also reported here:
https://github.com/Koenkk/zigbee2mqtt/issues/22808

Brucepbd, 2024-09-24:
I have the same issue.  The device does report in ZHA.  Current is in mA.
[image]
[image]

Brucepbd, 2024-09-24:
To convert current from mA to A I use a template helper:

    sm2_amps:
      friendly_name: "SM2 Amps"
      unit_of_measurement: "A"
      device_class: "current"
      value_template: "{{ states('sesensor.sm2_30a_current')|int /1000}}"

Brucepbd, 2024-09-24:
I found the external converter works best for z2m.  It reports in Amperes.  Everything seems to work.

To install see https://github.com/Koenkk/zigbee2mqtt/issues/22808

[image]

closed 2024-12-22 by github-actions[bot]