Koenkk/zigbee2mqtt #23684: SMSZB-120 Fails to configure due to TABLE_FULL

petermnt, 2024-08-21:
### What happened?

When pairing a SMSZB-120 HW version 5.2.0 I get TABLE_FULL during configuration

It happens on my zstack CC2652P (ZBDongle-P) and on ember on the SLZB-06M.
On ezsp on the SLZB-06M it seems to work fine (tried twice)

I have the same (or very similar) issues with 4 different devices.

Configuration tries to bind 
35: genBasic, genBinaryInput, genPollCtrl, PowerCfg, ssIasWd, ssIasZone
38: Temperatur

After it fails to configure I can manage to manually bind some.  But if I check all of them (especially the ssIas ones  I get the TABLE_FULL eventually)

### How to reproduce it (minimal and precise)

* Reset the SMSZB-120
* Open the network
* Let it configure

### Zigbee2MQTT version

1.39.1

### Adapter firmware version

EmberZNet 7.4.1 & zStack3x0 20230507

### Adapter

SLZB-06M & ZBDongle-P

### Setup

HA Add-on & x86-64 LXC 

### Debug log

[ember.txt](https://github.com/user-attachments/files/16691828/ember.txt)

mrmaximas, 2024-08-21:
try this https://github.com/Koenkk/Z-Stack-firmware/discussions/505

petermnt, 2024-08-21:
[zstack new firmware.txt](https://github.com/user-attachments/files/16700629/zstack.new.firmware.txt)

Same problem with the new firmware.

```
[2024-08-22 00:26:57] debug: 	zh:controller:endpoint: Error: Bind 0x0015bc0031026dbb/35 genBinaryInput from '0x00124b002c3fdedb/1' failed (ZDO error: bind failed with status 'TABLE_FULL' (140))
```

Koenkk, 2024-08-22:
> On ezsp on the SLZB-06M it seems to work fine consistently

This is because ezsp doesn't check the result of the device, it ignores the error.

Does the device work correctly?

petermnt, 2024-08-22:
It seems to depend on what actually manages to bind and what doesn't.
(It's also a bit hard to test all aspects of the device)

It seems that it never manages to bind all 7 things, but does seem to work (at least the temperature, power and test) in some attempts.

Could it be that it doesn't need bind (or support) all 7 items?  And is there any way to figure out which ones would be optional/unnecessary?

It seems very consistent with all 4 SMSZB-120's, so it does sound like it's 'as intended'.

Koenkk, 2024-08-24:
Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/e444ae56556c9b7af5615adae4dc2f0a): 

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, 
- in the frontend, check if the description of this device now ends with `CUSTOM` (this indicates the external converter has been loaded correctly)
- re-pair the device, check if issue is fixed and everything is still working, if not try adding back some of the commented things on line 59

petermnt, 2024-08-24:
Thanks for the suggestion.  I am currently unable to test it but I will test it as soon as I'm back.

RobLeighton67, 2024-09-04:
I'm receiving the same error during pairing with a CC2652P coordinator - Sonoff P .
Device seems to working OK.

peter4200, 2024-09-04:
I have the same problem using a Sonoff Dongle-E and frient SMSZB-120
```
Zigbee2MQTT version                 1.40.0 commit: unknown
Coordinator type                    EmberZNet
Coordinator revision                7.4.3 [GA]
Frontend version                    0.7.4
zigbee-herdsman-converters version  20.8.4
zigbee-herdsman version             0.57.1
```

Koenkk, 2024-09-05:
@peter4200 please post the debug log of this

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

peter4200, 2024-09-05:
This one?
[log.log](https://github.com/user-attachments/files/16898334/log.log)

Koenkk, 2024-09-05:
Yes, but can you:
- Force remove the device
- Stop z2m
- Start z2m and pair the device <- provide the log of this

peter4200, 2024-09-06:
Forced remove of the device
Turned on debug logging
Restareted z2m
Paired the device

[log.log](https://github.com/user-attachments/files/16903529/log.log)

Koenkk, 2024-09-06:
Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/3311969512f2e4307dcee8b6a22e1cbb): 

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, 
- in the frontend, check if the description of this device now ends with `CUSTOM` (this indicates the external converter has been loaded correctly)
- check if issue is fixed and that the device still works.

peter4200, 2024-09-06:
Strange, I have followed the instructions but the description does not change to CUSTOM. I can see in the log that ext_converter.js is being loaded. I have also tried deleting the device and adding it again. Nothing helps. What am I doing wrong?

Koenkk, 2024-09-06:
Forgot to add it, try with https://gist.github.com/Koenkk/3311969512f2e4307dcee8b6a22e1cbb

peter4200, 2024-09-06:
Should I save ext.js i the same folder as configuration.yams and ext_converter.js? 
And should it be added to external_converters i configuration.yaml?

Koenkk, 2024-09-07:
See https://www.zigbee2mqtt.io/guide/configuration/more-config-options.html#external-converters

eerr0EERR, 2024-09-07:
Z2M: 1.40.0-dev commit: [0d46354]  -  zStack3x0: 20230507
Force remove: "0x0015bc003100dd4e"
Log: debug
External convertor: https://gist.github.com/Koenkk/3311969512f2e4307dcee8b6a22e1cbb
Restart Z2M => log in annex (filtered out all but "0x0015bc003100dd4e" related)
[log.log](https://github.com/user-attachments/files/16918482/log.log)

After pairing:
  - Clusters correctly binded: genPollCtrl, PowerCfg
  - Clusters not correctly binded: genBasic, genBinaryInput, ssIasWd, ssIasZone

Binding in Frontend:
  - genBasic: OK, genBinaryInput: OK
  - ssIasZone: 2024-09-07 15:39:14z2m: Failed to bind cluster 'ssIasZone' from '0x0015bc003100dd4e' to 'Coordinator' (Error: Bind 0x0015bc003100dd4e/35 ssIasZone from '0x00124b00258b8164/1' failed (ZDO error: bind failed with status 'TABLE_FULL' (140)))
  - ssIasWd: 2024-09-07 15:40:06z2m: Failed to bind cluster 'ssIasWd' from '0x0015bc003100dd4e' to 'Coordinator' (Error: Bind 0x0015bc003100dd4e/35 ssIasWd from '0x00124b00258b8164/1' failed (ZDO error: bind failed with status 'TABLE_FULL' (140)))

Thx 4 all your wonderful work

peter4200, 2024-09-08:
Both of my alarms are now in Z2M, however Reliability and Fault of both alarms have a status of NULL:
```
{
    "battery": 100,
    "battery_low": false,
    "linkquality": 196,
    "restore_reports": true,
    "smoke": false,
    "supervision_reports": true,
    "temperature": 26.62,
    "test": false,
    "voltage": 3000,
    "alarm": null,
    "fault": null,
    "max_duration": null,
    "reliability": null
}
```

eerr0EERR, 2024-09-10:
With updated convertor https://gist.github.com/Koenkk/3311969512f2e4307dcee8b6a22e1cbb ... exactly the same as in my previous feedback ... https://github.com/Koenkk/zigbee2mqtt/issues/23684#issuecomment-2335192712

[2024-09-10 16:25:30] error: 	z2m: Failed to bind cluster 'ssIasWd' from '0x0015bc003100dd4e' to 'Coordinator' (Error: Bind 0x0015bc003100dd4e/35 ssIasWd from '0x00124b00258b8164/1' failed (ZDO error: bind failed with status 'TABLE_FULL' (140)))
[2024-09-10 16:25:30] error: 	z2m: Failed to bind cluster 'ssIasZone' from '0x0015bc003100dd4e' to 'Coordinator' (Error: Bind 0x0015bc003100dd4e/35 ssIasZone from '0x00124b00258b8164/1' failed (ZDO error: bind failed with status 'TABLE_FULL' (140)))

Haraade, 2024-09-10:
Managed to get more data from a device after OTA update. But only one of three units. Firmware was obtained from https://github.com/Develco-Products/frient_upgrade_images/blob/main/IntelligentSmokeAlarm_4.0.8.zigbee

{
    "battery": 100,
    "battery_low": false,
    "enrolled": true,
    "fault": false,
    "linkquality": 168,
    "max_duration": 5,
    "reliability": "no_fault_detected",
    "restore_reports": true,
    "smoke": false,
    "supervision_reports": true,
    "temperature": 16.43,
    "test": false,
    "update": {
        "installed_version": 262152,
        "latest_version": 262152,
        "state": "idle"
    },
    "voltage": 3000,
    "zone_id": 23
}

peter4200, 2024-09-10:
I see the same result as mentioned in previous posts and still have NULL values in Reliability and Fault

Haraade, 2024-09-10:
Condition reported from another unit.
This device does not report "ias_cie_address":

{
    "battery": 80,
    "battery_low": false,
    "linkquality": 216,
    "max_duration": 5,
    "restore_reports": true,
    "smoke": false,
    "supervision_reports": true,
    "test": false,
    "update": {
        "installed_version": 262152,
        "latest_version": 262152,
        "state": "idle"
    },
    "voltage": 2900,
    "alarm": null,
    "fault": null,
    "reliability": null,
    "temperature": null,
    "update_available": null
}

Haraade, 2024-09-10:
One thing I wonder, are these units completely identical ?. They are supplied by Frient (https://frient.com/products/intelligent-smoke-alarm/) and Develco (https://www.develcoproducts.com/products/sensors-and-alarms/intelligent-smoke-alarm/#SmokeAlarmTechnicalSpecifications). These three that I have are supplied by Frient, but are registered as Develco in Z2M. Which firmware (OTA) will then be right to use?

[image]

[image]

RobLeighton67, 2024-09-10:
Frient is the B2C company, Develco Products is the B2B White label company
- same people, met most of them!

Haraade, 2024-09-10:
Found some old discussions related to my previous comment.

https://github.com/Koenkk/zigbee2mqtt/issues/9250#issuecomment-947937091

https://community.home-assistant.io/t/frient-smoke-detector-zha/394354/1

Koenkk, 2024-09-12:
@Haraade can you try if everything is OK with this one: https://gist.github.com/Koenkk/3311969512f2e4307dcee8b6a22e1cbb ?

Haraade, 2024-09-12:
> @Haraade can you try if everything is OK with this one: https://gist.github.com/Koenkk/3311969512f2e4307dcee8b6a22e1cbb ?

Have tested and here is the log. Same feedback on all three devices.

error 2024-09-12 21:38:22z2m: Failed to configure 'Fire Reporter 1', attempt 1 (Error: Bind 0x0015bc0031021d68/38 msTemperatureMeasurement from '0xe0798dfffed3c026/1' failed (Delivery failed for {"profileId":0,"clusterId":33,"sourceEndpoint":0,"destinationEndpoint":0,"options":4416,"groupId":0,"sequence":3}) at EmberOneWaitress.deliveryFailedFor (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/oneWaitress.ts:101:31) at EmberAdapter.onMessageSent (/app/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:480:42) at Ezsp.emit (node:events:517:28) at Ezsp.ezspMessageSentHandler (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:4982:14) at Ezsp.callbackDispatch (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:738:26) at Ezsp.onAshFrame (/app/node_modules/zigbee-herdsman/src/adapter/ember/ezsp/ezsp.ts:398:22) at UartAsh.emit (node:events:517:28) at Immediate.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/ember/uart/ash.ts:1169:45) at processImmediate (node:internal/timers:476:21))

Remark! The units have not been re-pair, as they are not physically available to me now.

sarthurdev, 2024-09-16:
Having the same issue, external converter didn't seem to help.

```
z2m: Failed to configure '0x0015bc0031020b45', attempt 1 (Error: Bind 0x0015bc0031020b45/35 genPowerCfg from '0x00124b00258f1ae5/1' failed (ZDO error: bind failed with status 'TABLE_FULL' (140)) at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:1126:36)
```

I have two other devices of this model on older firmware 4.0.7 and they work without issue.

RobLeighton67, 2024-09-18:
Just to confirm what @sarthurdev reports: 

I have a 4.0.8 Develco SMSZB-120 generating the same Bind failed (TABLE_FULL) error for genPowerCfg - happens when joining and every time I restart Z2M - no obvious effect on functaionality though? 

I get the same ZDO error for msRelativeHumidity on a Heiman HS2AQ-EM air sensor (app version 35), again when joining and every time z2m restarts afterwards.

[Sonoff CC2652P dongle 240315 fw)

peter4200, 2024-09-19:
Just tried to delete it and rejoin. Reliabillty and Fault are still Nule and now Battery and Voltage alså become null, so it just get worse :-)
```
Reliability    Null
Fault          Null
Temperature    25.18°C
Battery        Null
Voltage        Null
```

I appreciate the great work you do

Koenkk, 2024-09-20:
Did the configure succeed now?

peter4200, 2024-09-20:
No, it got worse. Battery and Voltage are now missing and Reliability and Fault are still missing.

Koenkk, 2024-09-20:
Updated https://gist.github.com/Koenkk/3311969512f2e4307dcee8b6a22e1cbb, can you try to re-pair the device with this again?

peter4200, 2024-09-22:
Of course can I try again.
I deleted the device
Enabled debug log
Restarted Z2M
Rejoined the device

Log file attached
[z2mlog.log](https://github.com/user-attachments/files/17088561/z2mlog.log)

Koenkk, 2024-09-23:
Can you try again with : https://gist.github.com/Koenkk/3311969512f2e4307dcee8b6a22e1cbb ?

peter4200, 2024-09-23:
Of course, but the result are still the same:
```
Reliability    Null
Fault          Null
Temperature    Null °C
Battery        Null %
Voltage        Null mV

```
[z2mlog.log](https://github.com/user-attachments/files/17102681/z2mlog.log)

Koenkk, 2024-09-23:
Found the issue, please try again with https://gist.github.com/Koenkk/3311969512f2e4307dcee8b6a22e1cbb

peter4200, 2024-09-23:
And it is a little bit better:
```
Reliability    Null
Fault          Null
Temperature    25.18°C
Battery        Null %
Voltage        Null mV
```

[z2mlog.log](https://github.com/user-attachments/files/17102937/z2mlog.log)

Schlammschlumpf, 2024-09-24:
Same problem here. I'm complete new to zigbee (my first device). So sorry for my silly question:
SMSZB-120 Firmware	20240319 08:08
SLZB-06 Firmware version:	20240710	v2.5.4.dev
I want to buy 7 addtional smoke detectors. I saw a few yt-videos and everyone was happy with the HA-Integration. I have to decide, if I should send the SMSZB-120 back.
Is there something I could do, to get it work. May be downgrade something to an earlier version? 
Are there some better smoke detectors, which could also be integrated into HA (starting and receiving alarms? which are not connected by default and costs not much more?)  Thx.

Schlammschlumpf, 2024-09-24:
Interesting. It seems to take some time. Meanwhile I have temperature, voltage and battery... Missing Reliability and Fault, but I don't know if it is okay

Koenkk, 2024-09-25:
Please try with const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const lumi = require('zigbee-herdsman-converters/lib/lumi');
const utils = require('zigbee-herdsman-converters/lib/utils');
const {logger} = require('zigbee-herdsman-converters/lib/logger');
const {develcoModernExtend} = require('zigbee-herdsman-converters/lib/develco');
const {battery, quirkAddEndpointCluster} = require('zigbee-herdsman-converters/lib/modernExtend');

const NS = 'extconverter';

const fzLocal = {
    fault_status: {
        cluster: 'genBinaryInput',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            if (msg.data.reliability !== undefined) {
                const lookup = {0: 'no_fault_detected', 7: 'unreliable_other', 8: 'process_error'};
                result.reliability = utils.getFromLookup(msg.data['reliability'], lookup);
            }
            if (msg.data.statusFlags !== undefined) {
                result.fault = msg.data['statusFlags'] === 1;
            }
            return result;
        },
    },
};
const definition = {
    zigbeeModel: ['SMSZB-120', 'GWA1512_SmokeSensor'],
    model: 'SMSZB-120',
    vendor: 'Develco',
    description: 'Smoke detector with siren CUSTOM',
    whiteLabel: [
        {vendor: 'Frient', model: '94430', description: 'Smart Intelligent Smoke Alarm'},
        {vendor: 'Cavius', model: '2103', description: 'RF SMOKE ALARM, 5 YEAR 65MM'},
    ],
    fromZigbee: [fz.ias_smoke_alarm_1_develco, fz.ignore_basic_report, fz.ias_enroll, fz.ias_wd, fzLocal.fault_status],
    toZigbee: [tz.warning, tz.ias_max_duration, tz.warning_simple],
    // ota: ota.zigbeeOTA,
    extend: [
        develcoModernExtend.addCustomClusterManuSpecificDevelcoGenBasic(),
        develcoModernExtend.readGenBasicPrimaryVersions(),
        develcoModernExtend.temperature(), // TODO: ep 38
        battery({
            voltageToPercentage: {min: 2500, max: 3000},
            percentage: true,
            voltage: true,
            lowStatus: false,
            voltageReporting: false,
            percentageReporting: false,
        }),
    ],
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(35);
        for (const cluster of ['ssIasZone', 'ssIasWd', 'genBasic', 'genBinaryInput', 'genPowerCfg']) {
            try {
                await endpoint.bind(cluster, coordinatorEndpoint);
            } catch (error) {
                if (error.message.includes('TABLE_FULL')) {
                    // Device throws TABLE_FULL errors even though bind works.
                    // https://github.com/Koenkk/zigbee2mqtt/issues/23684
                    logger.debug(`TABLE_FULL error while binding '${cluster}', ignoring...`, NS);
                } else {
                    throw error;
                }
            }
        }

        await reporting.batteryPercentageRemaining(endpoint);
        await reporting.batteryVoltage(endpoint);

        //     await reporting.bind(endpoint, coordinatorEndpoint, ['ssIasZone', 'ssIasWd', 'genBasic', 'genBinaryInput']);
        await endpoint.read('ssIasZone', ['iasCieAddr', 'zoneState', 'zoneId']);
        await endpoint.read('genBinaryInput', ['reliability', 'statusFlags']);
        //     await endpoint.read('ssIasWd', ['maxDuration']);
    },
    endpoint: (device) => {
        return {default: 35};
    },
    exposes: [
        e.smoke(),
        e.battery_low(),
        e.test(),
        e.numeric('max_duration', ea.ALL).withUnit('s').withValueMin(0).withValueMax(600).withDescription('Duration of Siren'),
        e.binary('alarm', ea.SET, 'START', 'OFF').withDescription('Manual Start of Siren'),
        e.enum('reliability', ea.STATE, ['no_fault_detected', 'unreliable_other', 'process_error']).withDescription('Indicates reason if any fault'),
        e.binary('fault', ea.STATE, true, false).withDescription('Indicates whether the device are in fault state'),
    ],
};

module.exports = definition;

peter4200, 2024-09-25:
The result is still the same, no changes have occurred.
I have "heard" from a Danish Home Assistant Facebook group that others who have installed firmware 4.0.8 work if the device is upgraded from firmware 4.0.7. Don't know if that could be a hint
[z2mlog.log](https://github.com/user-attachments/files/17136999/z2mlog.log)

peter4200, 2024-09-25:
An issue has been created at frient / Develco: [https://github.com/Develco-Products/frient_upgrade_images/issues/1](url)

Schlammschlumpf, 2024-09-25:
I have baught
3x Frient SMSZB-120 Firmware Date 20240319 08:08
1x SLZB-06 Firmware version:	radio: 20240710	core:v2.5.4.dev
Installed zigbee2mqtt at my HomeAssistant

After connecting the SMSZB-120 I got a lot of errors "TABLE_FULL" and more.

After a few hours temperature, battery are now part of my device:
{
    "battery": 100,
    "battery_low": false,
    "linkquality": 0,
    "restore_reports": true,
    "smoke": false,
    "supervision_reports": true,
    "temperature": 21,
    "test": false,
    "update": {
        "installed_version": 262152,
        "latest_version": 262152,
        "state": "idle"
    },
    "voltage": 3100
}

I installed the next one. After two hours:
{
    "battery_low": false,
    "linkquality": 72,
    "restore_reports": true,
    "smoke": false,
    "supervision_reports": true,
    "test": false,
    "update": {
        "installed_version": 262152,
        "latest_version": 262152,
        "state": "idle"
    }
}
I'm interested if battery, temperature & co will come tomorrow. Very strange

Still missing at my first device realbility, fault, ax duration at my first device. But I also installed a third device and this has also at the bginning no temperature, battery,... But now:

{
    "battery_low": false,
    "linkquality": 80,
    "restore_reports": true,
    "smoke": false,
    "supervision_reports": true,
    "test": false,
    "update": {
        "installed_version": 262152,
        "latest_version": 262152,
        "state": "idle"
    },
    "alarm": null,
    "battery": null,
    "fault": null,
    "max_duration": null,
    "reliability": null,
    "temperature": null,
    "update_available": null,
    "voltage": null
}

null but more status...

(I'm new to zigbee, if my post makes no sense, just ignore it) But it seems that time seems to solve a few issues, 

I'm very interested in a solution before I buy more of them. 

@Koenkk Would it help, if you you have such a device?

sarthurdev, 2024-09-26:
Received the following email from Frient:

```
There isn’t really any difference between 4.0.8 and 4.0.7 that would have an influence on binding the clusters.

I just tested it myself with Z2M, ConBee II, and our SMSZB-120 @ 4.0.8.

In fact, when adding the device, it gave me the same issues and errors (in logs) as the one you’re referring to in your references.

However, I re-interviewed it by using this:

[screenshot interview button] 

This resulted in more things being properly set up.

Then I re-interviewed it a second time, and I got the device successfully configured without any errors:

[screenshot showing all state data present]

What it seems like is that Z2M doesn’t wait until getting data from the device while interviewing, since the device is polling only within a couple of settings.

Manually triggering the interview seems to fix it – at least using ConBee II.

Regarding your request – we’re unfortunately not able to deliver an older version of the firmware, because of the security policy.
```

Schlammschlumpf, 2024-09-27:
@sarthurdev What exactly di you?
i hit this little green "i" multiple times. At the second time, I got no errors::

Info 2024-09-27 02:03:07z2m: Interviewing 'Rauchmelder Flur EG'
Info 2024-09-27 02:03:10z2m: Configuring 'Rauchmelder Flur EG'
Error 2024-09-27 02:03:11z2m: Failed to configure 'Rauchmelder Flur EG', attempt 1 (Error: Bind 0x0015bc0031024092/35 genBinaryInput from '0x00124b002f8e61a9/1' failed (ZDO error: bind failed with status 'TABLE_FULL' (140)) at /app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:1126:36 at processTicksAndRejections (node:internal/process/task_queues:95:5))
Info 2024-09-27 02:03:11z2m: Configuring 'Rauchmelder Flur EG'
Error 2024-09-27 02:03:11z2m: Failed to configure 'Rauchmelder Flur EG', attempt 2 (Error: Bind 0x0015bc0031024092/35 genBinaryInput from '0x00124b002f8e61a9/1' failed (ZDO error: bind failed with status 'TABLE_FULL' (140)) at /app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:1126:36 at processTicksAndRejections (node:internal/process/task_queues:95:5))
Info 2024-09-27 02:03:12z2m: Configuring 'Rauchmelder Flur EG'
Error 2024-09-27 02:03:12z2m: Failed to configure 'Rauchmelder Flur EG', attempt 3 (Error: Bind 0x0015bc0031024092/35 genBinaryInput from '0x00124b002f8e61a9/1' failed (ZDO error: bind failed with status 'TABLE_FULL' (140)) at /app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:1126:36 at processTicksAndRejections (node:internal/process/task_queues:95:5))
Info 2024-09-27 02:03:25z2m: Successfully interviewed 'Rauchmelder Flur EG'
Info 2024-09-27 02:03:25z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/interview', payload '{"data":{"id":"Rauchmelder Flur EG"},"status":"ok","transaction":"aqbhe-1"}'
Info 2024-09-27 02:04:42z2m: Interviewing 'Rauchmelder Flur EG'
Info 2024-09-27 02:04:58z2m: Successfully interviewed 'Rauchmelder Flur EG'
Info 2024-09-27 02:04:58z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/interview', payload '{"data":{"id":"Rauchmelder Flur EG"},"status":"ok","transaction":"aqbhe-2"}'
Info 2024-09-27 02:08:28z2m: Interviewing 'Rauchmelder Flur EG'
Info 2024-09-27 02:08:47z2m: Successfully interviewed 'Rauchmelder Flur EG'
Info 2024-09-27 02:08:47z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/interview', payload '{"data":{"id":"Rauchmelder Flur EG"},"status":"ok","transaction":"6efdm-1"}'
Info 2024-09-27 02:09:01z2m: Interviewing 'Rauchmelder Flur EG'
Info 2024-09-27 02:09:18z2m: Successfully interviewed 'Rauchmelder Flur EG'
Info 2024-09-27 02:09:18z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/interview', payload '{"data":{"id":"Rauchmelder Flur EG"},"status":"ok","transaction":"6efdm-2"}'

but nothing was changed. Still only (status page on z2m):
{
    "battery_low": false,
    "linkquality": 91,
    "restore_reports": true,
    "smoke": false,
    "supervision_reports": true,
    "test": false,
    "update": {
        "installed_version": 262152,
        "latest_version": 262152,
        "state": "idle"
    }
}

My first binded device shows a temperatur of 21°C. But the value didn't change since two days. Is your temperature sensor working?

What exactly did you, to get it work? Thx

sarthurdev, 2024-09-27:
As per their email I clicked the interview button a couple of times, letting it complete and waiting for the next device poll.

Now I've got all fields present:
```
{
    "battery": 80,
    "battery_low": false,
    "enrolled": true,
    "fault": false,
    "ias_cie_address": "0x00124b00258f1ae5",
    "linkquality": 51,
    "max_duration": 240,
    "reliability": "no_fault_detected",
    "restore_reports": true,
    "smoke": false,
    "supervision_reports": true,
    "temperature": 10.31,
    "test": false,
    "update": {
        "installed_version": 262152,
        "latest_version": 262152,
        "state": "idle"
    },
    "voltage": 2900,
    "zone_id": 23
}
```

Schlammschlumpf, 2024-09-28:
This don't work at my side.
3x SMSZB-120 Firmware Date 20240319 08:08
1x SLZB-06 Firmware version: radio: 20240710 core:v2.5.4.dev
zigbee2mqtt
I think about to send them all back (including SLZB-06) . A working solution seems not be possible in the near future. (only one of three shows a temperature sensor, but the value of the temperatur sensor didn't change the last days, so it also not working)

willdurand, 2024-10-06:
I've finally managed to get more data coming from these smoke detectors thanks to the external converter from: https://gist.github.com/Koenkk/3311969512f2e4307dcee8b6a22e1cbb. I am using a ZBDongle-E, and here is my z2m config:

<details>

```
Zigbee2MQTT version
    [1.40.2](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.40.2) commit: [e06848d](https://github.com/Koenkk/zigbee2mqtt/commit/e06848d)

Coordinator type
    EmberZNet

Coordinator revision
    7.4.3 [GA]

Coordinator IEEE Address
    0xe0798dfffebccd7d

Frontend version
    0.7.4

zigbee-herdsman-converters version
    20.21.0

zigbee-herdsman version
    2.1.3
```

</details>

Previously (i.e. with no custom config), `ias_cie_address`, `zone_id`, `enrolled` were missing. `reliability` is also set to something other than `null`. The `alarm` field is still `null`, though, which causes the toggle switch in the UI to be in an unknown state by default.

```json
{
    "battery": 100,
    "battery_low": false,
    "enrolled": true,
    "fault": false,
    "ias_cie_address": "0xe0798df111c0ffee",
    "last_seen": "2024-10-06T12:42:06.350Z",
    "linkquality": 132,
    "max_duration": 240,
    "reliability": "no_fault_detected",
    "restore_reports": true,
    "smoke": false,
    "supervision_reports": true,
    "temperature": 18.4,
    "test": false,
    "update": {
        "installed_version": 262152,
        "latest_version": 262152,
        "state": "idle"
    },
    "voltage": 3000,
    "zone_id": 23,
    "alarm": null
}
```

As expected with the external converter, there is no more OTA (because this is commented out). I _think_ temperatures get updated, though this was already working since I bound endpoint 38 to temperature.

mamrai1, 2024-10-20:
I think if you manually unbind clusters genpollcntr and genbinaryinput then reconfiguring the device will succeed …that was the only way I managed to fix my heat alarm

mamrai1, 2024-10-22:
> I think if you manually unbind clusters genpollcntr and genbinaryinput then reconfiguring the device will succeed …that was the only way I managed to fix my heat alarm to bind the powercfg cluster and reporting

Removing bind from genbinaryinput and genpollcntrl device configurie succeeds without any error.

mamrai1, 2024-10-25:
https://github.com/Koenkk/zigbee2mqtt/issues/22492#issuecomment-2436231653

Koenkk, 2024-10-25:
Fixed in https://github.com/Koenkk/zigbee-herdsman-converters/pull/8189

closed 2024-10-25 by Koenkk

Tevion92, 2024-10-31:
> Fixed in [Koenkk/zigbee-herdsman-converters#8189](https://github.com/Koenkk/zigbee-herdsman-converters/pull/8189)
> 
> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

please make this fix available for the public/normal Z2M Version!
I am a absolut beginner and would be very thankfull!

Herbs851, 2024-11-01:
Hi all,

Despite today's update ([push requests](https://github.com/Koenkk/zigbee-herdsman-converters/pull/8189)).

[image]

I still have the error : 

```
**2024-11-01 13:50:49z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x0015bc0034001883', payload '{"alarm":null,"battery":null,"battery_low":false,"enrolled":true,"fault":false,"ias_cie_address":"0x00124b002a2ec733","last_seen":"2024-11-01T12:50:50.042Z","linkquality":185,"max_duration":240,"reliability":"no_fault_detected","restore_reports":true,"smoke":false,"supervision_reports":true,"temperature":21.93,"test":false,"update":{"installed_version":-1,"latest_version":-1,"state":null},"update_available":null,"voltage":null,"zone_id":23}'**
```

[image]

[image]

[image]

Could this be due to the fact that it's a “Frient” model and not a “Develco” ?

mamrai1, 2024-11-01:
Which error?

Herbs851, 2024-11-01:
Sorry, i messed my copy-paste in my previous message :/ 

This one : 

```
2024-11-01 14:14:36z2m: Failed to configure 'HD-Cuisine', attempt 4 (Error: Bind 0x0015bc0034001883/35 genPowerCfg from '0x00124b002a2ec733/1' failed (Status 'TABLE_FULL') at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:566:23) at setupAttributes (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:160:13) at configure (/app/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:211:17) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:205:21) at Configure.configure (/app/lib/extension/configure.ts:130:13) at Configure.onMQTTMessage (/app/lib/extension/configure.ts:57:21) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:206:17))
```

mamrai1, 2024-11-01:
Reset the device and pair again or unbind the genoollctrl cluster and then press the configuation button

Herbs851, 2024-11-01:
I applied the second solution, and it worked.

Thanks a lot !!!

[10 more replies]