Koenkk/zigbee2mqtt #23688: Failed to start

fraizzo, 2024-08-21:
### What happened?

I read the installation guide but i cannot start the addon 

### Zigbee2MQTT version

1.39.1.1

### Adapter firmware version

not found

### Adapter

ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Setup

Dongle on x86 mini pc

### Debug log

Starting Zigbee2MQTT without watchdog.
[2024-08-21 18:30:32] info: 	z2m: Logging to console, file (filename: log.log)
[2024-08-21 18:30:32] info: 	z2m: Starting Zigbee2MQTT version 1.39.1 (commit #unknown)
[2024-08-21 18:30:32] info: 	z2m: Starting zigbee-herdsman (0.55.3)
[2024-08-21 18:30:32] info: 	zh:zstack:znp: Opening SerialPort with {"path":"/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_b01bd61f54afed119c203d4e71c9e8b5-if00-port0","baudRate":115200,"rtscts":false,"autoOpen":false}
[2024-08-21 18:30:32] info: 	zh:zstack:znp: Serialport opened
[2024-08-21 18:30:32] info: 	zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2024-08-21 18:30:34] info: 	zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2024-08-21 18:30:52] error: 	z2m: Error while starting zigbee-herdsman
[2024-08-21 18:30:52] error: 	z2m: Failed to start zigbee
[2024-08-21 18:30:52] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-08-21 18:30:52] error: 	z2m: Exiting...
[2024-08-21 18:30:52] error: 	z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:119:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:127:29)
    at Zigbee.start (/app/lib/zigbee.ts:63:27)
    at Controller.start (/app/lib/controller.ts:139:27)
    at start (/app/index.js:154:5)

RobyRWK, 2024-08-23:
hi, i have the same errors, unlike you i use SONOFF with TCP ZBBridge Pro but the error is the same. here are my logs:

[2024-08-23 16:52:15] info: 	z2m: Logging to console, file (filename: log.log)
[2024-08-23 16:52:15] info: 	z2m: Starting Zigbee2MQTT version 1.39.1 (commit #unknown)
[2024-08-23 16:52:15] info: 	z2m: Starting zigbee-herdsman (0.55.3)
[2024-08-23 16:52:15] info: 	zh:zstack:znp: Opening TCP socket with 10.100.201.53:8888
[2024-08-23 16:52:17] info: 	zh:zstack:znp: Socket connected
[2024-08-23 16:52:17] info: 	zh:zstack:znp: Socket ready
[2024-08-23 16:52:17] info: 	zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2024-08-23 16:52:18] info: 	zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2024-08-23 16:52:36] error: 	z2m: Error while starting zigbee-herdsman
[2024-08-23 16:52:36] error: 	z2m: Failed to start zigbee
[2024-08-23 16:52:36] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-08-23 16:52:36] error: 	z2m: Exiting...
[2024-08-23 16:52:36] error: 	z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:119:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:127:29)
    at Zigbee.start (/app/lib/zigbee.ts:63:27)
    at Controller.start (/app/lib/controller.ts:139:27)
    at start (/app/index.js:154:5)

can anyone help us?

TheGabeMan, 2024-08-31:
Same here. After update to 1.39.1 I haven't been able to get Zigbee2MQTT running again. I removed it completely, deleted all the files in /config/zigbee2mqtt and then reinstalled, but the issue remains. 

`[09:07:28] INFO: Preparing to start...
[09:07:28] INFO: Socat not enabled
[09:07:28] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2024-08-31 09:07:29] info: 	z2m: Logging to console, file (filename: log.log)
[2024-08-31 09:07:29] info: 	z2m: Starting Zigbee2MQTT version 1.39.1 (commit #unknown)
[2024-08-31 09:07:29] info: 	z2m: Starting zigbee-herdsman (0.55.3)
[2024-08-31 09:07:29] info: 	zh:zstack:znp: Opening TCP socket with 192.168.0.119:8888
[2024-08-31 09:07:29] info: 	zh:zstack:znp: Socket connected
[2024-08-31 09:07:29] info: 	zh:zstack:znp: Socket ready
[2024-08-31 09:07:29] info: 	zh:zstack:znp: Writing CC2530/CC2531 skip bootloader payload
[2024-08-31 09:07:30] info: 	zh:zstack:znp: Skip bootloader for CC2652/CC1352
[2024-08-31 09:07:48] error: 	z2m: Error while starting zigbee-herdsman
[2024-08-31 09:07:48] error: 	z2m: Failed to start zigbee
[2024-08-31 09:07:48] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-08-31 09:07:48] error: 	z2m: Exiting...
[2024-08-31 09:07:48] error: 	z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:119:27)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:127:29)
    at Zigbee.start (/app/lib/zigbee.ts:63:27)
    at Controller.start (/app/lib/controller.ts:139:27)
    at start (/app/index.js:154:5)`

fraizzo, 2024-08-31:
remove or disable ZHA, i resolve like this

TheGabeMan, 2024-08-31:
> remove or disable ZHA, i resolve like this

I have many devices still on ZHA, so I'd rather run them next to each other. And since my Zigbee bridges are not USB but wifi connected, it shouldn't be a problem I think??????

fraizzo, 2024-08-31:
> > remove or disable ZHA, i resolve like this
> 
> I have many devices still on ZHA, so I'd rather run them next to each other. And since my Zigbee bridges are not USB but wifi connected, it shouldn't be a problem I think??????

ZHA and Z2M can't go together, yo have to do a choice

RobyRWK, 2024-09-04:
I did not install ZHA. In my case I think it was a wifi problem because after a few days an access point broke, replaced with a new one and the problem did not reoccur.

closed 2024-12-22 by github-actions[bot]