Koenkk/zigbee2mqtt #23689: Nous P3Z lightbulb

mihaimdinca, 2024-08-21:
### What happened?

Nous P3Z lightbulb works but is really slow (from 2 to 15 seconds) to respond. The issue is not present when using the Smart Life app and a Moes ZHUB-W gateway.

### What did you expect to happen?

The lightbulb should execute a command (on/off, change color etc ) in about 1 second.

### How to reproduce it (minimal and precise)

Pair with Z2M hub and send commands through the Z2M interface or through Home Assistant.

### Zigbee2MQTT version

1.39.1

### Adapter firmware version

20230507

### Adapter

zStack3x0 Sonoff ZigBee Dongle P

### Setup

Home Assistant OS Addon on Raspberry Pi 5 8GB

### Debug log

2024-08-21 19:47:00z2m: Publish 'set' 'color' to 'Kitchen Light' failed: 'Error: ZCL command 0xa4c138c3924b8529/1 lightingColorCtrl.moveToColor({"transtime":0,"colorx":36202,"colory":26732}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 2882 - 1 - 142 - 768 - 11 after 10000ms)'

mihaimdinca, 2024-09-21:
Had to revert to firmware version 20221226 of the Sonoff ZigBee Dongle P for stability reasons. Anyway, the problem with the Nous P3Z still persists. It becomes especially obvious when sending multiple commands in a short period of time (change color multiple times, of the white temperature etc.)

mihaimdinca, 2024-09-29:
Apparently it works better if you enable color sync in settings (specific). Still becomes non responsive for a few (10ish) seconds after a number of color changes but seems to work better than with having color sync disabled.

JorinL, 2024-11-29:
Can tell the situation is still the same.

OrbitingOcelot, 2024-11-30:
I am seeing the bulb falling out of the network every now and then. Do you also experience this?

mihaimdinca, 2024-11-30:
Now i have given up on these bulbs. I remember them falling out from time to time. Overall a bad experience.

litecross91, 2024-12-01:
Me too. 

[2024-12-01 17:37:38] error: 	z2m: Publish 'set' 'brightness' to 'Desk bulb' failed: 'Error: ZCL command 0xa4c138679316a2c3/1 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":29001,"apsFrame":{"profileId":260,"clusterId":8,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":167},"zclSequence":156,"commandIdentifier":11} timed out after 10000ms)'

JorinL, 2024-12-02:
Yes gave up also... will return them.

Catastrophic..... even outside of ambilight — unresponsive.

Bought the first hue bulb ever for me.... works as wanted. Rest is ikea or wled.

mihaimdinca, 2024-12-02:
Went for Ikea - cheap and very responsive also good as routers. Not very bright with white dough. Woox also works well but is terrible as a router, at least for me

closed 2025-02-09 by github-actions[bot]

Daniel-WW, 2025-05-08:
I’m experiencing the same issue described here — sending `color_temp`, `hue`, or `saturation` to a smart bulb fails with a timeout:

```
lightingColorCtrl.moveToSaturation(...) failed (no response received (241)) Error: waiting for response TIMEOUT
```

The bulb otherwise works (on/off, brightness), but any attempt to change color properties leads to this error. The command is correctly formatted and sent via MQTT. The issue seems reproducible and persistent.

Is there any known workaround or fix for this? Could it be related to unsupported ZCL commands, firmware limitations, or missing bindings? A clarification or update would be appreciated.

Thanks in advance!