Koenkk/zigbee2mqtt #23691: [Feature request]: Nimly touch pro

Teeseeone, 2024-08-21:
### Is your feature request related to a problem? Please describe

Nimly is not included in supported devices.

### Describe the solution you'd like

I would be able to the same as Homey does it.
- open close lock
- activate/deactivate auto lock
- last lock
- sett new code
- clear new code.

### Describe alternatives you've considered

This is the door lock i have and its not supported.

### Additional context

https://nimly.no/product/touch-pro/

jkaberg, 2024-09-09:
Oh but it is supported...

One of my two Nimly Touch Pro Ultimates paired with zigbee2mqtt:
[image]

Teeseeone, 2024-09-09:
yes it works, i had a bad unit. Nimly sent me a new one! :)

Teeseeone, 2024-12-13:
Issue is not resolved, not supported yet.

gustavostmo, 2025-02-13:
Any update in this case?

jkaberg, 2025-02-13:
Typically for Nimly zigbee locks this can be fixed with adding the proper zigbeeModel to this list https://github.com/Koenkk/zigbee-herdsman-converters/blob/4ab9c5e424eaba39741175f008dd65749a528622/src/devices/onesti.ts#L103

closed 2025-02-13 by Teeseeone

gustavostmo, 2025-02-13:
@jkaberg It looks exciting. Where can I find a description of how to add this to my installation?