Koenkk/zigbee2mqtt #23695: [Feature request]: OWON PC 321 - Add clear measurement data

Lacrymal3, 2024-08-22:
### Is your feature request related to a problem? Please describe

Hello,

Sorry for my English.

I have a proposal for a modification to add a reset command for the counters of the OWON PC321 clamp meter.
[image]

Thank you in advance.

### Describe the solution you'd like

Add a push button.
[image]

### Describe alternatives you've considered

Attached is the modified code (you need to replace .txt with .js.):
[PC321.txt](https://github.com/user-attachments/files/16706833/PC321.txt)

The modifications :
```
const tzLocal = {
    PC321_clearMetering: {
        key: ['clear_metering'],
        convertSet: async (entity, key, value, meta) => {
            const endpoint = meta.device.getEndpoint(1);
            const group = 0xFFE0;
            const command = 0x00;
            const payload = {};

            await endpoint.command(group, command, payload, {disableDefaultResponse: true});
            meta.logger.info(`Sent clear command to ${entity.ieeeAddr}`);
        },
    },
};
```

`toZigbee: [tzLocal.PC321_clearMetering],`

`e.enum('clear_metering', ea.SET, ['Clear']).withDescription('Clear measurement data'),`

### Additional context

Datasheet (clear measurement data - sheet 12) :
[PC321 zha clusters and attributes v1.2.4.pdf](https://github.com/user-attachments/files/16706594/PC321.zha.clusters.and.attributes.v1.2.4.pdf)

Koenkk, 2024-08-22:
Looks good, could you make a pull request for this?

Lacrymal3, 2024-08-23:
Alright, i'm trying to make a pull request

Lacrymal3, 2024-08-23:
Pull request : 
https://github.com/Koenkk/zigbee-herdsman-converters/pull/7895

closed 2024-12-22 by github-actions[bot]