Koenkk/zigbee2mqtt #23700: Failed to interview

sebastienserre, 2024-08-22:
### What happened?

I'm adding a 6th Sonoff ZBMIniL2. The 5 others added some weeks/months ago are working correctly.

Z2M is recognizing it but can't finish pairing it with this error:
```
error 2024-08-22 12:57:25z2m: Request 'zigbee2mqtt/bridge/request/device/interview' failed with error: 'interview of 'Inter Exterieur entrée' (0x7cc6b6fffe0982a0) failed: Error: Interview failed because can not get active endpoints ('0x7cc6b6fffe0982a0')'
info 2024-08-22 12:57:25z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/interview', payload '{"data":{},"error":"interview of 'Inter Exterieur entrée' (0x7cc6b6fffe0982a0) failed: Error: Interview failed because can not get active endpoints ('0x7cc6b6fffe0982a0')","status":"error","transaction":"r8fwp-1"}'
```
Anybody can help me ?

### Zigbee2MQTT version

1.39.1

### Adapter firmware version

7.4.2 [GA]

### Adapter

EmberZNet - ZBT-1

### Setup

HA Green

### Debug log

error 2024-08-22 12:57:25z2m: Request 'zigbee2mqtt/bridge/request/device/interview' failed with error: 'interview of 'Inter Exterieur entrée' (0x7cc6b6fffe0982a0) failed: Error: Interview failed because can not get active endpoints ('0x7cc6b6fffe0982a0')'
info 2024-08-22 12:57:25z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/interview', payload '{"data":{},"error":"interview of 'Inter Exterieur entrée' (0x7cc6b6fffe0982a0) failed: Error: Interview failed because can not get active endpoints ('0x7cc6b6fffe0982a0')","status":"error","transaction":"r8fwp-1"}'

sibero80, 2024-08-24:
Exact same issue:
`Error 2024-08-23 22:23:59zh:controller: Interview failed for '0x842712fffe550620 with error 'Error: Interview failed because can not get active endpoints ('0x842712fffe550620')'
Error 2024-08-23 22:23:59z2m: Failed to interview '0x842712fffe550620', device has not successfully been paired
Info 2024-08-23 22:23:59z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"0x842712fffe550620"},"type":"pairing"}'
Info 2024-08-23 22:23:59z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x842712fffe550620","ieee_address":"0x842712fffe550620","status":"failed"},"type":"device_interview"}'`

sebastienserre, 2024-08-24:
@sibero80 ; I've fixed it by paring to a nearer router.

closed 2024-08-24 by sebastienserre