Koenkk/zigbee2mqtt #23703: External Converters cannot be completely deleted

dhc25, 2024-08-22:
### What happened?

I have been testing some local External Converters for dimmer switches.
After removing the External Converters from the file system, and also removing the reference to the files in the UI, the functionality (and product description) that was included in the converters is still present on the device.

I have confirmed that the reference to the files has accurately be reflected in the configuration file, but even when removing all traces of a local external converter from config file and file system, the converter still appears to be cached and used.

How do I revert back to the device functionality without using the local external converter?

Thank you

### What did you expect to happen?

Deleting an external converter would then show no signs of it's use, within the UI.

### Zigbee2MQTT version

1.39.1

### Adapter firmware version

20230507

### Adapter

Sonoff Zigbee Dongle

### Setup

Home Assistant add-on. Home Assistant running on proxmox.

msalte, 2024-08-26:
I have this problem too

martyn-vesternet, 2024-08-28:
@Koenkk I've noticed this while testing too, there doesn't seem to be a way to fully remove / delete a custom converter once it's been loaded.

Things I've tried:

1) Remove custom converter file from file system
2) Remove all traces of custom converter entry from configuration yaml
3) Replace custom converter file with empty definition {}
4) Remove all devices that were previously using the custom converter

It seems the converter is cached somewhere and although the cache copy is updated if the contents of the custom converter file changes locally (aside from the empty definition test mentioned above), the cached copy never seems to get purged in situations where it should.

IMO the cached copy should get purged in either of the following cases:

1) The local custom converter file is removed
2) The custom converter entry is removed from the configuration yaml

Koenkk, 2024-08-29:
In order to unload it, a restart of z2m is required.

Koenkk, 2024-08-30:
- Is it a total stop of z2m, or a restart via the z2m web ui?
- I've added some extra logging in the dev branch, can you check if `Loaded external converter` is logged even when there are no external converters in your `configuration.yaml`?

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-12-22 by github-actions[bot]

Oddiwahn, 2025-06-26:
I have the same problem, tried deleting the file, changing, rebooting, deleting the device, etc. the first version of the converter was still persistent. Now after waiting for an hour or so and then rebooting the new version was loaded