Koenkk/zigbee2mqtt #23704: SPLZB-141 OTA Update crashes z2m on ember + ezsp : "ASH_ERROR_TIMEOUTS"

senna1992, 2024-08-22:
### What happened?

W

### What did you expect to happen?

Update shall go through

### How to reproduce it (minimal and precise)

issue ota update on the device using z2m.

### Zigbee2MQTT version

 1.39.1

### Adapter firmware version

7.4.3 [GA]

### Adapter

HA yellow

### Setup

HA Yellow: z2m

### Debug log

`error 2024-08-22 14:41:35zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
error 2024-08-22 14:41:35zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
error 2024-08-22 14:41:35zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
error 2024-08-22 14:41:35zh:ember: !!! ADAPTER FATAL ERROR reason=HOST_FATAL_ERROR. !!!
error 2024-08-22 14:41:35zh:ember:uart:ash: ASH disconnected: ASH_ERROR_RESET_FAIL | Adapter status: ASH_NCP_FATAL_ERROR
error 2024-08-22 14:41:44zh:ember:ezsp: ERROR Transaction failure; status=ASH_ERROR_TIMEOUTS. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=36 Len=7].
error 2024-08-22 14:41:44zh:ember: !!! ADAPTER FATAL ERROR reason=ASH_ERROR_TIMEOUTS. !!!
error 2024-08-22 14:42:04zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
error 2024-08-22 14:42:04zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
error 2024-08-22 14:42:04zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
error 2024-08-22 14:42:04zh:ember: !!! ADAPTER FATAL ERROR reason=HOST_FATAL_ERROR. !!!
error 2024-08-22 14:42:04zh:ember:uart:ash: ASH disconnected: ASH_ERROR_RESET_FAIL | Adapter status: ASH_NCP_FATAL_ERROR
error 2024-08-22 14:42:12zh:ember:ezsp: ERROR Transaction failure; status=ASH_ERROR_TIMEOUTS. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=35 Len=7].
error 2024-08-22 14:42:12zh:ember: !!! ADAPTER FATAL ERROR reason=ASH_ERROR_TIMEOUTS. !!!
error 2024-08-22 14:42:33zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
error 2024-08-22 14:42:33zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
error 2024-08-22 14:42:33zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
error 2024-08-22 14:42:33zh:ember: !!! ADAPTER FATAL ERROR reason=HOST_FATAL_ERROR. !!!
error 2024-08-22 14:42:33zh:ember:uart:ash: ASH disconnected: ASH_ERROR_RESET_FAIL | Adapter status: ASH_NCP_FATAL_ERROR
error 2024-08-22 14:42:42zh:ember:ezsp: ERROR Transaction failure; status=ASH_ERROR_TIMEOUTS. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=38 Len=7].
error 2024-08-22 14:42:42zh:ember: !!! ADAPTER FATAL ERROR reason=ASH_ERROR_TIMEOUTS. !!!
error 2024-08-22 14:43:21zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
error 2024-08-22 14:43:21zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
error 2024-08-22 14:43:21zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
error 2024-08-22 14:43:21zh:ember: !!! ADAPTER FATAL ERROR reason=HOST_FATAL_ERROR. !!!
error 2024-08-22 14:43:21zh:ember:uart:ash: ASH disconnected: ASH_ERROR_RESET_FAIL | Adapter status: ASH_NCP_FATAL_ERROR
error 2024-08-22 14:43:30zh:ember:ezsp: ERROR Transaction failure; status=ASH_ERROR_TIMEOUTS. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=38 Len=7].
error 2024-08-22 14:43:30zh:ember: !!! ADAPTER FATAL ERROR reason=ASH_ERROR_TIMEOUTS. !!!
error 2024-08-22 14:43:55zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
error 2024-08-22 14:43:55zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
error 2024-08-22 14:43:55zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
error 2024-08-22 14:43:55zh:ember: !!! ADAPTER FATAL ERROR reason=HOST_FATAL_ERROR. !!!
error 2024-08-22 14:43:55zh:ember:uart:ash: ASH disconnected: ASH_ERROR_RESET_FAIL | Adapter status: ASH_NCP_FATAL_ERROR
error 2024-08-22 14:44:03zh:ember:ezsp: ERROR Transaction failure; status=ASH_ERROR_TIMEOUTS. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=40 Len=7].
error 2024-08-22 14:44:03zh:ember: !!! ADAPTER FATAL ERROR reason=ASH_ERROR_TIMEOUTS. !!!`

senna1992, 2024-08-22:
Have to add: all 3 of the smart plugs updated even tho the controller has been crashing every 30s or so ❓

waroen, 2024-08-23:
Have the same exact error, my Skyconnect crashes repeatedly during OTA with either ASH_ERROR_TIMEOUTS or ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.

The frequecy varies, 30s to a few minutes.

senna1992, 2024-08-23:
> Have the same exact error, my Skyconnect crashes repeatedly during OTA with either ASH_ERROR_TIMEOUTS or ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
> 
> The frequecy varies, 30s to a few minutes.

Does it succeed doing the ota despite the errors as it did for me?

bjornarfjelldal, 2024-09-02:
Same experience here, for the past month or so. Crashes the HA addon, and the only way to get it back to a working state is to hard reset the bare metal (restart physical machine running proxmox and HA VM).

[log.log](https://github.com/user-attachments/files/16834530/log.log)
Attached is a debug log of it happening.

Nerivec, 2024-09-05:
Any chance one of you can try with one of the firmware described in [this post](https://github.com/Koenkk/zigbee2mqtt/discussions/23224#discussioncomment-10422347) and 1.40.0?

waroen, 2024-09-09:
Updated to the max version for my sjyconnect. Have since been able to update two IKEA switches without much problem. I'll go through the logs so make sure that I haven't missed any errors but it seems to be better.

bjornarfjelldal, 2024-12-13:
Can confirm this helped, @Nerivec

closed 2024-12-13 by Nerivec

Nerivec, 2025-07-01:
I advise you to stay on 8.0.2 for the moment. It is widely used by Z2M ember users, and has shown no major problem.
You can however update to the latest 8.0.2 bugfix release: https://github.com/Nerivec/silabs-firmware-builder/releases/tag/v2024.6.2-update7

8.1.x and 8.2.x have shown troubles (I've pre-released some 8.2.0 builds, but there are still issues with flow control to be solved on some models, and this brand new version is widely untested in general).

Nerivec, 2025-07-01:
zigbee_ncp => coordinator
zigbee_router => use an adapter as a dedicated router in the network
openthread_rcp => use in OpenThread applications or with [ZigBee on Host](https://github.com/Koenkk/zigbee2mqtt/discussions/26746)

Note: if you use Ember ZLI for flashing, depending on what model you select when entering the bootloader, it will automatically pick coordinator or router firmware if you then use the pre-defined lists.

bjornarfjelldal, 2025-07-01:
- Thanks! I did it through the Skyconnect web tool. Is it supposed to say 8.0.2 [GA] still, by the way? This is from the raw after update/reboot:
```

    },
    "coordinator": {
        "ieee_address": "redacted",
        "meta": {
            "build": 397,
            "ezsp": 14,
            "major": 8,
            "minor": 0,
            "patch": 2,
            "revision": "8.0.2 [GA]",
            "special": 0,
            "type": 170
        },
        "type": "EmberZNet"
```