Koenkk/zigbee2mqtt #23705: [New device support]: TZE204_ex3rcdha presence and lux sensor

djdevil, 2024-08-22:
### Link

https://it.aliexpress.com/item/1005007335061248.html?gatewayAdapt=glo2ita

### Database entry

{"id":54,"type":"Router","ieeeAddr":"0xa4c13887abb15e5d","nwkAddr":37763,"manufId":4417,"manufName":"_TZE204_ex3rcdha","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1724348619720}

### Comments

I purchased this new sensor, but even by modifying the converter it doesn't seem to work, is there anyone who can help me, thanks

### External definition

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const tzDatapoints = {
    ...tuya.tz.datapoints,
    key: [...tuya.tz.datapoints.key, 'minimum_range', 'maximum_range', 'detection_delay', 'fading_time']
}

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE204_ex3rcdha']),
    model: 'TS0601_smart_human_presence_sensor',
    vendor: 'TuYa',
    description: 'Smart Human presence sensor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tzDatapoints],
    exposes: [
        e.illuminance_lux(), e.presence(),
        exposes.numeric('target_distance', ea.STATE).withDescription('Distance to target').withUnit('m'),
        exposes.numeric('radar_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(9).withValueStep(1)
            .withDescription('sensitivity of the radar'),
        exposes.numeric('minimum_range', ea.STATE_SET).withValueMin(0).withValueMax(9.5).withValueStep(0.15)
            .withDescription('Minimum range').withUnit('m'),
        exposes.numeric('maximum_range', ea.STATE_SET).withValueMin(0).withValueMax(9.5).withValueStep(0.15)
            .withDescription('Maximum range').withUnit('m'),
        exposes.numeric('detection_delay', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(0.1)
            .withDescription('Detection delay').withUnit('s'),
        exposes.numeric('fading_time', ea.STATE_SET).withValueMin(0.5).withValueMax(1500).withValueStep(1)
            .withDescription('Fading time').withUnit('s'),
    ],
    meta: {
        tuyaDatapoints: [
            [104, 'illuminance_lux', tuya.valueConverter.raw],
            [105, 'presence', tuya.valueConverter.trueFalse1],
            [106, 'radar_sensitivity', tuya.valueConverter.raw],
            [107, 'maximum_range', tuya.valueConverter.divideBy100],
            [108, 'minimum_range', tuya.valueConverter.divideBy100],
            [109, 'target_distance', tuya.valueConverter.divideBy100],
            [110, 'fading_time', tuya.valueConverter.divideBy10],
            [111, 'detection_delay', tuya.valueConverter.divideBy10],
        ],
    }
};

module.exports = definition;
```

seoeaa, 2024-08-22:
same problem

draserver, 2024-08-27:
And me too

therealchunkyd, 2024-08-29:
Also following this one

BeastyBoy339, 2024-08-29:
Following

DougD1991, 2024-08-30:
Following

Yub0, 2024-09-04:
i have the same presence detector too, following

Koenkk, 2024-09-05:
In order to add support for this Tuya device, we need to figure out what datapoints it uses.
See [this guide](https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html) on how to do that

fabianteunissen, 2024-09-09:
Not able to figure out the datapoints, but following!

tomas123, 2024-09-11:
I found here the datapoints for the ZY-HPS01-5.8G
https://github.com/make-all/tuya-local/blob/main/custom_components/tuya_local/devices/zyhps01_human_presence_sensor.yaml

motezuki, 2024-09-11:
Got this device working with illuminance and presence states, other parameters can be set but not read at first

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
//    zigbeeModel: ['TS0601'],
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE204_ex3rcdha',
        },
    ],
    model: 'TS0601_?',
    vendor: 'Tuya',
    description: 'mmWave radar ZY_HPS01 V1.2',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    extend: [    ],
    exposes: [
        e.illuminance_lux(),
        e.presence(),
        exposes.numeric('presence_timeout', ea.STATE_SET).withValueMin(0).withValueMax(180).withValueStep(1)
            .withDescription('Presence timeout').withUnit('s'),
        exposes.numeric('move_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(1)
           .withDescription('sensitivity of the radar').withUnit('X'),
        exposes.numeric('move_minimum_range', ea.STATE_SET).withValueMin(0).withValueMax(600).withValueStep(10)
            .withDescription('Movement minimum range').withUnit('cm'),
        exposes.numeric('move_maximum_range', ea.STATE_SET).withValueMin(0).withValueMax(600).withValueStep(10)
            .withDescription('Movement maximum range').withUnit('cm'),
        exposes.numeric('breath_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(1)
            .withDescription('Breath sensitivity of the radar').withUnit('X'),
        exposes.numeric('breath_minimum_range', ea.STATE_SET).withValueMin(0).withValueMax(600).withValueStep(10)
            .withDescription('Breath minimum range').withUnit('cm'),
        exposes.numeric('breath_maximum_range', ea.STATE_SET).withValueMin(0).withValueMax(600).withValueStep(10)
            .withDescription('Breath maximum range').withUnit('cm'),

    ],
    meta: {
        tuyaDatapoints: [
            [12, 'illuminance_lux', tuya.valueConverter.raw],
            [101, 'presence', tuya.valueConverter.trueFalse0],
            [104, 'presence_timeout', tuya.valueConverter.raw],
            [105, 'movesensitivity', tuya.valueConverter.raw],
            [107, 'breath_sensitivity', tuya.valueConverter.raw],
            [109, 'move_maximum_range', tuya.valueConverter.raw],
            [110, 'move_minimum_range', tuya.valueConverter.raw],
            [111, 'breath_maximum_range', tuya.valueConverter.raw],
            [112, 'breath_minimum_range', tuya.valueConverter.raw],

        ],
    }
};
```

djdevil, 2024-09-11:
> Got this device working with illuminance and presence states, other parameters can be set but not read at first
> 
> ```
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const reporting = require('zigbee-herdsman-converters/lib/reporting');
> const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
> const e = exposes.presets;
> const ea = exposes.access;
> const tuya = require('zigbee-herdsman-converters/lib/tuya');
> 
> const definition = {
>     fingerprint: [
>         {
>             modelID: 'TS0601',
>             manufacturerName: '_TZE204_ex3rcdha',
>         },
>     ],
>     model: 'TS0601_?',
>     vendor: 'Tuya',
>     description: 'mmWave radar ZY_HPS01 V1.2',
>     fromZigbee: [tuya.fz.datapoints],
>     toZigbee: [tuya.tz.datapoints],
>     configure: tuya.configureMagicPacket,
>     extend: [    ],
>     exposes: [
>         e.illuminance_lux(),
>         e.presence(),
>         exposes.numeric('presence_timeout', ea.STATE_SET).withValueMin(0).withValueMax(180).withValueStep(1)
>             .withDescription('Presence timeout').withUnit('s'),
>         exposes.numeric('move_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(1)
>            .withDescription('sensitivity of the radar').withUnit('X'),
>         exposes.numeric('move_minimum_range', ea.STATE_SET).withValueMin(0).withValueMax(600).withValueStep(10)
>             .withDescription('Movement minimum range').withUnit('cm'),
>         exposes.numeric('move_maximum_range', ea.STATE_SET).withValueMin(0).withValueMax(600).withValueStep(10)
>             .withDescription('Movement maximum range').withUnit('cm'),
>         exposes.numeric('breath_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(1)
>             .withDescription('Breath sensitivity of the radar').withUnit('X'),
>         exposes.numeric('breath_minimum_range', ea.STATE_SET).withValueMin(0).withValueMax(600).withValueStep(10)
>             .withDescription('Breath minimum range').withUnit('cm'),
>         exposes.numeric('breath_maximum_range', ea.STATE_SET).withValueMin(0).withValueMax(600).withValueStep(10)
>             .withDescription('Breath maximum range').withUnit('cm'),
> 
>     ],
>     meta: {
>         tuyaDatapoints: [
>             [12, 'illuminance_lux', tuya.valueConverter.raw],
>             [101, 'presence', tuya.valueConverter.trueFalse0],
>             [104, 'presence_timeout', tuya.valueConverter.raw],
>             [105, 'movesensitivity', tuya.valueConverter.raw],
>             [107, 'breath_sensitivity', tuya.valueConverter.raw],
>             [109, 'move_maximum_range', tuya.valueConverter.raw],
>             [110, 'move_minimum_range', tuya.valueConverter.raw],
>             [111, 'breath_maximum_range', tuya.valueConverter.raw],
>             [112, 'breath_minimum_range', tuya.valueConverter.raw],
> 
>         ],
>     }
> };
> ```

the code gives me a syntax error and doesn't seem to function, as if we are missing 

data const tzDatapoints = {
    ...tuya.tz.datapoints,
    key: Array.isArray(tuya.tz.datapoints.key)
        ? [...tuya.tz.datapoints.key, 'minimum_range', 'maximum_range', 'detection_delay', 'fading_time']
        : ['minimum_range', 'maximum_range', 'detection_delay', 'fading_time']
};

motezuki, 2024-09-12:
Sorry I missed one edition on tuyaDatapoints

where is:
[105, 'movesensitivity', tuya.valueConverter.raw],

should be:
[105, 'move_sensitivity', tuya.valueConverter.raw],

I will fix on the post

tomas123, 2024-09-12:
your code give me a syntax error
```
[2024-09-12 11:38:04] error: 	z2m: Failed to load external converter file 'ZY_HPS01.js' (Cannot read properties of undefined (reading 'push'))

[2024-09-12 11:38:04] error: 	z2m: Probably there is a syntax error in the file or the external converter is not compatible with the current Zigbee2MQTT version
```

I used your code with the snippet from
https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html
and now the sensor works fine
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE204_ex3rcdha',
        },
    ],
    model: 'ZY-HPS01-5.8G',
    vendor: 'Tuya',
    description: 'mmWave radar ZY_HPS01 5.8GHz',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        // Here you should put all functionality that your device exposes
        e.illuminance_lux(),
        e.presence(),
        exposes.numeric('presence_timeout', ea.STATE_SET).withValueMin(0).withValueMax(180).withValueStep(1)
            .withDescription('Presence timeout').withUnit('s'),
        exposes.numeric('move_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(1)
           .withDescription('sensitivity of the radar').withUnit('X'),
        exposes.numeric('move_minimum_range', ea.STATE_SET).withValueMin(0).withValueMax(600).withValueStep(10)
            .withDescription('Movement minimum range').withUnit('cm'),
        exposes.numeric('move_maximum_range', ea.STATE_SET).withValueMin(0).withValueMax(600).withValueStep(10)
            .withDescription('Movement maximum range').withUnit('cm'),
        exposes.numeric('breath_sensitivity', ea.STATE_SET).withValueMin(0).withValueMax(10).withValueStep(1)
            .withDescription('Breath sensitivity of the radar').withUnit('X'),
        exposes.numeric('breath_minimum_range', ea.STATE_SET).withValueMin(0).withValueMax(600).withValueStep(10)
            .withDescription('Breath minimum range').withUnit('cm'),
        exposes.numeric('breath_maximum_range', ea.STATE_SET).withValueMin(0).withValueMax(600).withValueStep(10)
            .withDescription('Breath maximum range').withUnit('cm'),

    ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
            [12, 'illuminance_lux', tuya.valueConverter.raw],
            [101, 'presence', tuya.valueConverter.trueFalse0],
            [104, 'presence_timeout', tuya.valueConverter.raw],
            [105, 'move_sensitivity', tuya.valueConverter.raw],
            [107, 'breath_sensitivity', tuya.valueConverter.raw],
            [109, 'move_maximum_range', tuya.valueConverter.raw],
            [110, 'move_minimum_range', tuya.valueConverter.raw],
            [111, 'breath_maximum_range', tuya.valueConverter.raw],
            [112, 'breath_minimum_range', tuya.valueConverter.raw],
        ],
    },
    extend: [
        // A preferred new way of extending functionality.
    ],
};

module.exports = definition;

```

tomas123, 2024-09-12:
I add this line for an external *.js script
`module.exports = definition;`

and now the code works fine!

Koenkk, 2024-09-13:
@motezuki could you submit a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts)?

Javisen, 2024-09-17:
@motezuki Thank you very much for the work. It works fine for me. Is there any way to change the icon?

motezuki, 2024-09-18:
@Javisen maybe you have a slightly different device, with a different data point for illuminance.
Could you verify the datapoints using [these instructions](https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html)?

The icons can be changed directly on the configuration of the entity card

Javisen, 2024-09-18:
> @Javisen maybe you have a slightly different device, with a different data point for illuminance. Could you verify the datapoints using [these [...]

I think it might be the device. When you bring a flashlight close to the sensor you get lux values but very low (25 lux). Also, I don't know how I can find the device on yours if I'm using it locally.

BeastyBoy339, 2024-09-21:
Thanks for al the work.
It works but cant use the distance settings, getting an error like this "z2m: No converter available for 'move_minimum_range' (340)"

dheerajgudipati, 2024-09-25:
Occupancy not working 

Occupancy
Indicates whether the device detected occupancy
Null

paulsijben, 2024-11-18:
following.  I was looking at my TS0601 quirk, and I do not see how I can add this code in.

Madmorgan007, 2024-12-08:
Hi I'm new to home assistant how do I add this fix to my system ??

pipiche38, 2024-12-14:
do you know what breath_maximum_range',  'breath_minimum_range mean ?

closed 2025-02-21 by github-actions[bot]