Koenkk/zigbee2mqtt #23706: Error when sending a multicaast frame: ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT

borgmatthew, 2024-08-22:
### What happened?

I have multiple zigbee bulbs which are grouped in zigbee groups. These are turned on/off according to motion detected by a PIR sensor. Occasionally (more than once in a day), when a sensor is triggered, I getting the following error: `Received ERROR from NCP, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT`, which is causing a restart.

This is causing some events from the PIRs to be missed, since the adapter is restarting. I tried to upgrade to `1.39.1` but this seemed to be worse, it was occurring much more frequently, so I reverted to `1.39.0`.

Could this be related to weak zigbee signal?

### What did you expect to happen?

The error is handled without restarting

### Zigbee2MQTT version

1.39.0

### Adapter firmware version

7.4.3 [GA]

### Adapter

Sonoff Dongle E, using ember driver

### Setup

Z2M on Docker, on Raspberry pi 5

### Debug log

[zigbee-crash-log.txt](https://github.com/user-attachments/files/16716143/zigbee-crash-log.txt)

Nerivec, 2024-08-26:
If this is signal/interference related, it's possible dev branch already fixes the issue.
Can you confirm the issue is still present with latest-dev? (Or after September 1st release).

borgmatthew, 2024-08-31:
Updated to the latest dev, the error is slightly different as it does not mention the "transaction failure", but the adapter was disconnected and restarted.
[zigbee-crash-log-dev.txt](https://github.com/user-attachments/files/16825458/zigbee-crash-log-dev.txt)

Nerivec, 2024-10-01:
Can you confirm the issue is still present after updating firmware to 7.4.4?

borgmatthew, 2024-10-03:
I've updated to `7.4.4` 4 days ago and didn't have a single reset ever since! :D This is now resolved.

closed 2024-10-03 by borgmatthew