Koenkk/zigbee2mqtt #23707: Zigbee2MQTT doesn't allow changing dongle IP address

canoben, 2024-08-22:
### What happened?

I changed the IP address of my SLZB-06 dongle (from x.x.x.218 to x.x.x.141). No matter what I try (restarting the plugin, restarting Home Assistant, reinstalling the plugin), Zigbee2MQTT refuses to use the new address.

Here is my configuration:

[image]

and here is the log spewing an error about no connection:

[image]

I'm at a loss here what to do.

### What did you expect to happen?

I expected Zigbee2MQTT to connect with the new address.

### Zigbee2MQTT version

1.39.1-1

### Adapter firmware version

Zigbee coordinator 20240510

### Adapter

SLZB-06M

### Setup

Virtual machine on Synology NAS

### Debug log

[22:16:08] INFO: Preparing to start...
[22:16:08] INFO: Socat not enabled
[22:16:11] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2024-08-22 22:16:21] info: 	z2m: Logging to console, file (filename: log.log)
[2024-08-22 22:16:21] info: 	z2m: Starting Zigbee2MQTT version 1.39.1 (commit #unknown)
[2024-08-22 22:16:21] info: 	z2m: Starting zigbee-herdsman (0.55.3)
[2024-08-22 22:16:22] warning: 	zh:ezsp: Deprecated driver 'ezsp' currently in use, 'ember' will become the officially supported EmberZNet driver in next release. If using Zigbee2MQTT see https://github.com/Koenkk/zigbee2mqtt/discussions/21462
[2024-08-22 22:16:25] error: 	zh:ezsp:ezsp: Connection attempt 1 error: Error: connect EHOSTUNREACH 192.168.1.218:6638
    at TCPConnectWrap.afterConnect [as oncomplete] (node:net:1555:16)
[2024-08-22 22:16:33] error: 	zh:ezsp:ezsp: Connection attempt 2 error: Error: connect EHOSTUNREACH 192.168.1.218:6638
    at TCPConnectWrap.afterConnect [as oncomplete] (node:net:1555:16)
[2024-08-22 22:16:46] error: 	zh:ezsp:ezsp: Connection attempt 3 error: Error: connect EHOSTUNREACH 192.168.1.218:6638
    at TCPConnectWrap.afterConnect [as oncomplete] (node:net:1555:16)
[2024-08-22 22:17:04] error: 	zh:ezsp:ezsp: Connection attempt 4 error: Error: connect EHOSTUNREACH 192.168.1.218:6638
    at TCPConnectWrap.afterConnect [as oncomplete] (node:net:1555:16)
[2024-08-22 22:17:04] error: 	z2m: Error while starting zigbee-herdsman
[2024-08-22 22:17:04] error: 	z2m: Failed to start zigbee
[2024-08-22 22:17:04] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-08-22 22:17:04] error: 	z2m: Exiting...
[2024-08-22 22:17:05] error: 	z2m: Error: Failure to connect
    at Ezsp.connect (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:364:19)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
    at Driver.startup (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/driver.ts:176:13)
    at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:127:29)
    at Zigbee.start (/app/lib/zigbee.ts:63:27)
    at Controller.start (/app/lib/controller.ts:139:27)
    at start (/app/index.js:154:5)

nblount, 2024-10-04:
Have a similar issue but mine is I can not change from a USB Sonoff to a SLZB-06 on the network. I set the IP address either in the configuration file or via settings in the application and it is ignored and defaults back to USB. I am on version 1.40.2

closed 2024-12-22 by github-actions[bot]

Smith166, 2025-04-29:
Was there any solution, please?
I can't change the IP. I found the configuration.yaml, I have changed, but it's not working, after restarts it's always changing back to old IP.

Smith166, 2025-04-29:
Thank you. I just find it out before read you post, but thank you very much.

I just setup completely a new network infrastructure with separate VLANs, subnets, DHCPs, and slowly moving everything to the new subnets, and today I have changed the IP of HAOS and it caused a chaos, it took almost a day the fight with Z2M 😀

It1s a big journey to move everything but it's also refreshed my mind regarding the hidden places of configs and IPs.