Koenkk/zigbee2mqtt #23711: seMetering currentSummDelivered Reporting Configuration behaviour

martyn-vesternet, 2024-08-23:
### What happened?

@Koenkk what is the correct format for configuration in the currentSummDelivered attribute on the seMetering cluster?

In https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/lib/modernExtend.ts I see this:

```
seMetering: {
            // Report change with every 5W change
            power: {attribute: 'instantaneousDemand', divisor: 'divisor', multiplier: 'multiplier', forced: args.power, change: 5},
            // Report change with every 0.1kWh change
            energy: {attribute: 'currentSummDelivered', divisor: 'divisor', multiplier: 'multiplier', forced: args.energy, change: 0.1},
            // produced_energy: {attribute: 'currentSummReceived', divisor: 'divisor', multiplier: 'multiplier', forced: args.energy, change: 0.1},
        },
```

Which suggests "0.1" format should be used and will equal 0.1kWh.

However some herdsman converters are using this format:

`await reporting.currentSummDelivered(endpoint, {min: 1, max: 600, change: [0, 5]});`

And some herdsman converters are using this format:

`await reporting.currentSummDelivered(endpoint, {min: 10, max: 3600, change: 5});`

In my testing, the divisor and multiplier are being read correctly from the device:

[image]

So this particular device counts 3,600,000 per kWh.

Something is amiss somewhere.

If I use this in an external converter:

`await reporting.currentSummDelivered(endpoint11, {min: 10, change: 0.1, max: 600});`

Then the result is this in the UI:

[image]

And 0 being sent to the device during pairing:

[image]

On the other hand, if I use this in an external converter:

`await reporting.currentSummDelivered(endpoint11, {min: 10, change: [0, 1], max: 600});`

Then the result is this in the UI:

[image]

And 1 being sent to the device during pairing:

[image]

Or if I use this in an external converter:

`await reporting.currentSummDelivered(endpoint11, {min: 10, change: [0, 360000], max: 600});`

This is the UI:

[image]

And this is sent to the device during pairing:

[image]

If I use this in the external converter:

`await reporting.currentSummDelivered(endpoint11, {min: 10, change: 3600000, max: 600});`

Then this is the result in the UI:

[image]

But again, 0 is being sent to the device during pairing:

[image]

If all of these formats *should* be acceptable:

0.1
[0,1]
360000

Then presumably they should all produce the correct end result being sent to the device?

Finally, if I make a manual entry in the UI with any value, always 0 is sent to the device.

1000:

[image]

0 is sent to the device:

[image]

0.5:

[image]

Again 0 to the device:

[image]

3600000:

[image]

Again 0 to the device:

[image]

Or even 1:

[image]

Again 0 to the device:

[image]

### Zigbee2MQTT version

1.39.1

### Adapter firmware version

6.10.3.0 build 297

### Adapter

 EZSP v8

### Setup

Home Assistant Supervised Z2M addon

Koenkk, 2024-08-24:
In modernExtend, the `change` for `currentSummDelivered` is converted here (according to divisor/multiplier):  https://github.com/Koenkk/zigbee-herdsman-converters/blob/8d7f1c4aa7c253646fef7603bb1e07a307d64349/src/lib/modernExtend.ts#L1680

- 0.1 -> should not be supported, only ints
- [0,1] -> ideally this should also not be supported (but a nodejs number cannot fit a uint48 afaik)
360000 -> only this should be supported

Koenkk, 2024-08-29:
It's indeed wrong, I will put it on my backlog

closed 2024-09-17 by Koenkk

Koenkk, 2024-09-17:
Fixed now! Removed the `[number, number]`, it's always a `number` now.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.