Koenkk/zigbee2mqtt #23713: [Feature request]: Cover state should inclide STATE_OPENING/STATE_CLOSING

Xyaren, 2024-08-23:
### Is your feature request related to a problem? Please describe

I would like to have my covers to show an animation on the custom card https://github.com/Deejayfool/hass-shutter-card.
The card is using the (official) cover states "opening"/"closing" to identify if the cover is currently moving.

My z2m mqtt covers are not providing the moving state in the state of the cover entity, instead they provide additional sensors for the moving state via auto discovery.

### Describe the solution you'd like

Implement STATE_OPENING + STATE_CLOSING (see https://developers.home-assistant.io/docs/core/entity/cover/#states) for cover states additionally or instead of the separate sensors.

### Describe alternatives you've considered

Custom sensors in ha.. (not very keen of doing all of that)

### Additional context

Additionally the state of the cover is currently uppercase despite the HA constatns using lowercase values.

Xyaren, 2024-12-12:
Still an issue

Xyaren, 2024-12-17:
Found out the issue:

The autoconfiguration state template is wrong:

Current 
```json
{
  "availability": [
    {
      "topic": "zigbee2mqtt/bridge/state",
      "value_template": "{{ value_json.state }}"
    },
    {
      "topic": "zigbee2mqtt/Rolläden Wohnzimmer/availability",
      "value_template": "{{ value_json.state }}"
    }
  ],
  "availability_mode": "all",
  "command_topic": "zigbee2mqtt/Rolläden Wohnzimmer/left/set",
  "device": {
    "configuration_url": "https://zigbee2mqtt.mydomain.de/#/device/0xa4c138018cf95021/info",
    "identifiers": [
      "zigbee2mqtt_0xa4c138018cf95021"
    ],
    "manufacturer": "Lonsonho",
    "model": "Dual curtain/blind module (TS130F_dual)",
    "name": "Rolläden Wohnzimmer",
    "suggested_area": "Wohnzimmer",
    "via_device": "zigbee2mqtt_bridge_0xe0798dfffec7b47c"
  },
  "device_calss": "shutter",
  "name": "Left",
  "object_id": "rolläden_wohnzimmer_left",
  "origin": {
    "name": "Zigbee2MQTT",
    "sw": "1.42.0",
    "url": "https://www.zigbee2mqtt.io"
  },
  "position_template": "{{ value_json.position }}",
  "position_topic": "zigbee2mqtt/Rolläden Wohnzimmer/left",
  "set_position_template": "{ \"position_left\": {{ position }} }",
  "set_position_topic": "zigbee2mqtt/Rolläden Wohnzimmer/left/set",
  "state_closing": "DOWN",
  "state_opening": "UP",
  "state_stopped": "STOP",
  "state_topic": "zigbee2mqtt/Rolläden Wohnzimmer/left",
  "unique_id": "0xa4c138018cf95021_cover_left_zigbee2mqtt",
  "value_template": "{% if \"moving_left\" in value_json and value_json.moving_left %} {{ value_json.moving_left }} {% else %} STOP {% endif %}"
}
```

Correct JSON (last line changed)
```json
{
  "availability": [
    {
      "topic": "zigbee2mqtt/bridge/state",
      "value_template": "{{ value_json.state }}"
    },
    {
      "topic": "zigbee2mqtt/Rolläden Wohnzimmer/availability",
      "value_template": "{{ value_json.state }}"
    }
  ],
  "availability_mode": "all",
  "command_topic": "zigbee2mqtt/Rolläden Wohnzimmer/left/set",
  "device": {
    "configuration_url": "https://zigbee2mqtt.mydomain.de/#/device/0xa4c138018cf95021/info",
    "identifiers": [
      "zigbee2mqtt_0xa4c138018cf95021"
    ],
    "manufacturer": "Lonsonho",
    "model": "Dual curtain/blind module (TS130F_dual)",
    "name": "Rolläden Wohnzimmer",
    "suggested_area": "Wohnzimmer",
    "via_device": "zigbee2mqtt_bridge_0xe0798dfffec7b47c"
  },
  "device_calss": "shutter",
  "name": "Left",
  "object_id": "rolläden_wohnzimmer_left",
  "origin": {
    "name": "Zigbee2MQTT",
    "sw": "1.42.0",
    "url": "https://www.zigbee2mqtt.io"
  },
  "position_template": "{{ value_json.position }}",
  "position_topic": "zigbee2mqtt/Rolläden Wohnzimmer/left",
  "set_position_template": "{ \"position_left\": {{ position }} }",
  "set_position_topic": "zigbee2mqtt/Rolläden Wohnzimmer/left/set",
  "state_closing": "DOWN",
  "state_opening": "UP",
  "state_stopped": "STOP",
  "state_topic": "zigbee2mqtt/Rolläden Wohnzimmer/left",
  "unique_id": "0xa4c138018cf95021_cover_left_zigbee2mqtt",
  "value_template": "{% if \"moving\" in value_json and value_json.moving %} {{ value_json.moving }} {% else %} STOP {% endif %}"
}
```

Manually sending an updated mqtt event fixes this.

Koenkk, 2024-12-17:
Thanks for the PR!

closed 2024-12-17 by Koenkk

hsantens, 2025-03-03:
Hi
Sorry to get back in this older topic, but I was wondering is this was fully implemented?
I am using [ROBB ROB_200-029-0] controllers and still don't have and opening or closing state when the shutters are moving. Is there something that I should do to enable this? 
I'm running version 2.1.3

PR #25240 (2024-12-17): fix: cover moving state not reflected in HA state
Closes #23713

Tempalte was using non-existent value_json property