Koenkk/zigbee2mqtt #23715: [BUG] Aqara B1 motor ZNCLDJ12LM not updating states

Dillton, 2024-08-23:
### What happened?

Hi.
My Aqara B1 motor (ZNCLDJ12LM) is not updating it's states often but not every time.
Simple example curtain is opened and I set position to xy% the action is done as expected but when the action is finished the state in Z2M is not updated as expected. Still showing open with position 100.
Another example but the same result when I set from open to close the state after it is still open or from close to open the state remains close.

If I would do some math then approx 70% state updates is just wrong. 

I tried to update the entity via HA service but no luck.
I tried to request state update manually in Z2M device exposes which is working, state and positions is updated correctly.
Based on these findings I managed to hotfix this by creating additional automation which sending mqtt request zigbee2mqtt/FRIENDLY_NAME/get with payload {"state": ""} after every detected change on sensor.motor _state entity.

### What did you expect to happen?

Afted the position change it should update the state to open with xy% position.

### How to reproduce it (minimal and precise)

Not exactly sure as I think it's happening since some time ago but it didn't blocked me from anything until now when I need to set some automation with position conditions.

### Zigbee2MQTT version

1.39.1-1

### Adapter firmware version

20240710 happened also with release version

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus-P

### Setup

Add-on in HAOS in Proxmox on Intel NUC

demey, 2024-10-26:
Same behavior on version 1.40.2-1 and same setup.
Coordinator: SONOFF Zigbee 3.0 USB Dongle Plus-P
Coordinator type: zStack3x0
Coordinator revision: 20221226

Partially I resolved this issue by updating config mqtt topic for the cover device.

Dillton, 2024-12-26:
This is still an issue.

Dillton, 2024-12-26:
> Same behavior on version 1.40.2-1 and same setup. Coordinator: SONOFF Zigbee 3.0 USB Dongle Plus-P Coordinator type: zStack3x0 Coordinator revision: 20221226
> 
> Partially I resolved this issue by updating config mqtt topic for the cover device.

I have a bit of the time to more improve the workarounds for this. What did you changed in mqtt topic?

demey, 2024-12-26:
@Dillton I'm using this code:
`{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/bedroom_sh_motor/availability"}],"availability_mode":"all","command_topic":"zigbee2mqtt/bedroom_sh_motor/set","device":{"identifiers":["zigbee2mqtt_0x04cf8cdf3c738f3e"],"manufacturer":"Aqara","model":"Curtain controller B1 (ZNCLDJ12LM)","name":"bedroom_sh_motor","sw_version":"2019\u0000www.","via_device":"zigbee2mqtt_bridge_0x00124b001c2c90d4"},"json_attributes_topic":"zigbee2mqtt/bedroom_sh_motor","name":null,"object_id":"bedroom_sh_motor","origin":{"name":"Zigbee2MQTT","sw":"1.40.2","url":"https://www.zigbee2mqtt.io"},"position_template":"{{ value_json.position }}","position_topic":"zigbee2mqtt/bedroom_sh_motor","set_position_template":"{ \"position\": {{ position }} }","set_position_topic":"zigbee2mqtt/bedroom_sh_motor/set","state_closing":"closing","state_opening":"opening","state_stopped":"stopped","state_topic":"zigbee2mqtt/bedroom_sh_motor","unique_id":"0x04cf8cdf3c738f3e_cover_zigbee2mqtt","value_template":"{% if value_json.position >= 99 %} open {% elif value_json.position <= 1 %} closed {% else %} {{ value_json.motor_state }} {% endif %}"}`
in this topic:
`homeassistant/cover/0x04cf8cdf3c738f3e/cover/config`

Dillton, 2024-12-27:
So if I am correct you have just changes the value_tempate right?

Dillton, 2025-02-26:
This is not stale. Still open point, still not solved on Z2M side.

Dillton, 2025-04-28:
This is still a problem but after Z2M 2.0 update it's happening far less than before at least on my side.

closed 2025-07-07 by github-actions[bot]