Koenkk/zigbee2mqtt #23716: [New device support]: Tuya bidirectional energy meter TS0601 / _TZE204_kpg9hpb0

M66B, 2024-08-23:
### Link

https://aliexpress.com/item/1005007268370253.html

### Database entry

{"id":39,"type":"Router","ieeeAddr":"0xa4c138029edcd4d7","nwkAddr":43287,"manufId":4417,"manufName":"_TZ3000_kqvb5akv","powerSource":"Mains (single phase)","modelId":"TS0001","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"57344":{"attributes":{"53251":"AAAA"}},"genBasic":{"attributes":{"65503":"(\u0000\u0000\u0000\u0012","65506":56,"65508":0,"65534":0,"modelId":"TS0001","manufacturerName":"_TZ3000_kqvb5akv","powerSource":1,"zclVersion":3,"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":""}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":233,"rmsCurrent":0,"activePower":0}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,91]}},"genOnOff":{"attributes":{"onOff":1,"onTime":0,"offWaitTime":0,"moesStartUpOnOff":2}},"manuSpecificTuya_3":{"attributes":{"53248":0,"53249":0,"53250":0,"53251":0,"53252":0,"53253":0,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0029dc0998","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0029dc0998","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0029dc0998","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5,"manufacturerCode":null},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50,"manufacturerCode":null},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10,"manufacturerCode":null},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1],"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":559524985},"lastSeen":1724429381109}

### Comments

This is a partial implementation because I needed power measurements only.
Switching on/off works, and voltage/current/power measurements too, but there are a number of other features as well.
Perhaps somebody else can continue with this.

### External definition

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE204_kpg9hpb0',
        },
    ],
    model: 'ZBN-JLS-63',
    vendor: 'Tuya',
    description: 'Bidirectional energy meter',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [
            tuya.exposes.switch(),
            e.voltage(),
            e.power(),
            e.current(),
    ],
    meta: {
            tuyaDatapoints: [
                [16, 'state', tuya.valueConverter.onOff],
                [
                    6,
                    null,
                    {
                        from: (v) => {
                            return {
                                voltage: v.readUint16BE(0) / 10,
                                current: ((v.readUint8(2) << 16) + (v.readUint8(3) << 8) + v.readUint8(4)) / 1000,
                                power: (v.readUint8(5) << 16) + (v.readUint8(6) << 8) + v.readUint8(7),
                            };
                        },
                    },
                ],
                [6, 'voltage', tuya.valueConverter.raw],
                [6, 'current', tuya.valueConverter.raw],
                [6, 'power', tuya.valueConverter.raw],
            ],
    },
    extend: [
    ],
};

module.exports = definition;
```

M66B, 2024-08-23:
Probably same/similar device as #23613  #23502

M66B, 2024-08-23:
Probably similar to this device too: #18417

Koenkk, 2024-08-26:
Could you check if the device works when changing all occurrences of `_TZE204_kpg9hpb0` to: `_TZE200_lsanae15` or `_TZE200_rhblgy0z` or `_TZE204_l6llgoxq` or `_TZE204_q22avxbv`

M66B, 2024-08-27:
@Koenkk the mentioned devices are similar in some aspects, but all missing:

```
[110, 'reverse_energy', tuya.valueConverter.divideBy100]
```

This type of device is new because formerly there were no devices on AliExpress that could measure bidirectional, targeted at solar panels, etc.

Koenkk, 2024-08-28:
I see, feel free to make a PR for this device by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/tuya.ts)

M66B, 2024-11-09:
Another reference implementation: https://forum.hacf.fr/t/utilisation-dun-module-on-off-energy-meter-volatge-et-courant-protection-model-zbn-dj-63-sur-aliexpress/49410/13

M66B, 2024-11-13:
Improved version, still without threshold values (can't test those):

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE204_lb0fsvba', // ZBN-DJ-63
        },
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE204_kpg9hpb0', // ZBN-JLS-63
        },
    ],
    model: 'ZBN-DJ/JLS-63',
    vendor: 'NoName',
    description: 'Bidirectional energy meter',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [
            tuya.exposes.switch(),
            e.energy(),
            e.voltage(),
            e.power(),
            e.current(),
            e.produced_energy(),
            exposes.numeric('temperature', ea.STATE)
                .withUnit('°C')
                .withDescription('Temperature'),
            exposes.enum('indicator_mode', ea.ALL, ['off', 'normal', 'inverse'])
                .withDescription('LED indicator mode'),
            exposes.binary('power_outage_memory', ea.ALL, 'ON', 'OFF')
                .withDescription('Retain state after power outage (ON - remembers last state)'),
    ],
    meta: {
            tuyaDatapoints: [
                [1, 'energy', tuya.valueConverter.divideBy100], // Forward energy total
                [6, null, tuya.valueConverter.phaseVariant2], // Handles voltage, current, power
                [16, 'state', tuya.valueConverter.onOff],
                [22, 'indicator_mode', tuya.valueConverterBasic.lookup({'off': 0, 'normal': 1, 'inverse': 2})],
                [23, 'power_outage_memory', tuya.valueConverter.onOff],
                [103, 'temperature', tuya.valueConverter.raw],
                [110, 'produced_energy', tuya.valueConverter.divideBy100], // Reverse Energy Total
            ],
    },
    extend: [
    ],
};

module.exports = definition;
```

closed 2025-01-20 by github-actions[bot]