Koenkk/zigbee2mqtt #23718: NOUS B2Z/B3Z relay configuration error

jnorbi, 2024-08-23:
### What happened?

I have NOUS B2Z and NOUS B3Z relays, the issue is the same. I got this errors on configuration/reconfiguration:

Alert #1: Failed to configure (Bind 0xa4c13869317d88d8/1 genOnOff from '170/1' failed (addr.indexOf is not a function))

Alert #2: z2m: Failed to configure '0xa4c13869317d88d8', attempt [X] (TypeError: Bind 0xa4c13869317d88d8/1 genOnOff from '170/1' failed (addr.indexOf is not a function)
    at Driver.macAddrStringToArray (/app/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:803:18)
    at DeconzAdapter.bind (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:847:37)
    at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:509:34)
    at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:42:24)
    at configure (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:3434:29)
    at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/index.ts:190:21)
    at Configure.configure (/app/lib/extension/configure.ts:126:13)
    at Configure.onMQTTMessage (/app/lib/extension/configure.ts:56:21)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:174:17))

### How to reproduce it (minimal and precise)

Easy to reproduce, join or reconfigure the NOUS B2Z/B3Z relay will generate this error.

### Zigbee2MQTT version

1.39.1

### Adapter firmware version

0x26720700

### Adapter

ConBee2

### Setup

Add-on on Home Assistant on RPi 4

Koenkk, 2024-08-26:
Could you provide the debug log of this?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

jnorbi, 2024-08-27:
@Koenkk I've switched the logging level to debug and the only relevant log entry what is belongs to the device is what I pasted before.

lgjunk17, 2024-09-28:
Hello

Same with HEIMAN HS3AQ, in 1.39.1 or 1.40.1

`error 2024-09-28 22:37:38z2m: Failed to configure '0x84fd27fffe08106a', attempt 1 (TypeError: Bind 0x84fd27fffe08106a/1 msRelativeHumidity from '28/1' failed (addr.indexOf is not a function) at Driver.macAddrStringToArray (/var/wwerror 2024-09-28 22:37:38z2m: Failed to configure '0x84fd27fffe08106a', attempt 1 (TypeError: Bind 0x84fd27fffe08106a/1 msRelativeHumidity from '28/1' failed (addr.indexOf is not a function) at Driver.macAddrStringToArray (/var/www/html/plugins/z2m/resources/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:803:18) at DeconzAdapter.bind (/var/www/html/plugins/z2m/resources/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:847:37) at Endpoint.bind (/var/www/html/plugins/z2m/resources/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:509:34) at Object.bind (/var/www/html/plugins/z2m/resources/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:42:24) at Object.configure (/var/www/html/plugins/z2m/resources/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/devices/heiman.ts:741:29) at Configure.configure (/var/www/html/plugins/z2m/resources/zigbee2mqtt/lib/extension/configure.ts:126:37) at Immediate.<anonymous> (/var/www/html/plugins/z2m/resources/zigbee2mqtt/lib/extension/configure.ts:75:28))w/html/plugins/z2m/resources/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/deconz/driver/driver.ts:803:18) at DeconzAdapter.bind (/var/www/html/plugins/z2m/resources/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:847:37) at Endpoint.bind (/var/www/html/plugins/z2m/resources/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:509:34) at Object.bind (/var/www/html/plugins/z2m/resources/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:42:24) at Object.configure (/var/www/html/plugins/z2m/resources/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/devices/heiman.ts:741:29) at Configure.configure (/var/www/html/plugins/z2m/resources/zigbee2mqtt/lib/extension/configure.ts:126:37) at Immediate.<anonymous> (/var/www/html/plugins/z2m/resources/zigbee2mqtt/lib/extension/configure.ts:75:28))`

Koenkk, 2024-09-30:
May be fixed in tomorrow z2m release, please check.

closed 2024-12-20 by github-actions[bot]