Koenkk/zigbee2mqtt #23719: Sonoff SWV smart water valve - cyclic timed/quantitative irrigation controls not exposed in Home Assistant

Elu43, 2024-08-23:
### What happened?

Hello,

With Sonoff Smart Water Valve (SWV) integrated with Zigbee2MQTT, we can control various parameters for cyclic timed or quantitative irrigation :
[image]

However, in Home Assistant it's impossible to control or modify any of the irrigation parameters directly :

[image]

It's just giving the state without the possibility to modify it with a slider.

Best regards 

### Zigbee2MQTT version

1.39.1

### Adapter firmware version

20240710

### Adapter

ZigStar UZG-01

### Setup

Docker

ghhatint, 2024-08-28:
In HA I manage SWV via script. This is an example:

TimedIrrigation:
  alias: Timed irrigation
  sequence:
  - service: mqtt.publish
    data:
      qos: '0'
      payload: '{"cyclic_timed_irrigation": {"current_count": 1, "total_number": 2,
        "irrigation_duration": 30, "irrigation_interval": 10}}'
      topic: zigbee2mqtt/Valve001/set
      retain: false
  icon: mdi:sprinkler-variant
  description: ''

this script start 2 times irrigation for 30 sec (paused 10 sec between irrigation). Insert it in script.yaml (or in configuration.yaml).
You can also manage it in HA from Settings+Automations&Scene+Scripts+Add script
[image]

(Valve001 is friendly name of my SWV)
Here you can find other command: https://www.zigbee2mqtt.io/devices/SWV.html

I hope it's useful to you. 
Ciao!

closed 2024-12-20 by github-actions[bot]

splash112, 2025-06-26:
I use the following, with lots of help from the above posts:

1 script:
`action: mqtt.publish
metadata: {}
data:
  evaluate_payload: false
  qos: "1"
  retain: true
  topic: zigbee2mqtt/{{ var_topic }}/set
  payload: >
    {"cyclic_quantitative_irrigation": {"current_count": {{ var_current_count |
    int }}, "total_number": {{ var_total_number | int }}, "irrigation_capacity":
    {{ var_irrigation_capacity | int }}, "irrigation_interval": {{
    var_irrigation_interval | int }}} }
`

And then you can make automations like this:
`
alias: Sunset heat irrigation
description: ""
triggers:
  - trigger: sun
    event: sunset
    offset: 0
conditions:
  - type: is_temperature
    condition: device
    device_id: 9ebc8906a4f5599fd1abb37a79028bce
    entity_id: bd33c456889e66ddfe92331bd482e644
    domain: sensor
    above: 28
actions:
  - variables:
      var_topic: SWV Back
      var_current_count: 0
      var_total_number: 3
      var_irrigation_capacity: 5
      var_irrigation_interval: 3600
  - action: script.cyclic_quantitative_irrigation
    metadata: {}
    data:
      var_topic: "{{ var_topic }}"
      var_current_count: "{{ var_current_count }}"
      var_total_number: "{{ var_total_number }}"
      var_irrigation_capacity: "{{ var_irrigation_capacity }}"
      var_irrigation_interval: "{{ var_irrigation_interval }}"
`
The variables in the automation make it just easier to edit, the script is reusable, so you can repeat the variables script part for different valves.