Koenkk/zigbee2mqtt #23720: npm ci takes forever after trying to use update.sh

its-Zeke, 2024-08-23:
### What happened?

I wanted to add a new zigbee device to my devices, but it wasnt working. Therefore i decided to update my z2m version using the update.sh script in the opt/z2m folder. 
After waiting for about 5h nothing more seemed to happen:
[image]

I tryed to use the dev branch, same behaviour. System completely stucks.
When i abort the script, this is the end of the logfile:
[image]

Also since trying to update, my z2m service keeps crashing and restarting itself.

### What did you expect to happen?

I was expecting to use the update.sh script to update things that needs to get updates and done. 

### How to reproduce it (minimal and precise)

run './update.sh'

### Zigbee2MQTT version

1.39.1 at least thats what npm run build says, which also fails

### Adapter firmware version

20210708

### Adapter

sonoff stick

### Setup

plain

its-Zeke, 2024-08-23:
This is now the status of the service :
[image]

lux73, 2024-08-24:
upgrade NodeJS to v20 minimum...

you are running 18.20.4 which is not supported anymore

its-Zeke, 2024-08-24:
Hm, okay i updated the node version now to >20 using nvm, since npm didnt update version higher than 18..
Using the update.sh script i thought that it would check what node version i am using, but instead it installed further dependencies and killed my service :D
Now, after the update finally went throught, I'm having trouble starting the service because it cant find /usr/bin/node ? I guess thats due to the installation with nvm ?

lux73, 2024-08-24:
correct...  thats not the correct way for upgrading

i am using this Way for my Debian based System: https://phoenixnap.com/kb/debian-install-nodejs - Method 2

its-Zeke, 2024-08-24:
I added a symlink now as described here:
[image]

Seems to work . Service started, GUI is finally accessable.

Lot of trouble just to add a new zigbee device :)

lux73, 2024-08-24:
i don't know if these Hack is working in future Updates ^^

its-Zeke, 2024-08-24:
Alright, the Pi just died again.. gonna check that

closed 2024-12-20 by github-actions[bot]