Koenkk/zigbee2mqtt #23726: Bticino N4027C tilt position not exposes anymore

flyDlight, 2024-08-24:
### What happened?

I have eight Bticino N4027C purchased all together a few months ago. I installed two of them immediately and they showed the tilt position. Now I'm installing the other six, but I noticed that they no longer show the tilt position (not even the two previously installed).

### What did you expect to happen?

I expected that the devices expose the tilt position again

### Zigbee2MQTT version

1.39.1

### Adapter firmware version

7.4.2 [GA]

### Adapter

Sonoff zigbee dongle E

### Setup

Plain on docker on linux debian

flyDlight, 2024-09-19:
Any solutions or suggestions?

Koenkk, 2024-09-20:
I think you need to enable the tilt control in the z2m frontend -> device -> settings specific, and then restart z2m.

flyDlight, 2024-09-25:
it doesn't work. I can see the "tilt" value in the "state" tab but I have no controls on the dashboard or even in home assistant.
Here the state output:
{
    "calibration_mode": "specific_nllv",
    "device": {
        "applicationVersion": 0,
        "dateCode": " \u000020220113\u0000\u0000\u0000\u0000\u0000",
        "friendlyName": "tapp cucina",
        "hardwareVersion": 9,
        "ieeeAddr": "0x0004740001136286",
        "manufacturerID": 4129,
        "manufacturerName": " Legrand",
        "model": "K4027C/L4027C/N4027C/NT4027C",
        "networkAddress": 46114,
        "powerSource": "Mains (single phase)",
        "softwareBuildID": "003d",
        "stackVersion": 69,
        "type": "Router",
        "zclVersion": 8
    },
    "led_if_on": "ON",
    "led_in_dark": "OFF",
    "linkquality": 176,
    "position": 100,
    "state": "OPEN",
    "tilt": 100,
    "update": {
        "installed_version": 4015615,
        "latest_version": 4015615,
        "state": "idle"
    },
    "update_available": false,
    "action": null
}

closed 2024-12-20 by github-actions[bot]