Koenkk/zigbee2mqtt #23729: TS0601_soil_2 (_TZE284_sgabhwa6) soil sensor battery is linked to its temperature

shift-del1, 2024-08-24:
### What happened?

The battery level is wrong, changing its value together with the temperature. In the official Tuya app it is working fine.

[image]

### What did you expect to happen?

I expect to see the real battery level values. See the one from the official Tuya app:

[image]

### How to reproduce it (minimal and precise)

Just check the actual values + history.

### Zigbee2MQTT version

1.39.1

### Adapter firmware version

0x26780700

### Adapter

ConBee2/RaspBee2

### Setup

Add-on on Home Assistant OS on  virtualized x86-64 PC (Proxmox 8.2.4))

nprez83, 2024-09-30:
Thanks @shift-del1. Looks like DP14 and 15 are both labeled Battery level, so still unclear to me which of them is supposed to be used for battery. Could you try to get the DPs a slightly different way please? God to API Explorer -> Device Control -> Query Properties, and plug in your device ID. See an example below for one of my devices. I like that way as it lists the DP number, but also the type of variable and the current state, which may be useful in sorting this out. 

[image]

shift-del1, 2024-09-30:
I'd love to, but my trial period expired and my only choice is to pay 25k or 50k for a license:

"msg": "No permissions. Your subscription to cloud development plan has expired.",

By the way this was the method I was familiar with. Any other option to try?

nprez83, 2024-09-30:
That's interesting, I was under the impression that it's free indefinitely, but perhaps only if you registered before 2021 ([see here](https://support.tuya.com/en/help/_detail/K9zsowlj19oaf)). Hopefully they'll renew it so we can get to the bottom of this.

shift-del1, 2024-09-30:
Hello @nprez83,

Tuya platform was really fast, my trial is extended. Here is the log:

```
{
  "result": {
    "properties": [
      {
        "code": "humidity",
        "custom_name": "",
        "dp_id": 3,
        "time": 1727729758665,
        "type": "value",
        "value": 14
      },
      {
        "code": "temp_current",
        "custom_name": "",
        "dp_id": 5,
        "time": 1727729758665,
        "type": "value",
        "value": 233
      },
      {
        "code": "temp_unit_convert",
        "custom_name": "",
        "dp_id": 9,
        "time": 1727714405935,
        "type": "enum",
        "value": "f"
      },
      {
        "code": "battery_state",
        "custom_name": "",
        "dp_id": 14,
        "time": 1727714505306,
        "type": "enum",
        "value": "high"
      },
      {
        "code": "battery_percentage",
        "custom_name": "",
        "dp_id": 15,
        "time": 1727714505444,
        "type": "value",
        "value": 100
      },
      {
        "code": "temp_alarm",
        "custom_name": "",
        "dp_id": 101,
        "time": 1727714406436,
        "type": "enum",
        "value": "cancel"
      },
      {
        "code": "hum_alarm",
        "custom_name": "",
        "dp_id": 102,
        "time": 1727714406560,
        "type": "enum",
        "value": "loweralarm"
      },
      {
        "code": "maxtemp_set",
        "custom_name": "",
        "dp_id": 103,
        "time": 1727714406683,
        "type": "value",
        "value": 400
      },
      {
        "code": "minitemp_set",
        "custom_name": "",
        "dp_id": 104,
        "time": 1727714505956,
        "type": "value",
        "value": 100
      },
      {
        "code": "maxhum_set",
        "custom_name": "",
        "dp_id": 105,
        "time": 1727714406813,
        "type": "value",
        "value": 70
      },
      {
        "code": "minihum_set",
        "custom_name": "",
        "dp_id": 106,
        "time": 1727714406813,
        "type": "value",
        "value": 20
      },
      {
        "code": "temp_sensitivity",
        "custom_name": "",
        "dp_id": 107,
        "time": 1727714506226,
        "type": "value",
        "value": 5
      },
      {
        "code": "hum_sensitivity",
        "custom_name": "",
        "dp_id": 108,
        "time": 1727714406945,
        "type": "value",
        "value": 2
      },
      {
        "code": "report_sensitivity",
        "custom_name": "",
        "dp_id": 109,
        "time": 1727714407080,
        "type": "value",
        "value": 5
      },
      {
        "code": "temp_current_f",
        "custom_name": "",
        "dp_id": 110,
        "time": 1727729758665,
        "type": "value",
        "value": 739
      },
      {
        "code": "temp_sensitivity_f",
        "custom_name": "",
        "dp_id": 111,
        "time": 1727714506226,
        "type": "value",
        "value": 329
      },
      {
        "code": "maxtemp_set_f",
        "custom_name": "",
        "dp_id": 112,
        "time": 1727714506614,
        "type": "value",
        "value": 1040
      },
      {
        "code": "minitemp_set_f",
        "custom_name": "",
        "dp_id": 113,
        "time": 1727714506614,
        "type": "value",
        "value": 500
      }
    ]
  },
  "success": true,
  "t": 1727732712111,
  "tid": "452c2a3f7f7511efb77736033862a1ba"
}
```
Let me know if you need anything else.

shift-del1, 2024-10-02:
I attached 2 screenshots which could help to understand the values from my previous log. The screenshots were taken a couple minutes later, so maybe they are not representing the datapoints' values.

[image]

[image]

nprez83, 2024-10-05:
Ok, I think I'm realizing what's going on. The device just does not seem to ever publish the battery status. I've had the device set up for several days, so it's not fixed by waiting. Here is the state as published by the device: 
```
z2m: Received Zigbee message from 'Front yard soil sensor', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,0],"type":"Buffer"},"datatype":2,"dp":3},{"data":{"data":[0,0,0,212],"type":"Buffer"},"datatype":2,"dp":5},{"data":{"data":[0,0,2,189],"type":"Buffer"},"datatype":2,"dp":110}],"seq":62465}' from endpoint 1 with groupID 0
```

As you can see, the device reports DP3 (humidity), DP 5 (temperature), and DP 110 (temperature_f). DP 15 is never reported. The other DPs are reported after you've set them (sensitivities, etc), but for the purposes of this test I left them unset. 

@baggiowu, any idea why the device never publishes DP 15?

dave21w, 2024-12-09:
My issue is very similar, I only see temperature and soil moisture, batt level is either a question mark or flashing red percentage sign Z2M main device screen. If i look in the exposes screen both battery and batery state just say Null.

nprez83, 2025-02-15:
@baggiowu, any updates on this?

closed 2025-04-25 by github-actions[bot]