Koenkk/zigbee2mqtt #23730: network commissioning timed out - most likely network with the same panId or extendedPanId already exists nearby since I upgradesd my Sonoff dongle CC2652P again to 20240710

Nardol, 2024-08-24:
### What happened?

I already updated to 20240710 durring test phase but thought firmwares could be different so I flashed again by downloading the firmware from the release page today.
But since I upgraded, Zigbee2MQTT fails to start. I also tried to downgrade, and tried also with my old ZZH after updating because without update I had a ping timeout after 6000MS, unfortunately everytime with same result.
I also tried another USB port, another USB extension cable but without success. Only solution I think about requires pairing again but I don't understand why it happens since it just worked before.
I have changed the Zigbee channel to 25 with success but since I also restarted Zigbee2MQTT after Node update it worked, only today this issues started.

### What did you expect to happen?

I expected that as usually, Zigbee2MQTT start again successfully.

### How to reproduce it (minimal and precise)

Update twice to 20240710

### Zigbee2MQTT version

1.39.1 (commit #e132316a)

### Adapter firmware version

20240710

### Adapter

Sonoff dongle P if I make no confusion with names

### Setup

Plein on an Odroid N2+ under Raspbian

### Debug log

[zigbee2mqtt_2024-08-24.22-55-24.log](https://github.com/user-attachments/files/16738366/zigbee2mqtt_2024-08-24.22-55-24.log)

emillassen, 2024-08-25:
I'm experiencing the same error after upgrading a tubeszb-cc2652p7-poe-2023 running CC1352P2_CC2652P_launchpad_coordinator_20230507 to CC1352P2_CC2652P_launchpad_coordinator_20240710. Downgrading to 20230507 does not fix the issue, and a reinstallation of zigbee2mqtt does also not work.

Nardol, 2024-08-25:
I've set the channel back to 20, the channel I changed from last week and Zigbee2MQTT starts successfully.
I have not tested to switch again to channel 25 yet, just want to enjoy a functional Zigbee network again.
FYI I have to configure again the binding for my Hue dimmer switch (324131092621) but could be more related to the fact I tried some other things before E.G. doing without coordinator_backup.json which could have break it.

@Koenkk if you are interested in some experiments to see what could have caused it I would be happy to help but until that I will not touch anything except my Hue dimmer switch.

Nardol, 2024-08-25:
OK, in fact it is not the binding which does not work but nothing is really connected. Zigbee2MQTT started but nothing can be used at all, timeout are logged when switching something on/off.

Koenkk, 2024-08-25:
Do you have the old coordinator_backup.json file? (from the data folder). If yes, re-flash your new dongle, put the coordinator_backup.json in the data directory and start z2m

Cyclone411, 2024-08-27:
I am running into this same issue with the same adapter on a brand new setup. I've redriven GENERATE multiple times in the hopes it helps, but still stuck. Similarly, upgraded, downloaded firmware, all to no avail. I opened a separate issue at https://github.com/Koenkk/zigbee2mqtt/issues/23751 to not clutter @Nardol 's thread.

AThomsen, 2024-09-22:
I experience the same with CC2652R (zzh): upgrading from CC2652R_coordinator_20230507 to CC2652R_coordinator_20240710.hex gives the error.

I've tried flashing with the previous version (multiple times) and restoring coordinator_backup.json to a version from yesterday. I now simply can't get the network back up.

`z2m: Error: network commissioning timed out - most likely network with the same panId or extendedPanId already exists nearby (Error: AREQ - ZDO - stateChangeInd after 60000ms`

meks007, 2024-11-29:
I had this happen to me yesterday after flashing a slaesh and a sonoff-p. @Koenkk I found out that after I corrected invalid nwk_address hex values from my slaesh coordinator_backup,json everything started working. Without those corrections I kept getting _z2m: Error: network commissioning timed out - most likely network with the same panId or extendedPanId already exists nearby (Error: AREQ - ZDO - stateChangeInd after 60000ms_ with both slaesh and sonoff-p coordinators.

The correction I made to nwk_address was making sure they're full 2 byte values. It appears they're only 3 digits when the 4th digit should be 0.

nwk_address: ad0 to nwk_address: 0ad0

I found that out while trying to restore the backup to the coordinator using zigpy-cli. Zigpy errord out with ValueErrors because it wasn't a valid hex values. With leading zeros zigpy managed to restore the coordinator backup. After that Z2M started working.

  _"devices": [
    {
      "nwk_address": "ad0",
      "ieee_address": "00158d00022c75de",
      "is_child": false,
      "link_key": {
        "key": "5f5e7bb47e298e444898d281ea683973",
        "rx_counter": 0,
        "tx_counter": 1799
      }
    },_

closed 2025-02-05 by github-actions[bot]

meks007, 2025-02-05:
@Koenkk I still think this should be addressed. There's not much point in having backup functionality that produces invalid backups. A few simple zero's decide between being able to restore or having to rebuild the whole network.

usego, 2025-06-19:
This might be unrelated, but an issue disappeared for me immediately after I connected the stick to an extension cord.

1aranzant, 2025-11-14:
same issue... I updated the core and firmware of my SLZ06, gives me the same issue

obbardc, 2025-11-17:
OK, reflashing the firmware on the Sonoff adapter and restarting z2mqtt and it works now. Maybe its worth having a note in the docs that you need a freshly flashed/reset adapter ?