Koenkk/zigbee2mqtt #23733: Since Ember a lot of warnings/errors mostly on Philips Hue lights: delivery failed, ping failed etc...

xabix, 2024-08-25:
### What happened?

Hello Team,

I am strugling to understand if the issues are linked to a config or a bug but my network (82 devices) was working ok with ezsp (and Jeedom) and since I moved to HA, it was OK but the ezsp adaptor got deprecated so I had to move to ember. I have a dongle E with ncp-uart-hw-v7.4.3.0-zbdonglee-230400.gbl firmware.

Type of issues I get:
[image]

even restarting i get them right away so I don't think it s bc of the network being overloaded but maybe more a change between the 'format' or the 'timers' between ezsp and ember?

Any help will be appreciated as it seems only batterie devices are working and most of the plugs one not. Is it recommended to change the stick?

Merci

### What did you expect to happen?

Network to work as with ezsp without having to start from 0 (reset of the config)

### How to reproduce it (minimal and precise)

just try to change a state of a bulb (Salon Plafond light for example)

### Zigbee2MQTT version

1.39.1 by unknown?

### Adapter firmware version

7.4.3 GA

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus V2

### Setup

Proxmox/HA 2024.8.2

### Debug log

full logs:
[Z2M.log](https://github.com/user-attachments/files/16739515/Z2M.log)

Nerivec, 2024-08-26:
Any chance you can confirm if the behavior is the same with latest dev? (Otherwise, can you can report back after September 1st release, it will have several fixes/improvements for `ember`.)

xabix, 2024-08-27:
Great, I can confirm that this works much better so this can be closed and it's covered in the coming release. MERCI.

BTW I saw this msg:
error 2024-08-27 09:02:26z2m: Failed to read state of 'Pool House' after reconnect (ZCL command 0x00178801061ed090/11 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":37742,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":11,"options":4416,"groupId":0,"sequence":178},"zclSequence":36,"commandIdentifier":1} timed out after 10000ms))

billygr, 2025-03-21:
Hi,

i am encountering the same situation (Hue lamps recently joined on the network 3 items, running 67.116.3) 
Switched to just using two of them and it is always one of the two random stuck with messages like the above (ping didn't help either)

Sonoff ZBdongle-E with zbdonglee_zigbee_ncp_8.0.2.0_sw_flow_115200.gbl from https://github.com/darkxst/silabs-firmware-builder/releases

Zigbee2MQTT version (docker)
2.1.3-dev commit: 9792466aac8b94747fe41fe47db766b7a7e93367

Adapter firmware version
8.0.2 [GA]

Can you provide some guidance on what is needed to check with ember-zli as a starting point ?

There are cases where just restarting the zigbee2mqtt solves it for 10-30 minutes