Koenkk/zigbee2mqtt #23741: [Feature request]: OWON PC321 - Power factor

rommess, 2024-08-26:
### Is your feature request related to a problem? Please describe

It's not a problem, but improvement. 
I'd like to change the default statistics for OWON Meter Power factor l1 bar graph

### Describe the solution you'd like

I'll like to be in line graph to be more "readable"

### Describe alternatives you've considered

i don't know in wich .yaml it can be changed.

### Additional context

[image]

Koenkk, 2024-08-27:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-08-27 by Koenkk

rommess, 2024-09-01:
Beauty, Thank you very much.