Koenkk/zigbee2mqtt #23743: [New device support]:

Teeseeone, 2024-08-26:
### Link

https://frient.com/products/smart-din-relay/

### Database entry

{"id":23,"type":"Router","ieeeAddr":"0x0015bc0014030eaa","nwkAddr":39348,"manufId":4117,"manufName":"frient A/S","powerSource":"Mains (single phase)","modelId":"SMRZB-342","epList":[1,2,242],"endpoints":{"1":{"profId":49353,"epId":1,"devId":1,"inClusterList":[5,6],"outClusterList":[],"clusters":{},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xbc026efffefe6a41","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":81,"inClusterList":[0,2,3,4,5,6,9,1794,2820,2821],"outClusterList":[3,10,25,1030],"clusters":{"genBasic":{"attributes":{"modelId":"SMRZB-342","manufacturerName":"frient A/S","powerSource":1,"zclVersion":7,"dateCode":"2023-11-24 08:20"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"dateCode":"2023-11-24 08:20","zclVersion":7,"interviewCompleted":true,"meta":{},"lastSeen":1724672468973}

### Comments

I am very new to this, can you please add support?

### External definition

```shell
const {deviceEndpoints, onOff, deviceTemperature, identify, electricityMeter} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['SMRZB-342'],
    model: 'SMRZB-342',
    vendor: 'frient A/S',
    description: 'Automatically generated definition',
    extend: [deviceEndpoints({"endpoints":{"1":1,"2":2}}), onOff({"powerOnBehavior":false,"endpointNames":["1","2"]}), deviceTemperature({"endpointNames":["2"]}), identify(), electricityMeter()],
    meta: {"multiEndpoint":true},
};

module.exports = definition;
```

Teeseeone, 2024-09-06:
Did i do it corect?

Koenkk, 2024-09-07:
Does the external definition work?

Teeseeone, 2024-09-07:
Not 100% sure what external defitnition is, but this is what is exposed.

[image]
 
Checked with chatgpt and did not get any smarter..

[image]

Koenkk, 2024-09-08:
Does something happen when trying to control it?

Teeseeone, 2024-09-08:
I can turn it on and off from switch 2 that have a slider.

Koenkk, 2024-09-09:
- When refreshing the voltage for example, does this also work?
- Does the configure succeed when pressing the yellow configure button?

Teeseeone, 2024-09-09:
No, only switch 2 works.

Get and error.
[image]

Teeseeone, 2024-10-02:
> * When refreshing the voltage for example, does this also work?
> * Does the configure succeed when pressing the yellow configure button?

Will SMRZB-342 be added any time soon?

Teeseeone, 2024-12-12:
Device is not yet added

closed 2025-02-20 by github-actions[bot]