Koenkk/zigbee2mqtt #23746: Aqara T1 water leak sensor SJCGQ12LM - No Battery status

Technol7, 2024-08-26:
### What happened?

Hello,
I have paired 3 Aqara SJCGQ12LM leak sensors via zigbee2mqtt but I can't get battery feedback (it's been 5-6 days).
I found an identical problem from the beginning of the year but the topic (#20764) was closed even though I don't seem to be the only one in this case.
Sorry for my approximate English.
Thank you in advance for your help or clue to find a solution.

### Zigbee2MQTT version

1.39.1

### Adapter firmware version

20210708

### Adapter

Sonoff ZB Dongle-P

### Setup

Add-ob on Home Assistant OS 13.0 on HA Yellow

Technol7, 2024-12-14:
Bonjour,
Malgré les MàJ de Zigbee2mqtt, l'information des piles ne fonctionnent toujours pas.
Quelle solution auriez-vous ?
Je vous remercie par avance,

crivchri, 2025-01-19:
Hi,

I have the same issue and would like to know whether with Zigbee2MQTT version 2.0.0 the workaround with [external converter](https://gist.github.com/Koenkk/bedcf4cc4f3c63058349151571e7a82a) mentioned in issue https://github.com/Koenkk/zigbee2mqtt/issues/20764#issuecomment-1912423506 is still necessary, or if this fix was included in the default integration (no external converter necessary)?

Thank you and kind regards!

crivchri, 2025-03-21:
> Hi,
>
> I have the same issue and would like to know whether with Zigbee2MQTT version 2.0.0 the workaround with [external converter](https://gist.github.com/Koenkk/bedcf4cc4f3c63058349151571e7a82a) [...]

Hi,

Could anyone advise on this issue, whether with Zigbee2MQTT version 2.1.3 the workaround with [external converter](https://gist.github.com/Koenkk/bedcf4cc4f3c63058349151571e7a82a) mentioned in issue https://github.com/Koenkk/zigbee2mqtt/issues/20764#issuecomment-1912423506 is still necessary, or if this fix was included in the default integration (no external converter necessary)?

Thank you and kind regards.

xbmcnut, 2025-04-05:
> > Hi,
> > I have the same issue and would like to know whether with Zigbee2MQTT version 2.0.0 the workaround with [external converter](https://gist.github.com/Koenkk/bedcf4cc4f3c63058349151571e7a82a) mentioned in issue [#20764 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/20764#issuecomment-1912423506) is still necessary, or if this fix was included in the default integration (no external converter necessary)?
> > Thank you and kind regards!
> 
> Hi,
> 
> Could anyone please advise on this issue, whether with Zigbee2MQTT version 2.1.3 the workaround with [external converter](https://gist.github.com/Koenkk/bedcf4cc4f3c63058349151571e7a82a) mentioned in issue [#20764 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/20764#issuecomment-1912423506) is still necessary, or if this fix was included in the default integration (no external converter necessary)?
> 
> Thank you and kind regards.

Battery status works natively now but is reported at 50% value so you'll need a template helper to fix.
```
{{ states('sensor.aqara_t1_water_valve_battery')| float (0)
* 2 }}
```

closed 2025-06-12 by github-actions[bot]