Koenkk/zigbee2mqtt #23751: network commissioning timed out - most likely network with the same panId or extendedPanId already exists nearby

Cyclone411, 2024-08-27:
### What happened?

Got a new CC2652P (SONOFF-Zigbee-3.0-USB-dongle-plus P variant) to set up for a family member on a pi4 via z2mqtt on docker. New setup, no devices already paired. I have the same dongle on a different machine running my home network flawlessly for the past 3+ years.

On this new installation, when starting up, after a minute or two of waiting, I'll fail to start with the following message:

Error: network commissioning timed out - most likely network with the same panId or extendedPanId already exists nearby

### What did you expect to happen?

Successful start of a new Zigbee2MQTT installation

### How to reproduce it (minimal and precise)

All starts of Zigbee2MQTT via docker compose up fails

### Zigbee2MQTT version

1.39.1 (commit #e132316)

### Adapter firmware version

20230507

### Adapter

SONOFF-Zigbee-3.0-USB-dongle-plus CC2652P

### Setup

Pi4 via docker compose

### Debug log

[log.log](https://github.com/user-attachments/files/16754912/log.log)

siandco, 2024-08-27:
As you are setting up a new dongle with no devices i'd recommend to edit the configuration.yaml file to get zigbee2mqtt to generate new values for **pan_id**, **ext_pan_id** and **network_key**

See: [https://www.zigbee2mqtt.io/guide/configuration/zigbee-network.html#network-config](https://www.zigbee2mqtt.io/guide/configuration/zigbee-network.html#network-config)

Cyclone411, 2024-08-27:
> As you are setting up a new dongle with no devices I'd recommend editing the configuration.yaml file to get zigbee2mqtt to generate new values for **pan_id**, **ext_pan_id** and **network_key**
> 
> See: https://www.zigbee2mqtt.io/guide/configuration/zigbee-network.html#network-config

OMG Thank you!!! Changed all three to generate and it worked on the first try! It has been so long since I did that for my own network, it just didn't click to set the other two to generate as well.

Odd that the [Getting Started guide](https://www.zigbee2mqtt.io/guide/getting-started/#installation) includes a sample config, but the sample doesn't include generating the pan or ext_pan ids, nor even the comments for them. It does include doing a generate for the network_key though. At least it got all worked in the end!

closed 2024-08-27 by Cyclone411

rickyoleary, 2025-11-30:
Given this is the first post on Google for what I searched people should be aware, like me, they're pulling their non-existent hair out because this isn't the solution for my case. The problem is you need to be using the USB extension cable. I plugged directly into my server, same error. After more research it reminded me to use the extension. It DOES matter. With that, it worked. So if you're plugging in directly, I think an extension will help. Otherwise the original fix as per this thread. But worth mentioning. It's the exact same error message so very generic.