Koenkk/zigbee2mqtt #23754: [New device support]: TS0601 TRV Devices - Unsupported

Forstfb1, 2024-08-27:
### Link

https://www.aliexpress.com/item/1005005454936849.html?spm=a2g0o.order_list.order_list_main.5.5f321802cPC0l2

### Database entry

{"id":88,"type":"EndDevice","ieeeAddr":"0xa4c1389273b1279f","nwkAddr":14059,"manufId":4417,"manufName":"_TZE204_zljqtner","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65487":14400,"65503":"G�`.iH�`.iI�`.iK�`.iN�`.iR�`.iU�`.fV�`.\u0012","65506":56,"65508":0,"stackVersion":0,"dateCode":"","appVersion":74}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1724761166920} {"id":89,"type":"EndDevice","ieeeAddr":"0xa4c138425dcc44d1","nwkAddr":41530,"manufId":4417,"manufName":"_TZE204_zljqtner","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65487":14400,"65503":"\u0000\u0000\u0000\u0000\u0011\u0000\u0000\u0000\u0000\u0011","65506":56,"65508":1,"modelId":"TS0601","manufacturerName":"_TZE204_zljqtner","powerSource":3,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1724761172904}

### Comments

[image]
I'm a complete newbie, and love the way that 99.9% of the items i have added "Just Work" this is the first time i have come across an item that did... please help :

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_zljqtner',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

atzetot, 2024-09-23:
Any update and advise yet how to integrate this device properly ?

banolka, 2024-09-27:
I have exact same issue - valve is not recognized

lenivez123, 2024-10-12:
У меня такая же проблема. Решение кто нибудь подскажет?

rdkm82, 2024-11-04:
https://houseiq.pl/pl/p/Glowica-termostatyczna-GTZ06-ZigBee-3.0-TUYA/1620

I bought my thermostatic radiator valve at the end of September. I couldn't add it correctly - no support. is there any chance to add support for this thermostatic head?

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_zljqtner',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

I also failed to add it using custom quirks

module.exports = definition;
[image]

makobpl, 2025-01-06:
I managed to get SOME functionality using z2m-edge and converters file.
I can get/set temperature, offset, window, child lock. I get idle/heating state back, not sure if battery works correctly.
I have not tested the schedules.

Here is the converters.js content:

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [
        {modelID: 'TS0601', manufacturerName: '_TZE204_zljqtner'},
    ],
    model: 'TS0601_thermostat_1',
    vendor: 'TuYa',
    description: 'Thermostatic radiator valve',
    whiteLabel: [
        {vendor: 'Unknown/id3.pl', model: 'GTZ06'},
    ],
    onEvent: tuya.onEventSetLocalTime,
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [
        e.battery(), 
        e.child_lock(), 
        e.max_temperature().withValueMin(15).withValueMax(45), 
        e.min_temperature().withValueMin(5).withValueMax(15),
        e.window_detection(),
        e.binary('window', ea.STATE, 'CLOSED', 'OPEN').withDescription('Window status closed or open '),
        e.binary('alarm_switch', ea.STATE, 'ON', 'OFF').withDescription('Thermostat in error state'),
        e.climate()
            .withLocalTemperature(ea.STATE)
            .withSetpoint('current_heating_setpoint', 5, 35, 0.5, ea.STATE_SET)
            .withLocalTemperatureCalibration(-30, 30, 0.1, ea.STATE_SET)
            .withPreset(['auto', 'manual', 'off', 'on'],
                'MANUAL MODE ☝ - In this mode, the device executes manual temperature setting. ' +
            'When the set temperature is lower than the "minimum temperature", the valve is closed (forced closed). ' +
            'AUTO MODE ⏱ - In this mode, the device executes a preset week programming temperature time and temperature. ' +
            'ON - In this mode, the thermostat stays open ' +
            'OFF - In this mode, the thermostat stays closed')
            .withSystemMode(['auto', 'heat', 'off'], ea.STATE)
            .withRunningState(['idle', 'heat'], ea.STATE),
        ...tuya.exposes.scheduleAllDays(ea.STATE_SET, 'HH:MM/C HH:MM/C HH:MM/C HH:MM/C'),
    ],
    meta: {
        tuyaDatapoints: [
            [1, null,
                {
                    from: (v) => {
                        utils.assertNumber(v, 'system_mode');
                        const presetLookup = {0: 'auto', 1: 'manual', 2: 'off', 3: 'on'};
                        const systemModeLookup = {0: 'auto', 1: 'auto', 2: 'off', 3: 'heat'};
                        return {preset: presetLookup[v], system_mode: systemModeLookup[v]};
                    },
                },
            ],
            [1, 'system_mode', tuya.valueConverterBasic.lookup({'auto': tuya.enum(1), 'off': tuya.enum(2), 'heat': tuya.enum(3)})],
            [1, 'preset', tuya.valueConverterBasic.lookup(
                {'auto': tuya.enum(0), 'manual': tuya.enum(1), 'off': tuya.enum(2), 'on': tuya.enum(3)})],
            [2, 'current_heating_setpoint', tuya.valueConverter.divideBy10],
            [3, 'local_temperature', tuya.valueConverter.divideBy10],
            [6, 'running_state', tuya.valueConverterBasic.lookup({'heat': 1, 'idle': 0})],
            [7, 'window', tuya.valueConverterBasic.lookup({'OPEN': 1, 'CLOSE': 0})],
            [8, 'window_detection', tuya.valueConverter.onOff],
            [9, 'max_temperature', tuya.valueConverter.divideBy10],
            [10, 'min_temperature', tuya.valueConverter.divideBy10],
            [12, 'child_lock', tuya.valueConverter.lockUnlock],
            [13, 'battery', tuya.valueConverter.raw],
            //[14, 'alarm_switch', tuya.valueConverter.onOff],
            [101, 'local_temperature_calibration', tuya.valueConverter.localTempCalibration1],
            [102, 'schedule_monday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(1)],
            [103, 'schedule_tuesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(2)],
            [104, 'schedule_wednesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(3)],
            [105, 'schedule_thursday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(4)],
            [106, 'schedule_friday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(5)],
            [107, 'schedule_saturday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(6)],
            [108, 'schedule_sunday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(7)],
        ],
    },
};

module.exports = definition;

vjaunet, 2025-01-16:
Hi there,
Here are the codes I obtained for the tuyadatapoints. Not sure how to implement that properly in an external converter, if someone could help, that'd be appreciated.
best,

{
"1":"Mode",
"2":"Target temperature",
"3":"Current temperature",
"6":"Working status",
"7":"Window status",
"8":"Open window",
"9":"Max. limit temperature",
"10":"Min. limit temperature",
"12":"Child lock",
"13":"Battery",
"14":"Fault alarm",
"101":"Room sensor calibration",
"102":"周程序1", // Weekly program 1
"103":"周程序2", // Weekly program 2
"104":"周程序3",	// Weekly program 3
"105":"周程序4", // Weekly program 4
"106":"周程序5", // Weekly program 5
"107":"周程序6", // Weekly program 6
"108":"周程序7", // Weekly program 7
"109":"机型", //model
"110":"Motor thrust",
"111":"Display brightness",
"112":"Software version",
"113":"Screen orientation",
"114":"Valve",
"115":"Switch deviation (energy-saving mode only)",
"116":"电机数据", //motor data
"117":"假期时长", //Holiday duration
"118":"Boost 模式时长", //Boost mode duration
"119":"舒适模式温度", //Comfort mode temperature
"120":"节能模式温度", //Energy saving mode temperature
"121":"防冻模式温度", //Antifreeze mode temperature
"122":"防冻使能", //Antifreeze enable
"123":"气压指数", //pressure index
"124":"临时模式开始时间", //Temporary mode start time
"125":"功能点",	//Function point
"126":"app支持特性", //app support features
"127":"System mode"
}

closed 2025-03-25 by github-actions[bot]

Levak, 2025-12-14:
The fix in [this post](https://github.com/Koenkk/zigbee2mqtt/issues/10489#issuecomment-1312817478) seems to work and resync the time properly for this device too. I removed the `onEvent` attribute. I re-paired, and the schedule was back on track.

Here is the updated external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const logger = require('zigbee-herdsman-converters/lib/logger');

const e = exposes.presets;
const ea = exposes.access;

const allModes = {
    0: { preset:"off"        , systemMode: "off",  deviceMode: "off"        , isSystemMode: true },
    1: { preset:"comfort"    , systemMode: "auto", deviceMode: "comfort"    , },
    2: { preset:"manual"     , systemMode: "auto", deviceMode: "manual"     , },
    3: { preset:"on"         , systemMode: "heat", deviceMode: "on"         , isSystemMode: true },
    4: { preset:"eco"        , systemMode: "auto", deviceMode: "eco"        , },
    5: { preset:"anti-frost" , systemMode: "off",  deviceMode: "anti-frost" , },
    6: { preset:"schedule"   , systemMode: "auto", deviceMode: "schedule"   , isSystemMode: true },
};
const allModesByName = Object.fromEntries(
    Object.entries(allModes)
        .map(([k, v]) => [
            v.preset, { ...v, 'index': k }
        ])
);

function customThermostatScheduleDay() {
    /// Each day is split into slots of 30minutes.
    /// Each slot is 4 bits, encoding the mode in which the valve should be put into.
    /// e.g. 69 20 ...
    /// Means at 00:00 use mode (69 >> 4) = 4 aka "eco"
    ///       at 00:30 use mode (69 & 15) = 5 aka "anti-frost"
    ///       at 01:00 use mode (20 >> 4) = 1 aka "comfort"
    ///       at 01:30 use mode (20 & 15) = 4 aka "eco"
    return {
        from: (v) => {
            const daySchedule = [];
            let previousMode = null;
            for (let index = 0; index < 48; index++) {
                const hour = Math.floor(index / 2);
                const minutes = (index % 2) ? 30 : 0;
                const hour_values = parseInt(v[hour], 10);
                const mode = (index % 2) ? (hour_values & 15) : (hour_values >> 4);
                if (allModes[mode] === undefined) {
                    throw new Error(`Invalid mode half-byte: "${mode}" (${hour_values} index ${index})`);
                }
                const modeString = allModes[mode].preset;
                if (mode != previousMode)
                {
                    const schedule = String(hour).padStart(2, '0') +
                          ':' +
                          String(minutes).padStart(2, '0') +
                          '/' +
                          modeString;
                    daySchedule.push(schedule);
                    previousMode = mode;
                }
            }
            return Array.from(new Set(daySchedule)).join(' ');
        },
        to: (v) => {
            const parseTransition = (transition) => {
[... 305 lines cut ...]
            [114, "valve", tuya.valueConverterBasic.lookup({ OPEN: 1000, CLOSED: 0 })],
            [115, "switch_deviation", tuya.valueConverter.divideBy10],
            [117, "holiday_duration", tuya.valueConverter.raw],
            [118, "boost_duration", tuya.valueConverter.raw],
            [119, "comfort_temperature", tuya.valueConverter.divideBy10],
            [120, "eco_temperature", tuya.valueConverter.divideBy10],
            [121, "frost_protection_temperature", tuya.valueConverter.divideBy10],
            [122, "frost_protection", tuya.valueConverter.onOff],

            // Unused
            [109, "mode", tuya.valueConverter.text],
            [112, "version", tuya.valueConverter.text],
            [116, "motor_data", tuya.valueConverter.raw],
            [123, "pressure_index", tuya.valueConverter.raw],
            [124, "temporary_mode_starttime", tuya.valueConverter.raw],
            [125, "function_point", tuya.valueConverter.raw],
            [126, "app_features", tuya.valueConverter.raw],
            [127, "system_mode2", tuya.valueConverter.raw],
        ],
    },

};

module.exports = definition;
```

bernardesarthur, 2025-12-15:
Hi @levak, thanks for your updated converter. I'll take a look on that. 🙏🏼