Koenkk/zigbee2mqtt #23755: ZigBee Smart Thermostat Module Part Touch Screen

tobatya, 2024-08-27:
### Link

https://aliexpress.ru/item/1005005288243919.html?spm=a2g2w.orderdetail.0.0.3e1d4aa6qJsxyC&sku_id=12000032501024262

### Database entry

{"id":23,"type":"Router","ieeeAddr":"0xa4c1383466b207eb","nwkAddr":22908,"manufId":4417,"manufName":"_TZE200_7rghpoxo","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":",�`.i.�`.i2�`.i5�`.f6�`.\u0012","65506":54,"65508":0,"stackVersion":0,"dateCode":"","appVersion":70}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1724768850936}

### Comments

I think the device is similar to MOES Zigbee Thermostat Room Temperature Controller of Water/Electric Floor Heating
[image]

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE200_7rghpoxo',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

tobatya, 2024-08-27:
I tried to add an external converter file but it doesn't help. The device is not detected
onst fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const utils = require('zigbee-herdsman-converters/lib/utils');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;
const globalStore = require('zigbee-herdsman-converters/lib/store');

const definition = {
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_7rghpoxo'}],
    model: 'TS0601_Smart Thermostat',
    vendor: 'TuYa',
    description: 'Smart thermostat',
	fromZigbee: [fz.legacy.tuya_thermostat_weekly_schedule, fz.etop_thermostat, fz.ignore_basic_report, fz.ignore_tuya_set_time],
	toZigbee: [tz.etop_thermostat_system_mode, tz.etop_thermostat_away_mode, tz.tuya_thermostat_child_lock,
		tz.tuya_thermostat_current_heating_setpoint, tz.tuya_thermostat_weekly_schedule],
	onEvent: tuya.onEventSetTime,
	meta: {
		thermostat: {
			weeklyScheduleMaxTransitions: 4,
			weeklyScheduleSupportedModes: [1], // bits: 0-heat present, 1-cool present (dec: 1-heat,2-cool,3-heat+cool)
			weeklyScheduleFirstDayDpId: tuya.dataPoints.schedule,
		},
	},
	exposes: [e.child_lock(), exposes.climate().withSetpoint('current_heating_setpoint', 5, 35, 0.5, ea.STATE_SET)
		.withLocalTemperature(ea.STATE)
		.withSystemMode(['off', 'heat', 'auto'], ea.STATE_SET).withRunningState(['idle', 'heat'], ea.STATE)
		.withAwayMode()],
};

module.exports = definition;

alehelley, 2024-09-10:
Hello, 

I have exactlly the same problem with a Samart Radiator valve. 
I have 20 of them. Is spent hours and hours trying to configure it. I'm desperated. 
I managed to make it supported by Z2M in HA by creating an external converter, but noting he exposed. 

This is what i got in database.db : 

{"id":23,"type":"EndDevice","ieeeAddr":"0xa4c138e87c289ab1","nwkAddr":8057,"manufId":4417,"manufName":"_TZE200_yqgbrdyo","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1725995036406}

Heres is my external converter : 
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const e = require('zigbee-herdsman-converters/lib/exposes').presets;
const ea = require('zigbee-herdsman-converters/lib/exposes').access;
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601_thermostat_3',
    vendor: 'Tuya',
    description: 'Thermostatic radiator valve',
    fromZigbee: [fz.thermostat, fz.ignore_basic_report],
    toZigbee: [tz.thermostat_local_temperature, tz.thermostat_occupied_heating_setpoint, tz.thermostat_system_mode, tz.on_off],
    exposes: [
        e.climate().withSetpoint('occupied_heating_setpoint', 5, 30, 0.5).withLocalTemperature(),
        e.binary('state', ea.ALL, 'ON', 'OFF').withDescription('Thermostat state'),
        e.enum('system_mode', ea.ALL, ['off', 'auto', 'heat']).withDescription('System mode')
    ],
    meta: {multiEndpoint: true},
};

module.exports = definition;

Please Help

alehelley, 2024-09-11:
Thank you for your answer. 
Unforunatly when I updated the external_converter.js with the fingerpint, the devices went to supported to unsuported ;( 
I'm lost...

Here is the .js I updated with the fingerprint : 

[image]

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const e = require('zigbee-herdsman-converters/lib/exposes').presets;
const ea = require('zigbee-herdsman-converters/lib/exposes').access;
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');

const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE200_yqgbrdyo',
        }
    ]
    zigbeeModel: ['TS0601'],
    model: 'TS0601_thermostat_3',
    vendor: 'Tuya',
    description: 'Thermostatic radiator valve',
    fromZigbee: [fz.thermostat, fz.ignore_basic_report],
    toZigbee: [tz.thermostat_local_temperature, tz.thermostat_occupied_heating_setpoint, tz.thermostat_system_mode, tz.on_off],
    exposes: [
        e.climate().withSetpoint('occupied_heating_setpoint', 5, 30, 0.5).withLocalTemperature(),
        e.binary('state', ea.ALL, 'ON', 'OFF').withDescription('Thermostat state'),
        e.enum('system_mode', ea.ALL, ['off', 'auto', 'heat']).withDescription('System mode')
    ],
    meta: {multiEndpoint: true},
};

module.exports = definition;

alehelley, 2024-09-11:
Here is my external definition : 

[image]

alehelley, 2024-09-13:
OMG !!! Thank you so much !!! 
It works ! You can't imagine how grateful I am ! 
You gave an end to weeks of late night sleep ! 
Thank you so much

Koenkk, 2024-09-13:
Added the device for out of the box support, thanks @dweng0 !

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-09-13 by Koenkk

Kdydis, 2024-09-15:
But problem with cheap touchscreen electric floor heating TS0601 _TZE200_7rghpoxo still remains.

I tried to use TS0601_floor_thermostat converter as external converter example. Small success: Switch, Child lock and Clock appeared. But no more.

From Tuya developer system I got datapoints, but with them my converter crashes.

"1":"Switch"
"2":"Temp Set"
"3":"Temp Current"
"9":"Child Lock"
"11":"Fault"
"15":"Upper Temp"
"16":"Lower Temp"
"19":"Temp Correction"
"11":"External temperature calibration"
"102":"Work State"
"103":"sleep"
"104":"Weekly programming"
"105":"week"
"106":"hour"
"107":"minute"
"108":"temp"
"109":"time frame"

Maybe a good person will show a working example?

agafonovkirill, 2024-10-20:
> But problem with cheap touchscreen electric floor heating TS0601 _TZE200_7rghpoxo still remains.

I have the same problem. Does you reach any solution?